python-microscopy/python-microscopy #1165: [SUPPORT] PYMEAcquire config doc pretty lacking

barentine, 2021-12-07:
We've added some metadata mandates / shifted things around a bit and I'm not sure we're keeping up with the doc.

Specifically we ought to update things after PR #1131 , 
[image]
to include exactly what one needs to to in order to start PYMEAcquire with their camera and acquire a snapshot without digging through the metadata rabbit hole

```
 Error whilst running Acquire>Snapshot	F5
==============================================
python-microscopy=21.09.23.post0.dev[git]
python=3.7.10, platform=win32
numpy=1.19.2, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\userfiles\code\python-microscopy\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\acquiremainframe.py", line 486, in OnMAquireOnePic
    mdgen(mdh)
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\Hardware\Camera.py", line 784, in GenStartMetadata
    mdh.setEntry('Camera.Name', self.GetName())
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\Hardware\Camera.py", line 177, in GetName
    raise NotImplementedError('Implemented in derived class.')
NotImplementedError: Implemented in derived class.

```

David-Baddeley, 2021-12-07:
ahh, I let the `GetName()` in by accident - I'm not sure it's really useful and it should definitely fail safely. Docs could also use an overhaul

David-Baddeley, 2021-12-13:
Low hanging fruit addressed in #1163

barentine, 2022-01-28:
other low hanging fruit would be to give a brief description of how to configure a scope if you aren't using init script / protocols in the main repo. Quick example of calling an external init script, and description of where to save protocols so they are automatically found (`config_dir/protocols/` would be good

barentine, 2022-01-28:
re: protocols I think it would also be nice to configure a separate protocols directory, such that one might be able to keep all of their PYME-custom scope stuff in a single repository of their own

barentine, 2023-01-24:
addressed for now with #1224 and #1163

closed 2023-01-24 by barentine