python-microscopy/python-microscopy #1176: [SUPPORT] More explanations for the data_xyztc 5D data model?

csoeller, 2021-12-26:
**What are you trying to do?**
I am still trying to get my head round the data_xyztc 5D data model and its intended use (+ potential backward compatibility).

**Where are you stuck?**
A number of issues.

**Acquiring xyt data with PYMEAcquire**

So we acquire data with PYMEAcquire, latest PYME from git (as of this writing). These are as acquired just time series of frames. How do we tell PYME this (e.g. in metadata)? I loaded one of these series with `dh5view` and if I do

```
      image.data_xyztc.shape

(512, 512, 711, 1, 1)
```

I.e. PYME thinks this is xyz data. Is this expected? Should there have been a metdata setting that tells PYME this is xyt data?
How can I change this after the fact? I wasn't sure I understand how to use the `set_dim_order_and_size` function, if that is indeed one of the tools to use?

**Converting older code**

I was going to use the `LabelsFromImage` module. This in turn uses `cluster_morphology.get_labels_from_image`. This one still uses the older `data` instead of `data_xyztc`. I feel like I want to convert this to `data_xyztc` but am now not so confident as I am not sure I understand the usage `data_xyztc` and its backward compat promotion behaviour fully.

If there are docs explaining this that I have overlooked, please just point me to it.

Generally, having a proper 5D model is the right way to go rather than implied (fuzzy) assumptions that can break at a whim. If not yet available a documentation page describing the 5D model would be great. I'd be happy to contribute to it (once I understand the intended behaviour).

zacsimile, 2021-12-27:
Hi @csoeller! You are right, there is not much documentation on this. @David-Baddeley  will have more to offer, but let me give it a crack and see if I can help.

Most of the code is in `PYME.Acquire.xyztc.py` and `PYME.IO.DataSources.BaseDataSource.XYZTCDataSource`. We currently acquire exclusively as `XYCZT`, but we can re-order after acqusition as needed. I believe you could change the data source order by changing the passed `input_order` in `PYME.IO.DataSources.BaseDataSource.XYZTCDataSource`. The `PYME.recipes.base.Redimension` recipe does this and has an example of how to use `set_dim_order_and_size`.

Re: the issue you are having in `dh5view`. If I record data with `PYMEAcquire`'s simulator, the resulting image I open in `dh5view` slides along the `t` direction. Similarly, I am able to properly acquire and open images using a custom object inherited from `xyztc.XYZTCAcquisition`. What acquisition object are you using/how are you acquiring images in `PYMEAcquire`?

I'm happy to work with you on an updated `LabelsFromImage` module. Maybe we could try a pair coding session some time soon?

David-Baddeley, 2021-12-27:
Hi Christian,

The documentation is indeed very patchy/non existent, partly because the
transition is still a bit of a work in progress...

What acquisition mode were you using (can I safely assume spooling?), and
were you saving locally or to a cluster? Xyztc support in PYMEAcquire is in
its infancy, and I'm not sure we're setting the metadata correctly for all
modes.

When loading, we assume xyz if metadata is missing.

The redimension recipe module (and processing->redimension in the dh5view
menu) will let you correct the dimensionality of a loaded image. I think
there are some de-facto docs for this in another issue, but the number
escapes me for now - might be retrievable through search.

The backwards compatibility mode simply gives you a flattened z/t
dimension. In general this will work as expected/as it did previously.

There might need to be some thought put in to making the labels from image
function z/t compatible, as t in the localisation data would usually not
map to t in the image (ie it currently only has an obvious use case for xyz
images).

In general, when converting modules, some thought should be given to memory
usage (i.e. If you get data which is truly xyztc, you might not want to
force it into a very large contiguous array). Partly for this reason, a lot
of the recipe module conversion is being delayed until we know what
chunking is going to look like.

csoeller, 2021-12-29:
Hi @zacsimile and @David-Baddeley, many thanks for your feedback, much appreciated.

I start with the acquisition. We have acquired via spooling to File or Cluster, using PYMEAcquire and actual cameras/hardware. When I open these types of `.h5`files with `dh5view`, then these seem to come up as x,y,z, see also screenshot.

[image]

Would it make sense to autoset a suitable metadata entry to mark these as x,y,t? I appreciate the possible difference of t between camera data and localisation data. Again, is a possible starting point to have the ability to distinguish these based on metadata entries that we provide (and possibly heuristics for legacy data)?

I appreciate that the 5D implementation is still in flux, just thinking that marking things suitably at acquisition time, e.g. via metadata, sounds like low hanging fruit...

Secondly, the `LabelsFromImage` module, I have made a module closely based on it, but not quite the same, as we are not setting discrete labels but looking up values from a map (to set local N0 values for the `SIMPLER` algorithm implementation). See `N0FromImage` in `PYMEcs.recipes.simpler`, see [here](https://github.com/csoeller/PYME-extra/blob/master/PYMEcs/recipes/simpler.py). For now, going with using `data_xyztc` directly was ok as we are really only dealing with 2D images here, and z, c, t are trivially of size 1. So this does not really take care of the chunking question...

@zacsimile, would it anyway be useful to discuss more about the module conversion?

David-Baddeley, 2021-12-29:
first reply was pretty fast (whilst juggling our 3 y.o.s breakfast), and the blocking aspect was a bit out of context. I neglected to state that this should not be an issue for `LabelsFromImage` as it's essentially just a look-up, and doesn't result in the copying of large chunks of data.

To the data format side of things, dimension ordering in an h5 file is specified by the `.DimOrder`, `.SizeZ`, `.SizeT`, `.SizeC` attributes on the actual data array itself (see, e.g. PYME/IO/dataExporter.py, lines 153-156, and PYME/IO/DataSources/HDFDataSource.py lines 56-64).

On the acquisition side, things are handled in `PYME.IO.acquisition_backends`. Notably this also sets the `DimOrder`, `SizeZ`, `SizeT`, and `SizeC` metadata entries (with the same meaning as the the array attributes above). As it stands, I think .h5 loading ignores these, assuming that the array attributes will be present. 

There are two options for why it's not working as expected - 1) array attributes are not being saved when we spool to h5 on the cluster/cluster of one, and 2) we might not actually set the dimension sizes correctly for time series spooling (as opposed to, eg, z stacks). For case 1), we should both fix this, and probably look in the metadata if the array attributes are missing. This won't help us if for case 2 however.

David-Baddeley, 2021-12-29:
Looking at the code again, case 2 definitely applies (case 1 might also apply) - we are not setting the dimensionality correctly for localisation spooling. In principle we might be able to change this by simply passing a `shape=[-1,-1,1,-1,1]` argument to the backend initialisation calls on line 85 of `HTTPSpooler_v2.py` and  line 84 of `HDFSpooler.py`. 

A `-1' in the shape argument indicates that the data will dictate this (for x and y this is the frame shape, a -1 in one of the z, y, or t dimensions implies that the data grows along this dimension  (i.e. for the above shape, `SizeT`  is inferred from `nFrames/(SizeZ*SizeC)`). Only one of the z, t, or c dimensions can be -1.

Part of the recent spooling refactors was aimed at enabling cluster-streaming of z stacks, tiling, and other acquisition types (as well as standard localisation series). The slightly open question here is whether the spoolers remain specific to time series acquisition (i.e. we can make the above change) or whether they end up being shared with other acquisition modes, in which case we'll need to bubble the setting upwards). I seem to remember that the intention was to have all the shared logic in `acquisition_backends`, but would need to work myself back in a little.

csoeller, 2022-01-04:
Ok, it looks like I was a bit naive how easy this would be to achieve via metadata info. It would be interesting to learn what use cases sharing spoolers with different acquisition modes would cover. In any case, happy to wait for this to shake out once you had a little more time to think this through.

zacsimile, 2022-01-04:
Hi @csoeller. It sounds like maybe the module doesn't need to be changed?

I'm happy to work on acquisition support, but it sounds like this should be done in conjunction with @David-Baddeley. Can you point us to the exact backend you're using for acquisition?

csoeller, 2022-01-04:
Thanks @zacsimile, yes, I'd say the module is ok for now.

On the acquisition side we are doing cluster-based localisation spooling into HDF files. According to one of @David-Baddeley's posts above this appears to involve some of the spooling modules. I have not delved into this part of the code base to date...

David-Baddeley, 2022-01-09:
The main use case of sharing spoolers amongst acquisition modes would be to avoid propagation of multiple parallel sets of code which do the same thing, and as a consequence reduce the maintenance (and adoption) burden.  

Having another look at the spoolers, most of the common logic has already gone into `acquisition_backends`, with the events being pretty much the only thing which is currently handled by the spoolers. For now, I think we can safely assume that spooler == blinking acquisition, and make pass the relevant shape to the backend. We can always revisit if/when we use the spoolers for something else.