python-microscopy/python-microscopy #1179: [BUG] IDS uEye camera stall-outs when changing settings/acquiring images

barentine, 2022-01-06:
**Describe the bug**
I have an IDS UI327x which runs 'fine' until I try to do something in the PYMEAcquire GUI - basically anything: changing integration time sometimes works, but toggling an ROI or acquiring a snapshot is ~gaurunteed to freeze up the camera similar to previous issue (#313). Note that with my previous IDS usage, the HTSMS focus lock, I never had to do anything with the GUI/change camera settings once loaded - I took care of setting my ROI / integration time in the init script and never needed to acquire images with it.

@csoeller is this all working fine in your hands?

**To Reproduce**
Steps to reproduce the behavior:
1. IDS UI327x
2. [image]
3.
```
@init_hardware('Camera')
def cam(scope):
    from PYME.Acquire.Hardware.ueye import UEyeCamera
    cam = UEyeCamera(0, 12)
    scope.register_camera(cam, 'IDS')
```
4. acquire snapshot
5. snapshot is acquired, but camera silently freezes.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Win10
 - Python version: 3.78
 - numpy version: 1.19
 - wxpython version [if a UI bug]: 4.0.4
 - pyueye                    4.95.0                   pypi_0    pypi

csoeller, 2022-01-07:
Hi @barentine, what you describe sounds pretty much like what I encountered. I figured out a workaround which basically involved downgrading to IDS software version 4.93. The nuts and bolts are described in my [installation writeup](https://github.com/csoeller/pyme-install-docs/blob/master/PYMEAcquire/IDS-camera-setup.md). Would be good to know if this works for you as well.

Note that (1) `pyueye` can stay on version `4.95.X` just fine (we are not using any of the new 4.95 features at the moment) and (2) it would be good to eventually tackle the question if upgrading to the 'new' 4.95 API functions for acqusition would allow us to use the latest software.

barentine, 2022-01-09:
Very helpful, @csoeller! Thank you. 

I had a crack at using the new functions for the queue handling. It's behaving a little goofy at the moment - the ImageQueue version of waitfornext is to pass a pointer to an `ImageQueueWaitBuffer` instance, which I do, but the nmemid field which we need to pass to the memcopy function never gets set - it remains a Null pointer.

Anyway, I'll maybe look more, but will at least PR the queue init/exit portions which work fine.

David-Baddeley, 2022-01-09:
Thanks for the help @csoeller! On an implementation side, I can see an argument for having a separate module/class for the new API version as it seems to be sufficiently different (and backwards incompatible) from the previous versions. The new version could be the default, with the old version accessible under a different name, but I think there are cases where we might still want to use older versions of the driver (mainly where this is already installed and/or obscure systems, but potentially also for older cameras).

csoeller, 2022-01-10:
A way to decide which interface to use could be decided on IDS installed software version and/or a switch/keyword we allow in the `__init__` arguments, for example to request legacy or new interface, explicitly, plus suitable defaults. DLL version can be queried using something like

```python
def GetDLLVersion():
    version = ueye.is_GetDLLVersion()
    build = version & 0xFFFF
    minor = version >> 16 & 0xFF
    major = version  >> 24 & 0xFF
 
    return (major,minor,build)
```

barentine, 2022-01-11:
I'm no longer stalling out he camera, which is nice, but still no luck on toggling ROI size (unless I hack in a multiview class for the uEye). Had tried 4.94.2, now downgraded further to 4.93 to match Christian's installs.

Multiview will work fine for me, and I suppose I might PR it, but little weird for the ROI toggle to not be working

csoeller, 2022-01-12:
What exactly do you call 'toggling ROI size'? Probably not the usual `F8` stuff? At least that does seem to work for us with 4.93.

csoeller, 2022-01-12:
I also notice that with the `pyueye` backend the integration time gets screwed up when you change the ROI. This is because the trick being used in the `uc480` does not work here because the last integration time that was set is not cached in the camera object whereas it is in the `uc480` interface. For me it therefore always reverts to 40 ms when changing ROI.

I suppose we should fix that...

barentine, 2022-01-12:
@csoeller I do actually mean the usual `F8` stuff for toggling the ROI! Glad to know it works for you. I tried 4.93 but am still having issues for some reason. 

Thanks for noticing the integration time oddity. I'm stuck on some alignment at the moment, but hopefully at some point in the near future I'll have a bit to focus on some software improvements for my setup and will work on some of these

csoeller, 2022-01-12:
@barentine, I suppose you have checked that it is not some ROI related internal logic issue, e.g. the camera ROI constraints were not correct or similar? I can't imagine you haven't checked for all these types of issues. All I can report from here is that we have ok behaviour now with 2 PCs on 4.93 and 2 different camera models, including ROI changes.

I tried a quick cached integration time implementation, that fixes the integration time reset on ROI toggling.

barentine, 2022-11-29:
closing as fixed with #1180

closed 2022-11-29 by barentine