python-microscopy/python-microscopy #1187: [BUG] Mac M1 support

David-Baddeley, 2022-01-20:
This is a bit of a super-issue to track anything we come up with

# Rosetta

Need to make an alias copy of `Applications/Utilities/Terminal.app`, and set terminal to use Rosetta

## existing conda development install (python=3.7)
from time machine backup, and likely indicative of the executable installer 

works with following caveats :

- sporadic "Illegal Insruction 4" message and hard crash (most likely the result of compilation with SSE/AVX support enabled). Reliably triggered by:
    - isosurfaces in dh5view
    - pointsprites in PYMEVis
    - running test suite

## Development install (python=3.7)

- Resolves visualisation related "Illegal Instruction 4" messages
- Still get "Illegal Instruction 4"  on test suite

# Native

- no experience so far

csoeller, 2022-08-17:
### Native build Aug 22

I just got my hands on an MBA-M2 and gave this a shot. Things seem to have become a little easier.
I started with a native miniconda install:

```shell
wget https://repo.anaconda.com/miniconda/Miniconda3-latest-MacOSX-arm64.sh # as seen at https://towardsdatascience.com/how-to-install-miniconda-x86-64-apple-m1-side-by-side-on-mac-book-m1-a476936bfaf0
sh Downloads/Miniconda3-latest-MacOSX-arm64.sh  # accept all defaults for miniconda prompts
```

That installed a small base install with Python 3.9.

I made an environment to try things out in and based that again on Python 3.9:

```shell
conda create -n pyme39_2 python=3.9
conda activate pyme39_2 
```

Then mostly followed @David-Baddeley s instructions above, the wxpython wheel seemed to install ok, I did not try the custom build (not sure if this wheel will be ARM native or not though?):

```shell
conda install -c conda-forge numpy scipy matplotlib pytables pyopengl jinja2 cython pip requests pyyaml psutil pandas scikit-image scikit-learn sphinx
conda install -c conda-forge traits traitsui==7.1.0 pyface==7.1.0
pip install wxpython # this seems to generate a working wxpython - I tried once with the version from conda-forge but no luck
conda install -c conda-forge python.app # without that no python.app was available!
```

Finally, build the latest python-microscopy from github (as of mid Aug 22):

```shell
/Users/csoe002/miniconda3/envs/pyme39_2/python.app/Contents/MacOS/python setup.py develop
conda install -c conda-forge toposort # also pick up toposort as we need it for dh5view etc
```

This seems to give a working install, still testing but most things with `dh5view` and `PYMEVis` seem fine. If anybody has further insights please comment.

David-Baddeley, 2022-08-17:
Thanks for the update. A couple of thoughts:

- I don't expect many issues with py 3.9, but it hasn't been well tested.
Py 3.8 is fairly well tested by now, and a safer option.

- it's probably worth checking the numpy build info. Back when I was
experimenting with this, the conda forge M1 package was not built against
any vector acceleration stuff and you could get a significant performance
boost by building it yourself (detailed above).

csoeller, 2022-08-18:
Thanks for the hints. I have tried the suggestions and this is the latest approach with accelerated numpy that seems to work well:

```shell
conda create -n pyme38_2 python=3.8
conda activate pyme38_2

# install accelerated numpy
conda install -c conda-forge cython pybind11
pip install --no-binary :all: --no-use-pep517 numpy # this gives us an accelerated numpy, see also https://stackoverflow.com/questions/69848969/how-to-build-numpy-from-source-linked-to-apple-accelerate-framework
conda config --set pip_interop_enabled true # apparently needed so no new numpy is pulled in in next conda command…, see also https://stackoverflow.com/questions/70240506/why-python-native-on-m1-max-is-greatly-slower-than-python-on-old-intel-i5

# install other dependencies
conda install -c conda-forge scipy matplotlib pytables pyopengl jinja2 cython pip requests pyyaml psutil pandas scikit-image scikit-learn sphinx
conda install -c conda-forge traits traitsui==7.1.0 pyface==7.1.0
conda install -c conda-forge python.app
conda install -c conda-forge toposort
pip install wxpython

# build PYME
cd pyme38 # this should be your directory with the latest PYME source
/Users/csoe002/miniconda3/envs/pyme38_2/python.app/Contents/MacOS/python setup.py develop
dh5view -t
```

I'll keep further updates at an [ARM64 version](https://github.com/csoeller/pyme-install-docs/blob/master/Installing-PYME-with-py3-macos-ARM64.md) of our [install docs](https://github.com/csoeller/pyme-install-docs).

barentine, 2023-01-26:
I have an M1 now and went through this today with py38 - thanks for the docs @David-Baddeley and @csoeller. 

I'm not trying to be particularly performant with my laptop at the moment, I just want to be able to code at home easily, so I installed the arm64 miniconda 3.8 as Christian did, but then I was lazy and took the conda-forge numpy and wx builds.

A couple packages not mentioned in Christian's above post (and possibly not in our M1 install docs right now) but required for our tests: ifaddr, zeroconf, ujson, cherrypy, pyfftw, pymecompress

Some packages aren't available on conda-forge. For M1 pyme, python 3.8, I had to build the following, which are available on my conda channel:
- [ifaddr](https://anaconda.org/barentine/ifaddr) 
- [zeroconf](https://anaconda.org/barentine/zeroconf) 
- [pymecompress](https://anaconda.org/barentine/pymecompress)

<details>  <summary> my conda list </summary>  

```
# packages in environment at /Users/andy/miniconda3/envs/pyme38:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12             pyhd3eb1b0_0  
anaconda-client           1.11.0           py38hca03da5_0  
attrs                     22.1.0           py38hca03da5_0  
autocommand               2.2.2              pyhd8ed1ab_0    conda-forge
babel                     2.11.0           py38hca03da5_0  
backports                 1.1                pyhd3eb1b0_0  
backports.functools_lru_cache 1.6.4              pyhd3eb1b0_0  
backports.tempfile        1.0                pyhd3eb1b0_1  
backports.weakref         1.0.post1                  py_1  
beautifulsoup4            4.11.1           py38hca03da5_0  
blas                      2.116                  openblas    conda-forge
blas-devel                3.9.0           16_osxarm64_openblas    conda-forge
blosc                     1.21.0               h98b2900_1  
bottleneck                1.3.5            py38heec5a64_0  
brotli                    1.0.9                h1a28f6b_7  
brotli-bin                1.0.9                h1a28f6b_7  
brotlipy                  0.7.0           py38h1a28f6b_1002  
brunsli                   0.1                  hc377ac9_1  
bzip2                     1.0.8                h620ffc9_4  
c-ares                    1.18.1               h1a28f6b_0  
ca-certificates           2022.12.7            h4653dfc_0    conda-forge
cctools                   949.0.1             hc179dcd_25  
cctools_osx-arm64         949.0.1             h332cad3_25  
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py38h80987f9_3  
cfitsio                   3.470                h7f6438f_7  
chardet                   4.0.0           py38hca03da5_1003  
charls                    2.2.0                hc377ac9_0  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
cheroot                   9.0.0              pyhd8ed1ab_0    conda-forge
cherrypy                  18.8.0           py38h10201cd_2    conda-forge
click                     8.0.4            py38hca03da5_0  
cloudpickle               2.0.0              pyhd3eb1b0_0  
clyent                    1.2.2            py38hca03da5_1  
colorama                  0.4.6            py38hca03da5_0  
conda                     22.11.1          py38h10201cd_1    conda-forge
conda-build               3.23.3           py38h10201cd_1    conda-forge
conda-package-handling    2.0.2            py38hca03da5_0  
conda-package-streaming   0.7.0            py38hca03da5_0  
conda-verify              3.4.2                      py_1  
contourpy                 1.0.5            py38h525c30c_0  
cryptography              38.0.4           py38h834c97f_0  
cycler                    0.11.0             pyhd3eb1b0_0  
cython                    0.29.32          py38hc377ac9_0  
cytoolz                   0.12.0           py38h1a28f6b_0  
dask-core                 2022.7.0         py38hca03da5_0  
docutils                  0.18.1           py38hca03da5_3  
expat                     2.5.0                hb7217d7_0    conda-forge
fftw                      3.3.9                h1a28f6b_1  
filelock                  3.9.0            py38hca03da5_0  
fonttools                 4.25.0             pyhd3eb1b0_0  
freetype                  2.12.1               h1192e45_0  
fsspec                    2022.11.0        py38hca03da5_0  
[... 175 lines cut ...]
tornado                   6.2              py38h1a28f6b_0  
tqdm                      4.64.1           py38hca03da5_0  
traitlets                 5.7.1            py38hca03da5_0  
traits                    6.3.1            py38hea4295b_0    conda-forge
traitsui                  7.1.0              pyh9f0ad1d_0    conda-forge
typing-extensions         4.4.0                hd8ed1ab_0    conda-forge
typing_extensions         4.4.0              pyha770c72_0    conda-forge
ujson                     5.5.0            py38h2b1e499_1    conda-forge
urllib3                   1.26.14          py38hca03da5_0  
wheel                     0.37.1             pyhd3eb1b0_0  
wxpython                  4.2.0            py38hddee93b_2    conda-forge
xz                        5.2.10               h80987f9_1  
yaml                      0.2.5                h1a28f6b_0  
zc.lockfile               2.0                pyhd8ed1ab_1    conda-forge
zeroconf                  0.26.3                   py38_3    barentine
zfp                       0.5.5                hc377ac9_6  
zipp                      3.11.0           py38hca03da5_0  
zlib                      1.2.13               h03a7124_4    conda-forge
zstandard                 0.18.0           py38h1a28f6b_0  
zstd                      1.5.2                h8574219_0  
```

</details>

With a couple publications on the near horizon it might be nice to have an M1 install route open to folks who just want to conda install.

David-Baddeley, 2023-01-26:
Last I checked github actions didn't let us build packages for M1, so we'd need to do it all manually every time we release (which is a bit of a hassle). I'm not sure we really want to go there.

What's a bit weird is that I got ifadddr and zeroconf from conda forge no problems. Both are pure python (no need for an architecture dependent version).

barentine, 2023-01-26:
Huh.... which zeroconf version did you pull? Ifaddr makes sense - sorry for the noise there

David-Baddeley, 2023-01-26:
0.38.1

David-Baddeley, 2023-01-26:
but it was a while ago

David-Baddeley, 2023-01-26:
Might have fixed things so we could remove the pin in pymedepends without then removing it.

barentine, 2023-01-26:
Fair enough. It'd be nice if we could remove the pin ([I've been confused about it before](https://github.com/python-microscopy/pyme-conda-recipes/issues/12)). Thing to check would be if we can move beyond 0.26.3 - previously we couldn't because of a backwards incompatible change in the ServiceInfo address constructor on 0.27.

Regardless - sorry for the adding the noise with ifaddr / zeroconf to this issue

David-Baddeley, 2023-01-26:
Think it's legit if we need to address pins etc ... to get it to install.

barentine, 2023-05-02:
Has anyone tried bioformats on M1 yet? I've got some obf files I'd like to open in PYME and am debating whether to give the JRE/bioformats stuff a shot or write / PR something native in Python based on e.g. [this MIT-licensed obf reader.](https://github.com/jkfindeisen/python-mix/tree/main/imspector/obf)

David-Baddeley, 2023-05-02:
I haven't tried bioformats yet (it's been on the TODO list for a while). It would definitely be worth a try, although it's hard to know how complex it will be. It does sound like there is an aarch64 version of the JDK which should be the principle requirement.

zacsimile, 2023-05-02:
I also haven't tried bioformats on the M1 yet.