python-microscopy/python-microscopy #1190: [BUG] PYMEImage crosshairs drawn between pixels

barentine, 2022-01-24:
**Describe the bug**
[image]

**Expected behavior**
crosshair should show which pixel is selected, not the corner of which (now one has to know the edge-case behavior to figure out what we've done)

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10
 - Python version: 3.7
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2022-02-09:
We'd probably also need to adjust the selection logic to add a half-pixel offset when you click as well.

barentine, 2022-04-26:
Rather than filing a separate ticket I'll add on to this one. If you display an image larger than you have room for on your screen, with the crosshair present (e.g. in the center of the image), if you copy the display or save a snapshot to png, the crosshairs will be (pretty wildly) the wrong size/place (potentially not scaled to the aspect ratio at all? not sure).

Note the crosshairs aren't always on a corner or in the center, sometimes they're just weirdly neither. 

While I'm on it I'll mention how nice it would be if we displayed intensity by channel next to the `x`, `y` pixel position display, more like a pyplot figure.

barentine, 2022-04-26:
forgot to mention, on windows 10, wx 4.0.4, python 3.7, like #1210 potentially

barentine, 2022-04-26:
Note that when scrolling an image larger than the screen, if you move small enough the pixels don't move but the crosshair does (i.e. it's a float instead of int somewhere)