python-microscopy/python-microscopy #1194: [BUG] context manager issue(?) in arduino code

barentine, 2022-01-28:
**Describe the bug**
First, I'd like to rename `IOSlave`. Mind if I do that @David-Baddeley? Some examples [here](https://en.wikipedia.org/wiki/Master/slave_(technology)). How about `IOAgent`?

Second, seems to be the case that my Uno is not enjoying the context management. Here I'm printing the argument we construct in `IOSlave.SetDigital` which is as expected, however nothing happens on the pin. If I use the same serial args and open the port (without a context manager) and write the command manually, no issue the pin changes (as readline confirms).
In [94]: uno = ioslave.IOSlave(port='COM3')

In [95]: uno.SetDigital(4, 1)
b'SD4 1\n'
Out[95]: "b''"

In [96]: ser = serial.Serial(**uno.ser_args)

In [97]: ser.write(b'SD4 1\n')
Out[97]: 6

In [98]: ser.readline()
Out[98]: b'1.00\r\n'

**To Reproduce**
1. plug in an arduino
2. create an `IOSlave` instance
3. try and send commands, wonder what happened
4. do it manually as described above and see that it works

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10
 - Python version: 3.7
 - numpy version: 1.19
 - pyserial version: 3.5

barentine, 2022-01-28:
ah, sorry, forgot to mention doing the context manager explicitly doesn't work:
```
In [108]: with serial.Serial(**uno.ser_args) as ser:
     ...:     ser.write(b'SD4 1\n')
     ...:     print(ser.readline())
     ...:
b''

In [109]: with serial.Serial(**uno.ser_args) as ser:
     ...:     ser.write(b'SD4 0\n')
     ...:     print(ser.readline())
     ...:
b''
```

David-Baddeley, 2022-01-28:
The context management stuff is weird as it should work, and, when it does,
should be more reliable. I wonder if there was a recent change to pyserial

As to the naming, that's a huge rabbit hole I'm not sure I really want to
go down, and certainly not a bridge I want to fight for, but the thought of
eliminating words from a language simply because of a remote and
unsubstantiated feat they *might* cause offense doesn't sit well with me.
In this case, the term slave accurately describes the function of the
module ... it exists soley to serve the main application, should do what
it's told, and does not recieve any reward for it's work. Using the word
slave is a far cry from saying "slavery is good". "Agent" is a much more
ambiguous description - an agent is expected to have agency and the
capability to make some decisions for themselves. It's a bit like calling a
waiter a customer service representative - it makes it much harder to
quickly guess what something does. If we have to change, I'd probably opt
for something like proxy (although that is still a little ambiguous).

David-Baddeley, 2022-01-28:
PS, if we ditch context management, we have to find another way to make it
thread-safe. In practice, this probably means having a separate polling
thread and using queues to communicate with it which adds a lot of
complexity. I think all calls to a serial port have to be made in the
thread that opened it (which is why we started opening and closing on each
call with the context manager in the first place).

I wonder if adding some `ser.flush()` calls might fix things,