python-microscopy/python-microscopy #1204: [SUPPORT] further rationalising logging in PYME?

csoeller, 2022-02-16:
**Issue:**

While there is now good support for configurable logging from `PYMEAcquire`, some issues remain, that are at least a little obscure to me in (a) if they be solved by configuration and (b) remaining potential inconsistencies.

**Issue a:**

`matplotlib` leaves substantial footprints in the log file when at debug level (happens with the default logging as set in the repo). I cannot suss out any easy way to get rid of these using just a logging config file. What does work is explicitly forcing matplotlib loggers to a higher debug level, i.e. the following one-line patch to `PYMEAcquire.py`:

```python
    else:
        logging.basicConfig(level=default_level)

    # PATCH line below: we want to somehow supress the level of matplotlib messages below debug level
    logging.getLogger('matplotlib').setLevel(logging.WARN)

class BoaApp(wx.App):
    def __init__(self, options, *args):

```

Can we add something like this, or via some other route, to supress `matplotlib` debug messages which are really noise within `PYMEAcquire` logs?

**Part b in next comment...**

csoeller, 2022-02-16:
**Issue b:**

While PYME now has greatly improved logging there are still a few inconsistencies where I am uncertain if this is by design or remnants from earlier itereations of the code. Case in point, in `HTTPTaskPusher.py` we have a mixture of `logging.XXX` and `logger.XXX` calls, e.g.

```python
...
    def fileTasksForInputs(self, **kwargs):
        from PYME.IO import clusterIO
        input_names = kwargs.keys()
        inputs = {k : kwargs[k] if isinstance(kwargs[k], list) else clusterIO.cglob(kwargs[k], include_scheme=True) for k in input_names}

        numTotalFrames = len(list(inputs.values())[0])
        self.currentFrameNum = 0

        logger.debug('numTotalFrames = %d' % numTotalFrames)
        logger.debug('inputs = %s' % inputs)

        while numTotalFrames > (self.currentFrameNum + 1):
            logging.debug('we have unpublished frames - push them')

...
```

The effect of this is that `logging.XXX` generally goes to stdout/stderr and bypasses any settings from the logging config file. This is a little annoying as when I consult the log file these statements are all missing and I am unclear if this is desired. I.e. should we loose these `logging.XXX` statements completely or convert them to `logger.XXX` calls?

**Issue c:**

This is pretty much a continuation of issue b) above, it refers to bunches of remaining `print()` statements in the code that pollute, for example

[image]

I'd be happy to convert these into `logger.debug()` or similar statements as I come across them and submit some clean-up PRs if we can agree what the policy may be to deal with these types of statements as PYME is maturing.

David-Baddeley, 2022-03-01:
sorry for being out of action for a couple of weeks - agree that rationalisation is desirable. Some quick thoughts ...

a) matplolib logging is patched in both dh5view and PYMEVis (just leaving PYMEAcquire), this should give us a template
b) the `logging.debug` calls should definitely be changed to `logger.debug` (I'm not sure that is significantly different to what you proposed above, but have some recollection that it wasn't just matplotlib that needed patching).
c) some of these can probably be ditched completely, but potentially need to be looked at on a case by case basis. In general most of the print statements are going to need a bit more work than a simple replacement with a call to `logger.debug()` - most of them need more context, and some should be at `info()`, `warning` or above levels.One advantage to print, rather than logging statements is that they do show up in e.g. jupyter whereas logging statements disappear into the ether. If we are going to switch all the print statements, we should probably do something so that we always get a basic/minimal logging setup regardless of where we run them from.

csoeller, 2022-03-02:
Sounds good, I will justy try a few obvious ones and the PYMEAcquire bit and we can go from there.

csoeller, 2023-12-09:
logging seems better now, probably still some bits need doing; still, ok to close now