python-microscopy/python-microscopy #1205: [BUG] PZF PCS series loading through ClusterOfOne / 6D data model issue

barentine, 2022-02-16:
**Describe the bug**
```
filename == pyme-cluster:///Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs
DEBUG:PYME.IO.cluster_directory:Request for b'http://127.0.0.1:56151/Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/' failed with error: 404
DEBUG:PYME.IO.clusterIO:Could not find Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/metadata.json, retrying ...
DEBUG:PYME.IO.cluster_directory:Request for b'http://127.0.0.1:56151/Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/' failed with error: 404
DEBUG:PYME.IO.clusterIO:Could not find Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/metadata.json, retrying ...
DEBUG:PYME.IO.cluster_directory:Request for b'http://127.0.0.1:56151/Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/' failed with error: 404
DEBUG:PYME.IO.clusterIO:could not find file Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/metadata.json on cluster: cluster nodes = ['PYMEDataServer [Andrews-MacBook-Pro-5]:Andrews-MacBook-PID:3158']
Traceback (most recent call last):
  File "/opt/miniconda3/envs/pdev/lib/python3.7/site-packages/wx/core.py", line 3259, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/dsviewer.py", line 478, in LoadData
    im = ImageStack(filename=args[0], queueURI=options.queueURI, mdh=options.metadata, haveGUI=True)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/image.py", line 242, in __init__
    self.Load(filename, prompt=load_prompt, haveGUI=self.haveGUI)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/image.py", line 1222, in Load
    self._loadClusterPZF(filename)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/image.py", line 570, in _loadClusterPZF
    self.dataSource = ClusterPZFDataSource.DataSource(filename)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/DataSources/ClusterPZFDataSource.py", line 58, in __init__
    self.mdh.update(json.loads(clusterIO.get_file(mdfn, self.clusterfilter)))
  File "/Users/Andrew/code/python-microscopy/PYME/IO/clusterIO.py", line 505, in get_file
    raise IOError("Specified file could not be found: %s" % filename)
OSError: Specified file could not be found: Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/metadata.json
Finished main loop
(pdev) Andrews-MacBook-Pro-5:code Andrew$ PYMEImage pyme-cluster:////Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs
INFO:numexpr.utils:NumExpr defaulting to 8 threads.
/Users/Andrew/code/python-microscopy/PYME/DSView/splashScreen.py:117: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
Starting main loop
Loading data
filename == pyme-cluster:////Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs
DEBUG:PYME.IO.cluster_directory:Request for b'http://127.0.0.1:56151/Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/' failed with error: 404
DEBUG:PYME.IO.cluster_directory:Request for b'http://127.0.0.1:56151//Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/' failed with error: 404
DEBUG:PYME.IO.cluster_directory:Request for b'http://127.0.0.1:56151/Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/' failed with error: 404
Traceback (most recent call last):
  File "/opt/miniconda3/envs/pdev/lib/python3.7/site-packages/wx/core.py", line 3259, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/dsviewer.py", line 488, in LoadData
    vframe = DSViewFrame(im, None, im.filename, mode = mode)
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/dsviewer.py", line 116, in __init__
    self.do = DisplayOpts(self.image.data_xyztc)
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/displayOptions.py", line 90, in __init__
    self.SetDataStack(datasource)
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/displayOptions.py", line 260, in SetDataStack
    if np.iscomplexobj(self.ds[0,0, 0, 0, 0]):
  File "/Users/Andrew/code/python-microscopy/PYME/IO/DataSources/BaseDataSource.py", line 347, in __getitem__
    return out.squeeze(axis=tuple(to_squeeze))
ValueError: cannot select an axis to squeeze out which has size not equal to one
```

**To Reproduce**
Steps to reproduce the behavior:
1. `PYMEClusterOfOne`
2. `PYMEAcquire` simulator, uncheck spool to h5
2. spool a series
3. try and open that series
4. See error

 - OS: OSX 10.15.7
 - Python version: 3.7
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

barentine, 2022-02-16:
If it wasn't obvious from that there is now also some leading separator silliness going on...
```
DEBUG:root:raw_requestline: b'GET //Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/ HTTP/1.1\r\n'
ERROR:root:127.0.0.1 - - [16/Feb/2022 12:56:36] code 404, message File not found - /Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/, [/Users/Andrew/PYMEData/Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/]

DEBUG:root:raw_requestline: b'GET /Users/Andrew/PYMEData/Andrew/2022_2_16/000/16_2_series_A.pcs/ HTTP/1.1\r\n'
```

barentine, 2022-02-16:
ClusterUI based `localize molecules` from the sidebar also failing to do anything on clusterofone (i.e. clicking Launch analysis with selected does nothing)
[image]

David-Baddeley, 2022-03-09:
Odd! I can't replicate ...

David-Baddeley, 2022-03-09:
For completeness - I tested on 

MacOS 12.1
Python 3.8 [aarm64]
numpy 1.23.0.dev0+477.gd410c8da4

If this persists, and is not down to something like a trivial (0x0) ROI or similar, I think that printing `out.shape` and `tosqueeze` on line 346 of BaseDataSource.py would be a good step in debugging.