python-microscopy/python-microscopy #1207: [ENHANCEMENT] photoelectrons/second PYMEImage hist option

barentine, 2022-03-02:
**Is your feature request related to a problem? Please describe.**
arguably better unit than ADU, of course, though I can see how it spirals quickly - maybe someone really wants to look at (photoelectrons/s)/(kW/cm^2) or whatever. Still, I think e-/s is straightforward if you've got basic metadata present, camera maps or not, and is something I want to semi-automate to save myself from manually typing my calc in the shell every time I go to slap an image into a presentation / do a calculation.

**Describe the solution you'd like**
drop down to select 'ADU', 'e-/s' in PYMEImage display settings. If e-/s is selected, could consider having a camera map entry box pre-populated from metadata like we do in LMAnalyser2 gui panel.

**Describe alternatives you've considered**
doing it in the shell, manually

**Are you prepared to work on it?**
absolutely, going to do it regardless, but would be great to do it in a pre-approved manner in case others would like it (like my labmates, who would like it).

David-Baddeley, 2022-03-07:
Can definitely see how it could snowball with metadata / camera map dependencies, but not fundamentally opposed

To clarify - do you just want to display a histogram of photo-electrons (would the need be served by, e.g. a menu function for "photoelectron histogram", or do you want to adjust scaling in photoelectrons etc ...?). 

The 3 ways I can see of addressing this would be ...

a) a menu function to display a histogram of image photoelectrons/s, leaving everything else in ADUs (this is probably the simplest, but least satisfactory way)
b) changing the display scaling histogram (I think this is what you are hinting at) with a drop down 
c) a "convert this image to photo-electrons/second" menu function which opens a new (potentially shallow - i.e. a datasource that re-scales the original data as you access it) copy of the image with the data re-scaled to photoelectrons/s

I think b) has the potential to get really messy quite quickly:
- changing the units in the display settings would not change the units in, e.g. line profiles, or any subsequent analysis (region detection etc) unless a lot of additional special case logic was added. This could lead to some pretty counter-intuitive user experiences.
- I can see a lot of people being interested in slightly different rescalings (e.g. photoelectrons/frame rather than /s which might be more relevant when looking at noise performance, even or ADUs/s), which means that you could end up with several different options and associated spaghetti

The more I think about it, the more strongly I'd lean towards c, as I think it is both most likely what people actually want (ie display and all subsequent analysis in photoelectrons/s), and the most robust solution (least spaghetti, as you only need to take care of the scaling in one place).

csoeller, 2022-03-07:
I think I have made an option akin to type c) quite a while ago in my (not very tidy) `PYME-extra` plugin collection, which has a menu entry under `Experimental>Map Tools` to  `Convert current frame to photo-electron counts` which will use maps to also offset-correct etc. See also [maptools.py](https://github.com/csoeller/PYME-extra/blob/master/PYMEcs/experimental/mapTools.py). Not necessarily exactly what @barentine is after, possibly also lacks robustness in implementation, but at least for our purposes this did everything we wanted to know. Obviously just for a frame in this case, but could be done differently if desired and IMHO exactly what the plugin system enables.

barentine, 2022-03-10:
Thanks both! @David-Baddeley is correct that the snazzy option b) was what I was after, but I can see the rational for c), which I think would be straightforward to do as a recipe module using a thin datasource and a menu item to call it and pop open a new PYMEImage window. Thanks for pointing out your photonconvert implementation, @csoeller !  David, if we do a datasource I can't help but think we'll want to log the units in the metadata, which I know is something we've talked about before. Any thoughts on what you'd want this recipe module to write? `'IntensityUnit: photoelectrons/s` for example?

What I was less clear on (dramatically so, given I used the term PYMEImage instead of PYMEAcquire), is that now that I'm in a scope debugging phase, I'd find a lot of display options useful 'on the fly' in PYMEAcquire, rather than having to acquire a series or snapshot first. This includes a photoelectrons/s histogram (could just be a second histogram displayed in a panel), a scalebar (had it for a minute, but I think we decided it was a fluke and removed it), fastGraph of a pixel in time (I implemented a ROI average version of that which is close enough most of the time), etc.