python-microscopy/python-microscopy #1213: [BUG] metadata caching between ImageStack instances

barentine, 2022-03-28:
**Describe the bug**
Opening two image stacks by filenames, acquired on different cameras. Metadata from the first changes after opening the second.

**To Reproduce**
```
(pyme) C:\Users\aesb>ipython
Python 3.7.11 (default, Jul 27 2021, 09:42:29) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.27.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from PYME.IO.image import ImageStack

In [2]: from os import path

In [3]: base_path = "censored"

In [4]: im1 = ImageStack(filename=path.join(base_path, '25_3_series_I.h5'))
filename == censored\25_3_series_I.h5
censored\25_3_series_I.h5r

In [5]: im1.mdh['Camera.ElectronsPerCount']
Out[5]: 10.0

In [6]: im2 = ImageStack(filename=path.join(base_path, '25_3_series_I_scmos.h5'))
filename ==censored\25_3_series_I_scmos.h5
censored\25_3_series_I_scmos.h5r

In [7]: im2.mdh['Camera.ElectronsPerCount']
Out[7]: 0.23

In [8]: im1.mdh['Camera.ElectronsPerCount']
Out[8]: 0.23
```

**Expected behavior**
absolutely not this. ImageStack instances should have independent metadata

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Win10
 - Python version: 3.7
 - numpy version: 1.19

barentine, 2022-03-29:
Issue might be that we use `PYME.Analysis.MetaData.TIRFDefault` to prefill h5 file metadata. Likely better practice to make a generator function for it.

barentine, 2022-03-29:
yep, that was it. Will PR shortly

David-Baddeley, 2022-03-29:
Should be fixed with #1214

closed 2022-03-29 by David-Baddeley