python-microscopy/python-microscopy #1215: [ENHANCEMENT] gain calibration / more flatfield map support

barentine, 2022-04-01:
**Is your feature request related to a problem? Please describe.**
Picked up a particularly junk camera (UI-3240 NIR - not one of the ones others have shown working for single-molecules. dark and variance maps are horrifically non-uniform compared to e.g. a UI-3270), looking rather unusable with just a dark-map calibration. Further, the company would not like to disclose their "proprietary" gain conversion, so I'm going to calibrate it using the sqrt(var)/mean relation, but didn't know if we already had code to fit that curve, or if I should just write my own. I see `PYME.DSView.modules.dark_and_flatfield.flat` for calculating a flatfield from a series, but not something to fill `Camera.ElectronsPerCount` from a couple series at different intensities. 

Further, I wonder if we want to extend support for flatmaps so they can get installed similarly to dark and var maps in `gen_sCMOS_maps.py`?

@csoeller how are you calibrating the gain on your IDS cameras? I see some notes in the comments about emailing the company, and I can see I'll want to account for the minimum hardware gain being non-unity.

**Describe the solution you'd like**
recipe module with a couple three `FileOrURI` inputs for series acquired at the same intensity

**Describe alternatives you've considered**
writing my own script

**Are you prepared to work on it?**
sure

David-Baddeley, 2022-04-06:
I feel that it's something we should have, although not 100% sure. Will have a little bit of a dig ... As far as I recall, there are a couple of caveats which make the data collection side a bit non-trivial, especially if you want to get per-pixel gains not overall gains.

- estimating the gain needs a very accurate estimate of offset, and read noise (i.e. the dark and read noise measurements need to be taken with a cover on the camera (or similar), not just room lights off.
- You need quite a large series (1000s of frames)
- a flatfield map and a per-pixel gain map are not exactly equivalent (the flatfield map includes dust on the sensor, apodisation etc ..., the per-pixel gain map doesn't). I'm of the opinion that you usually want a flat field rather than a per-pixel gain map (i.e. measure the response to uniform illumination, rather than each pixels mean vs var behaviour). This might change somewhat (you might even need both) when you have high read noise (for low read noise, anything between sample and sensor just looks like a straight scaling of sensor sensitivity - and hence detection probability / Poisson noise).
- The best way of getting the average gain (suitable for use in combination with a flatfield ....) is often to use quite a non-uniform image (e.g. a defocussed image of a slit / edge, as this lets you get away with fewer images, as well as sampling lots of different intensities so you can look at linearity as well.

Curiously I've had reasonably good experiences with the UI3240NIR, although mostly for focus locking.

David-Baddeley, 2022-04-06:
There is some pre-existing flatfield calculation logic in `DSView.modules.dark_and_flatfield`. This might be a good place to add, e.g. some gain estimation code.

I'm not 100% sure about the merits of putting something like this in a recipe module, as it should really be a once-per-camera operation, which might make automation a bit overkill. I'd be particularly wary about a recipe which needed too many `FileOrUri` inputs - they are OK if absolutely required (light-weight calibration data etc ...), but make the recipes they are in non-portable. For one-off calibrations like this, recipes might be overthinking it / adding complexity where it isn't really needed.

The other argument against recipe-based automation of maps is that we really should have some kind of interactive quality control on them (which is harder to do in a recipe context).

David-Baddeley, 2022-04-06:
as a follow up on that - for the gain calibration, we really should display the variance-vs mean graph and its associated fit which lets us get read noise and counts to adus  (in theory the intercept and slope of the graph) at least for an image with a large intensity range). Displaying the graph would, e.g. let us easily see if we had any non-linearity in sensor response, and if we had sampled enough different intensities to get gain and intercept accurately.

barentine, 2022-04-27:
One thing I'll mention having just notebooked that, is that if one tries to subtract off the read noise, i.e. their variance map which they made using `gen_sCMOS_maps.py`, they'll get a surprise if they forgot the variance map is in [e-^2] and therefore already has a potentially-bogus conversion applied. Of course, the simpler / more robust way is to not subtract off the read noise, since that doesn't scale with the conversion factor and therefore just modifies the intercept. 

Two observations:
- chicken & egg problem with gen_sCMOS_maps.py if you don't already have a conversion factor you trust from the manufacturer. You need a dark map to calibrate the gain, and you need a gain to generate a variance map in [e-^2]
- default noise properties in a camera class that gets used by wildly different cameras with different chip manufacturers etc is a recipe for silly mistakes like the one I made. I would like to remove the 'default' entry in ueye `baseProps` is that's OK with you @David-Baddeley and @csoeller. Calibrations from specific camera models seem fine, particularly with a warning that one should calibrate their camera.