python-microscopy/python-microscopy #1226: [PROPOSAL] add ADOffset to base Camera.GenStartMetadata

barentine, 2022-05-03:
**Is your feature request related to a problem? Please describe.**
Currently, ueye cameras don't get ADOffset into the metadata because there's no custom GenStartMetadata call, and we diligently skip adding the ADOffset entry there, presumably because of EMCCDs.

Solutions are of course to either hardcode
```
super().GenStartMetadata(mdh)
mdh.setEntry('Camera.ADOffset', self.noise_properties['ADOffset'])
```
in every camera class except for EMCCDs where someone wants to use dark frames at the start of each series, or we add a line to the camera base class method and have EMCCDs implement their own GenStartMetadata and not call the base class method.

Notably:
iXon skips the base class method
Zyla skips the base class method,
DCAM cameras use the base class method then add ADOffset
ueye only uses the base class method
PCO edge uses the base class method then adds ADOffset

**Describe the solution you'd like to implement**
grab the ADOffset by default in the base class, let iXon and other EMCCDs take care of themselves.

**Describe alternatives you've considered**
add 
```
super().GenStartMetadata(mdh)
mdh.setEntry('Camera.ADOffset', self.noise_properties['ADOffset'])
```
to the ueye class and write a note about it in documentation for adding your own camera so someone else doesn't bump into this later.

**Additional context**
Add any other context or screenshots about the feature request here.

**What potential side effects do the proposed changes have?**
- Is it new functionality, or a change to existing functionality?
- Will it change how users interact with the software?
- Will it change the dependency landscape?

David-Baddeley, 2022-05-04:
`ADOffset` is generally omitted from the base metadata as it is often not constant between acquisitions, and it is more reliable to include dark frames in your imaging protocol and estimate on based on these. This is particularly noticeable on EMCCD cameras, where `ADOffset` depends strongly on both EMGain and temperature, and tends to drift a bit during an imaging session as the sensor warms up, but might be expected to be seen on other cameras as well. The sCMOS cameras are a bit special, as they have pixels under a chrome mask (not exposed to light) which are used to perform an on-camera correction for the temperature dependence. This means that we can generally get away with specifying a constant `ADOffset` for these cameras (although having the dark calibration frames if possible is still advised). For machine vision cameras (such as the ueye) I would not expect the dark level to be particularly constant, so I'd opt for the dark frames here if possible.

barentine, 2022-05-04:
All fair, I just think it's annoying to e.g. get GUI popups in bakeshop 'ADOffset value fudged as 7 and possibly wrong' everytime I try to run a recipe, even if that recipe uses dark maps ;) 

For the ueye I'll likely then submit a PR to
```
super().GenStartMetadata(mdh)
mdh.setEntry('Camera.ADOffset', self.noise_properties['ADOffset'])
```
and leave the base class alone

David-Baddeley, 2022-05-04:
If you don't supply ADOffset, it is estimated and added in `PYME.Analysis.MetaData.fillInBlanks` - you can see the logic there - in the absence of protocol info on where to find dark frames it takes the median of the first 10 frames, hopes for the best, and gives you the warning. This is something which a) should be better documented and b) should probably just log a warning in bakeshop.

Without evidence that ADOffset is stable on the ueyes I'd be a bit wary about adding it to the metadata.

barentine, 2022-05-04:
Yep, and the estimate's not so bad if the laser's off on the first frames ;) 

All fair, just trying to make our code less annoying to me on a daily basis. If we implemented automatic darkmap grabbing in more places, that'd be great. In general I agree that if anyone wants to get really serious with localization etc they'll want to use a darkmap, but if I just want to have a +/- 15% errorbar on e-/s brightness from a frame I snapped I just want a metadata value that isn't crazy (or estimated from my image with lasers on)

Re: #1231 I actually think that would have killed me in this case. I calibrated my camera, I stuffed the ADOffset into my yaml, yet it doesn't end up in metadata - I run a recipe to convert to e-/s and I want a warning! Or I want it to use the metadata entry I put in.