python-microscopy/python-microscopy #1228: [PROPOSAL] don't write negative error in LatGaussFitFR (or other) fit results, give it a failed code

barentine, 2022-05-03:
**Is your feature request related to a problem? Please describe.**
I think it's reasonable to assume precision estimates are positive, but we have a hardcoded -5e3 in LatGaussFitFR for fits which have obviously failed

**Describe the solution you'd like to implement**
We could spike it high, e.g.
```
In [2]: np.finfo(np.float32).max
Out[2]: 3.4028235e+38
```
and then give the resultCode entry a failed code, e.g. 5

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

**What potential side effects do the proposed changes have?**
- Is it new functionality, or a change to existing functionality?
- Will it change how users interact with the software?
- Will it change the dependency landscape?

David-Baddeley, 2022-05-03:
Is there an example of where the negative errors cause a problem? 

As it stands, negative errors are a clear flag that something didn't work. A high positive error on the other hand would be easy to mistake for a legitimate high positive error. 

Although it doesn't seem to be that reliable in practice, `resultCode` should already give information on why the fit did not work - arbitrarily messing with that doesn't seem to be a good idea, as it could clobber the other info it already conveys. As far as I can tell from the minpack docs, `resultCode==5` should mean that the number of function calls exceeded the algorithms threshold (i.e. no convergence).

David-Baddeley, 2022-05-03:
So implicitly, `resultCode==5` suggest that you might want to improve on start parameter estimates.

barentine, 2022-05-09:
Only time negative errors cause a problem is if you're hacking and using your own filtering, rather than a `ResultsFilter` which would force you to pick a lower bound too. Bothered me briefly, but suspect others won't hit. Just trying to be in the habit of broadcasting when I bump into something in case we decide to shore it up. Will close.

closed 2022-05-09 by barentine