python-microscopy/python-microscopy #1229: [BUG] PSF extraction gives erroneous PSF at the end of the image stack

lukasfue, 2022-05-04:
**Describe the bug**
Using PSF extraction in PYME gives a PSF image stack that has 121 images. However, the last 5 or so images (117-121) look very wrong. In my particular case, I am extracting an astigmatic PSF. So the PSF starts stretched more in x than y and should end the opposite. Indeed, this is mostly the case until image 114. Then starting in image 115, it begins to stretch again in x and by image 121 it looks like a merge between the first image in the PSF file and image 114. Maybe it is just looping for some reason and this isn't a bug? I wasn't sure but poor localization results using the astigmatic localization algorithm made me more concerned about this. It's worth noting that I checked the raw data for my bead stacks and the weirdness I see in the extracted PSF is not present in the raw bead stacks.

**To Reproduce**
Steps to reproduce the behavior:
1. Image bead stacks (I did the extraction on two separate bead stacks. The first was 100 steps of 20 nm step size and the second was 200 steps of 20 nm step size. Both have this problem, but it seems worse for the first/shorter stack)
2. Open 'psfExtraction" module
3. Extract PSF using ROI Half Size "30,30,60" and default parameters for everything else
4. Observe weirdness of extracted PSF at the end.

**Expected behavior**
PSF extraction without the weird ends

**Screenshots**
Beginning of PSF (image 1/121)
[image]

Middle of PSF
[image]

Image 117/121
[image]

End (image 121/121)
[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Window 10
 - Python version: 3.7.9
 - numpy version: 1.19.1
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2022-05-04:
Hi Lucas,

Could you please send me a copy of the data in question. A couple of thoughts in the meantime ...

1) at 20nm axial steps you would need a raw stack of at least 150, if not 200 steps, as you are going to need to find a continuous region of 120 frames within the stack (i.e. the 100 frame stack should be guaranteed to fail).
2) it's really odd that you are getting anything from the 100 frame stack - it should give a hard error as it runs over the end of the data. I wonder if some of the newer data model stuff is letting it wrap.
3) this makes me also wonder if the 200 frame stack is somehow wrapping - it might be work looking where the box(es) are in the xz view and checking that they are reasonably far from both the top and bottom of the data.
4) the multiple bead images used for PSF reconstruction are aligned using Fourier domain shifts. There is a chance this could be causing some wrapping (particularly if there is a lot of variation in the bead z-positions), although it should only effect the first and/or last frames.

lukasfue, 2022-05-04:
Hi David,

Yes, I took both 100 and 200 step bead stacks. The effect is clearer in the 100 step data but I do still see it with the 200 step data. 

It looks like maybe he box is too small as you point out. 
[image]

Here is the bead stack I used with 200 steps.

I will email you the data, I guess github doesn't like .tif files.

lukasfue, 2022-05-04:
Sorry I did not mean to close this

David-Baddeley, 2022-05-04:
One more question - What NA lens are you using - the PSF does look fairly long in z

lukasfue, 2022-05-05:
It is a 63x 1.47 NA Objective.

barentine, 2023-06-07:
Just wanted to note we've seen something similar here with an extracted PSF last frame looking more like it should be at the beginning of the stack

David-Baddeley, 2023-06-09:
OK, I think I know the cause: 

- To align the PSFs with sub-pixel precision we use Fourier domain shifting
- Because FFTs assume circular symmetry, this can cause wrap-around
- The wrap-around should not exceed 0.5 a pixel (maybe 1 pixel worst case), and will typically only be significant if the axial extent of the stack is too small for the data (i.e. the PSF hasn't gone close to zero at the top and bottom).

As to a solution, there are two options:
- ignore (if it's a problem for subsequent analysis you really need a deeper PSF)
- extract 1-2 slices over size and crop after extraction (this might be the safest option)

closed 2023-06-09 by David-Baddeley

barentine, 2023-10-06:
This still appears to be an issue. I'm trying to extract the DH PSF bead stack hosted [here](https://srm.epfl.ch/DatasetPage?name=beads) and the last frame on the extracted PSF is obviously from the start.

Additionally, despite their simulated focal plane being in the center of this 151 slice stack, our expand Z option only gets you back to ~127 slices even if you only tag a single PSF. 

I wonder if we need a combobox to select _how_ we want the z ROI expanded? The line `dzh = np.min(im_shape[2] - pts[:,2]) -1` in `extractImages._expand_z` is what's chopping my Z size, because it finds the intensity center (for my single tag!) to be not in the exact center, and it doesn't want to expand my Z roi asymmetrically. I know it's a little dumbed down for what the PSF extraction module can do, but XY centering with the option to average multiple bead images is a relatively straightforward PSF extraction process for calibrating a 3D PSF for SMLM, and that work flow has a couple kinks at the moment. Am I missing a way to get that easily?

David-Baddeley, 2023-10-09:
Confirmed - suspect error is in the padding, we add twice as much as needed and don't remove it all (extracted PSF is around 4 frames longer than expected). Although still not sure why we get the wrap around.

David-Baddeley, 2023-10-09:
As to not getting the centroid right, I think there are 2 issues at play here:

1) - the axial extent of the PSF measurement is not large enough - double helix PSFs blow out a bit axially and you really need at least 3 and preferably ~5 times the axial FWHM to get an accurate axial centroid.

2) the axial step size (and as a result the default axial ROI size) is pretty small (the code is designed for 50nm axial steps). This means that we don't capture enough axial range to get a good centroid.

David-Baddeley, 2023-10-09:
I also think that there are some things with this PSF which could make the centroid calculation pathological - the decay on either side does not look even (and it doesn't go close to zero).

David-Baddeley, 2023-10-09:
OK, did find a minor bug where we were padding by twice as much as we remove, but the main error is in the size of the calibration data and ROI selection. The main issue is that the calibration data linked simply doesn't have enough z-range to capture enough of the PSF and accurately estimate z positions. The secondary part of this is that the default axial ROI size is simply too small for 10 nm steps.

David-Baddeley, 2023-10-09:
As to averaging without z alignment - I think that is only really worthwhile on simulated data - in a real microscope the focal plane is always curved enough to mean that axial alignment is necessary for a good PSF. And that's without any potential coverslip tilt ...

barentine, 2023-10-09:
Thanks for looking into the padding issue! 

Maybe we should change the TagPSF code to be aware of the expand Z ROI option? It's not intuitive that tweaking the z ROI size would have an effect on the extraction even with that box checked.

Generally folks try and avoid double-helix PSF stacks that starts to wrap and give degenerate angles - typically one would design/manufacture a different phase mask if they wanted to target a different axial extent. Obviously you can break the degeneracy other ways, and you can go ahead and capture a larger Z range than you plan on using later (e.g. the 4 um you suggest), but I would say it's not intuitive for people using a DH phase mask designed for 1.5 um to take a 4 um calibration stack, nor do I particularly see why it's necessary. I'm not pointing this out to be a pain, nor do I really know what the right answer is here, I just feel like we either need excellent PSF extraction documentation, for a variety of use-cases, or maybe we need to rework some things so it's more flexible / not wrong (re: wrapping) if a user isn't aware of some nuances in what might be a black-box to them. For the simulated PSF case here, I might just localize each frame, and average the fitted orientation on each frame to generate the Z look-up.

David-Baddeley, 2023-10-09:
A couple of quick thoughts ...

- The tag button uses the entire axial extent of the calibration stack for centroid calculation. It's just aligning the extracted ROIs that uses the reduced ROI range.

- The specific simulated case you linked now works (despite the warnings) as I increased the amount of padding to 4 pixels rather than 2 (technically, it was already padding by 4 pixels, but only removing 2 of them, it now pads by 4 and removes 4).

- In theory, the interpolated PSF fitting will break the double-helix degeneracy for you using other aspects of PSF shape, although getting a suitable starting position can be hard. I'd even go as far as to say that the degeneracy makes it even more  important to have a large axial extent in your DH PSF, otherwise you are going to get spurious localizations which are one rotation period out. If you use the interpolated PSF fitting and a suitable start position estimator (might need a new one) this should get taken care of.

-  The code is a little more forgiving than the warnings might imply (ie the warning that the stack should be 4 um doesn't kick in for stacks of 3um, and the ROI warning doesn't kick in until the ROI drops below 2um (rather than the recommended 3).

- I've generally erred a little on the side of being conservative with suggested sizes, partly because the PSF extraction is dual-use (i.e. both localization and deconvolution) and you are in for a world of pain if you use any less than the recommended sizes for deconvolution. Things like deconvolving out the bead shape from the extracted PSF (part of my recommended PSF workflow) and/or pupil extraction are also going to blow up if the size is any smaller than recommended. I made the warnings warnings (rather than errors) so that you could choose to disregard them if you had good reason to think that you would be OK with smaller sizes. 

- The code now raises a specific warning when the PSF is going to wrap. I'm in two minds as to whether this should be a warning or a hard error.

PR #1376 (2023-06-09): Pad to avoid wrapping in PSF extraction
Fixes #1229