python-microscopy/python-microscopy #1235: [PROPOSAL] auto-/cross-correction recipe module for image-based drift correction

barentine, 2022-05-09:
**Is your feature request related to a problem? Please describe.**
I have some non-localization data I need to drift correct. Intend to use autocorrelation as done in `localisations.AutocorrelationDriftCorrection`, but we don't have an interface to use directly with image data.

**Describe the solution you'd like to implement**
1. refactor the code in `localisations.AutocorrelationDriftCorrection` slightly so I can call it from another module to compute a shift on each frame of a series.
2. add a `measurement.AutocorrelationShiftCalculation` recipe module, outputing a tabular with per-frame shift estimates
4. add an `processing.ShiftRegistration` recipe module to apply the shifts to an input series, with the option for sub-pixel interpolation
5. add unit tests

**Describe alternatives you've considered**
already written it for myself, just with duplicated code / no refactoring. Alternative is I don't sub a PR.

**What potential side effects do the proposed changes have?**
- Is it new functionality, or a change to existing functionality?
new
- Will it change how users interact with the software?
not many
- Will it change the dependency landscape?
no

David-Baddeley, 2022-05-09:
Sounds like a good idea, with a couple of suggestions on structuring/naming.

- To me, "registration" is the process of determining the shifts, not applying them. I wonder about `processing.CorrectImageShifts` or similar instead of `ShiftRegistration`
- Do we need separate estimation and correction modules? I'm unclear as to the answer here - seperate modules allows future implementation/addition of alternative estimators, but this could also be accomplished by adding a `method` Enum to a `CorrectImageShifts` module. In terms of ease of use, one module is probably simpler. 
- We might need to be a bit careful about naming so we don't clash with chromatic shift correction. Maybe `CorrectImageDrift` rather than `CorrectImageShifts`

barentine, 2022-05-17:
Thanks for your thoughts on this @David-Baddeley ! 

Naming `ShiftRegistration` is fine by me. I do think separate modules would be nice - let's folks add their own alternatives in a plugin and still use the `CorrectImageDrift`. An alternative way to achieve that would be for one of the `method` Enum to be `direct` and you supply your own, but since the recipe inputs/outputs define API I don't know if you really want a loose input. Might be nicer as two modules. Use-case wise, I do think it's nice to have them separate/some flexibility. I wanted autocorrelation last week, and this week I want to use a fiducial localized on another camera (but for image drift correction not localization correction).