python-microscopy/python-microscopy #1236: [BUG] saving images from recipe datasources fails

csoeller, 2022-05-10:
**Describe the bug**
Make a recipe that creates new datasources and try to save an image from one of these, get error as below.

**To Reproduce**
Steps to reproduce the behavior:
1. Use, for example, the recipe below (just applies Gaussian filtering)
2. Try `Save as` of the created output image
5. See error

**Example recipe as YAML**
```yaml
- filters.GaussianFilter:
    inputName: input
    outputName: output
```

**Error Message**
```
Error whilst running File>&Save As
==============================================
python-microscopy=22.01.12.post0.dev[git]6a021906e625
python=3.7.6, platform=darwin
numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/ui/progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/DSView/dsviewer.py", line 312, in OnSave
    self.image.Save()
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/image.py", line 1335, in Save
    self.filename = dataExporter.ExportData(self.data_xyztc, self.mdh, self.events, filename=filename, progressCallback=progressCallback)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/dataExporter.py", line 504, in ExportData
    exp.Export(ds, filename, slice(0, ds.shape[0], 1), slice(0, ds.shape[1], 1), slice(0, ds.shape[2], 1),mdh, events, origName, progressCallback=progressCallback)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/dataExporter.py", line 281, in Export
    data, _, _, _, _, _, _, _ = self._prepare(data, xslice, yslice, zslice, tslice)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/dataExporter.py", line 84, in _prepare
    cropped = CropDataSource.DataSource(data, xrange=xslice, yrange=yslice, zrange=zslice, trange=tslice)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/DataSources/CropDataSource.py", line 100, in __init__
    self.set_dim_order_and_size(self.dataSource._input_order, szs[2], szs[3], szs[4])
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/dataWrap.py", line 72, in __getattr__
    return getattr(self.wrapList[0], name)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/dataWrap.py", line 72, in __getattr__
    return getattr(self.wrapList[0], name)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/dataWrap.py", line 72, in __getattr__
    return getattr(self.wrapList[0], name)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/DataSources/ArrayDataSource.py", line 69, in __getattr__
    return getattr(self.data, name)
AttributeError: 'numpy.ndarray' object has no attribute '_input_order'
```

**Expected behavior**
Should just work. For example, equivalent functionality via `Processing>Gaussian Filter` can be saved fine.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 see **Error Message** above

csoeller, 2022-05-11:
The same issue means that bakeshop generally fails.

I have come up with a patch that fixes the issue, it may or may not be the correct solution but pinpoints where the issue occurs and roughly how it could be fixed.

```diff
diff --git a/PYME/IO/DataSources/CropDataSource.py b/PYME/IO/DataSources/CropDataSource.py
index 621cb08f8..2acaff718 100644
--- a/PYME/IO/DataSources/CropDataSource.py
+++ b/PYME/IO/DataSources/CropDataSource.py
@@ -97,14 +97,14 @@ class DataSource(XYZTCDataSource):
         
         szs = [int(np.floor((r.stop-r.start)/r.step)) for r in [self.xslice, self.yslice, self.zslice, self.tslice]] + [self.dataSource.shape[4],]
 
-        self.set_dim_order_and_size(self.dataSource._input_order, szs[2], szs[3], szs[4])
+        self.set_dim_order_and_size(getattr(self.dataSource, '_input_order', None), szs[2], szs[3], szs[4])
         
         self._shape = tuple(szs)
         self._dtype = dataSource.dtype
         
-        self._i_z_stride = self.dataSource._z_stride
-        self._i_t_stride = self.dataSource._t_stride
-        self._i_c_stride = self.dataSource._c_stride
+        self._i_z_stride = getattr(self.dataSource, '_z_stride',self._z_stride)
+        self._i_t_stride = getattr(self.dataSource, '_t_stride',self._t_stride)
+        self._i_c_stride = getattr(self.dataSource, '_c_stride',self._c_stride)
 
 
     
@@ -202,4 +202,4 @@ def roi_crop_image(image, roi, z=True, t=False):
     xrange = roi[0]
     yrange = roi[1]
 
-    return crop_image(image, xrange, yrange, zrange, trange)
\ No newline at end of file
+    return crop_image(image, xrange, yrange, zrange, trange)
```

David-Baddeley, 2022-05-11:
With the proposed patch, the `getattr()` on `'_input_order'` should work, but the defaults on the strides could be really problematic (notably it will break badly if you are downsampling the datasource in the crop, which the CropDataSource permits, and input strides are different to output strides. It will probably also break badly if you are cropping as that will also make input and output strides different).

Another, and potentially better place to solve this would be to make sure that `ArrayDataSource` defines `_input_order` and the stride info.

csoeller, 2022-05-12:
Thanks for feedback. All I can say is that I take your word for it - I still find the 5D model confusing (or its implementation) and I am also a little bemused by the class hierarchy and wrapping that datawrap does. With a lot of work I could probably eventually come up with something, but not with much confidence.

I would appreciate some guidance how `ArrayDataSource` should best be tweaked to define `_input_order`, i.e. is copying the `set_dim_order_and_size` method from `XYZTCDataSource` fine? But then, should this be in `ArrayDataSource` or in `XYZTCDataSource`?

David-Baddeley, 2022-05-12:
yeah the implementation is a bit of a mess as it tries to maintain backwards compatibility. Will hopefully become cleaner once we no longer need to keep that.

Another solution to the problem you are seeing might be to ensure that recipe modules all coerce their outputs to an XYZTCDataSource or sub-class/wrapper, much as is done when loading data in dh5view. Alternatively that could be done in export (possibly safer). Will have a look once this grant is in.

csoeller, 2022-05-13:
Excellent, much appreciated, we can survive with our non-ideal fix until you find time to look at it. Good luck for the grant...

David-Baddeley, 2022-05-18:
should be fixed with #1241

csoeller, 2022-05-23:
Excellent, works now fine. Many thanks for fixing.

closed 2022-05-23 by csoeller