python-microscopy/python-microscopy #1237: [ENHANCEMENT] HTSMS simulator / demo

barentine, 2022-05-11:
**Is your feature request related to a problem? Please describe.**
We're trying to walk people through a semi-automated acquisition / chained analysis using the simulator

**Describe the solution you'd like**
overview on simulated cells, manually click, load from tile viewer, run protocol acq at each ROI with an associated rule chain to localize and then render a histogram image. 

Ideally this would work on the executable installer, but that raises several questions, such as whether we want to show clusterUI / need to include django in the packaged installer

Outstanding issues:
- [ ] simul642.py protocol has an OnBAnalyse call which annoys me since I'd rather just let it localize and look at after. For sure inconvenient if you set the rule chain to post on series start (cool feature) and then all of a sudden it's trying to open something (failed for me)
- [ ] with django 2.1, clusterUI 'Selected Files > Localize molecules' brings up the menu as it should, but clicking Launch takes you to the distributor page without actually posting any rule tasks.

David-Baddeley, 2022-05-11:
- Note that posting on series start has potential issues with incomplete/incorrect metadata (in practice this probably just means that we need to hard-code ADOffset in the camera metadata). This should not be too much of a problem in the simulator, and might already happen - will double check, but is not portable to all cameras and protocols.

barentine, 2022-05-11:
adding while we talk:
- [ ] make sure fakeCam adds ADOffset to metadata

barentine, 2022-05-18:
Running into an issue with launching localization on spool stop when spooling to h5
<details/>

```
Traceback (most recent call last):
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 56934)
  File "<string>", line 3, in raise_from
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 440, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\aesb\Miniconda3\lib\http\client.py", line 1369, in getresponse
    response.begin()
  File "C:\Users\aesb\Miniconda3\lib\http\client.py", line 310, in begin
    version, status, reason = self._read_status()
  File "C:\Users\aesb\Miniconda3\lib\http\client.py", line 279, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\aesb\Miniconda3\lib\site-packages\requests\adapters.py", line 450, in send
    timeout=timeout
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 756, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\util\retry.py", line 532, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\packages\six.py", line 769, in reraise
    raise value.with_traceback(tb)
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 706, in urlopen
    chunked=chunked,
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 445, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 440, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\aesb\Miniconda3\lib\http\client.py", line 1369, in getresponse
    response.begin()
  File "C:\Users\aesb\Miniconda3\lib\http\client.py", line 310, in begin
    version, status, reason = self._read_status()
Traceback (most recent call last):
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 706, in urlopen
    chunked=chunked,
Traceback (most recent call last):
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 706, in urlopen
    chunked=chunked,
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 445, in _make_request
    six.raise_from(e, None)
  File "C:\Users\aesb\Miniconda3\lib\http\client.py", line 279, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 445, in _make_request
    six.raise_from(e, None)
  File "C:\Users\aesb\Miniconda3\lib\socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "<string>", line 3, in raise_from
urllib3.exceptions.ProtocolError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
  File "<string>", line 3, in raise_from
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 440, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\aesb\Miniconda3\lib\site-packages\urllib3\connectionpool.py", line 440, in _make_request
    httplib_response = conn.getresponse()

During handling of the above exception, another exception occurred:

  File "C:\Users\aesb\Miniconda3\lib\socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\aesb\Miniconda3\lib\http\client.py", line 1369, in getresponse
    response.begin()
  File "C:\Users\aesb\Miniconda3\lib\http\client.py", line 1369, in getresponse
    response.begin()
[... 315 lines cut ...]
  File "c:\userfiles\andrew\code\python-microscopy\PYME\IO\h5File.py", line 122, in get_file
    return self.get_frame(frame_num)
----------------------------------------
  File "tables\hdf5extension.pyx", line 2155, in tables.hdf5extension.VLArray._read_array
  File "c:\userfiles\andrew\code\python-microscopy\PYME\IO\h5File.py", line 98, in get_frame
    data = self.image_data[idx]
ValueError: cannot set WRITEABLE flag to True of this array
  File "C:\Users\aesb\Miniconda3\lib\site-packages\tables\vlarray.py", line 681, in __getitem__
    return self.read(start, stop, step)[0]
----------------------------------------
  File "C:\Users\aesb\Miniconda3\lib\site-packages\tables\vlarray.py", line 821, in read
    listarr = self._read_array(start, stop, step)
  File "tables\hdf5extension.pyx", line 2155, in tables.hdf5extension.VLArray._read_array
ValueError: cannot set WRITEABLE flag to True of this array
----------------------------------------
  File "C:\Users\aesb\Miniconda3\lib\site-packages\tables\vlarray.py", line 681, in __getitem__
    return self.read(start, stop, step)[0]
  File "C:\Users\aesb\Miniconda3\lib\site-packages\tables\vlarray.py", line 821, in read
    listarr = self._read_array(start, stop, step)
  File "tables\hdf5extension.pyx", line 2155, in tables.hdf5extension.VLArray._read_array
ValueError: cannot set WRITEABLE flag to True of this array
----------------------------------------
```

</details>

barentine, 2022-05-18:
OK, @David-Baddeley , wx issue you might be faster at solving: if we change the fitfactory the v sizer appears not to necessarily update nicely. E.g. we start with:
[image]
and switching to AstigGaussFitFR we end up with:
[image]
and can no longer add the localization rule to the chain through the GUI

David-Baddeley, 2022-05-18:
Re the WRITABLE error above - I wonder if this is a pytables version error. Is also super weird as we are reading from the array, not writing to it.

barentine, 2022-09-28:
Regression in latest package due to #1271 and incomplete fix in #1272 

```
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\ExecTools.py", line 140, in _bginit
    self.codeObj(defLocals['scope'])
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Scripts\init.py", line 61, in cm
    pixel_size_nm=70.,
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\Simulator\fakeCam.py", line 386, in __init__
    self.SaturationThreshold = (2**16) - 1
AttributeError: can't set attribute
Exception in thread Thread-24:
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\ExecTools.py", line 149, in _bginit
    raise e
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\ExecTools.py", line 140, in _bginit
    self.codeObj(defLocals['scope'])
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Scripts\init.py", line 61, in cm
    pixel_size_nm=70.,
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\Simulator\fakeCam.py", line 386, in __init__
    self.SaturationThreshold = (2**16) - 1
AttributeError: can't set attribute

Exception ignored in: <function FakeCamera.__del__ at 0x0000011B860BA048>
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\Simulator\fakeCam.py", line 640, in __del__
    self.Shutdown()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\Simulator\fakeCam.py", line 512, in Shutdown
    self.compT.kill = True
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\Simulator\fakeCam.py", line 637, in __getattr__
    else:  raise AttributeError(name)  # <<< DON'T FORGET THIS LINE !!
AttributeError: compT
ERROR:PYME.Acquire.ExecTools:Error running background init task Lasers
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\ExecTools.py", line 140, in _bginit
    self.codeObj(defLocals['scope'])
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Scripts\init.py", line 143, in lasers
    scope.l488 = lasers.FakeLaser('l488',scope.cam,1, initPower=10)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\lasers.py", line 174, in __init__
    Laser.__init__(self,name,turnOn, **kwargs)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\lasers.py", line 40, in __init__
    self.TurnOff()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\lasers.py", line 185, in TurnOff
    self.cam.compT.laserPowers[self.chan] = 0
AttributeError: 'NoneType' object has no attribute 'compT'
Exception in thread Thread-25:
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\ExecTools.py", line 149, in _bginit
    raise e
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\ExecTools.py", line 140, in _bginit
    self.codeObj(defLocals['scope'])
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Scripts\init.py", line 143, in lasers
    scope.l488 = lasers.FakeLaser('l488',scope.cam,1, initPower=10)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\lasers.py", line 174, in __init__
    Laser.__init__(self,name,turnOn, **kwargs)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\lasers.py", line 40, in __init__
    self.TurnOff()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\lasers.py", line 185, in TurnOff
    self.cam.compT.laserPowers[self.chan] = 0
AttributeError: 'NoneType' object has no attribute 'compT'

DEBUG:PYME.Acquire.acquiremainframe:Backround initialization done
DEBUG:PYME.Acquire.acquiremainframe:Starting post-init
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\acquiremainframe.py", line 170, in _check_init_done
    self.doPostInit()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\acquiremainframe.py", line 232, in doPostInit
    if self.scope.cam.CamReady():# and ('chaninfo' in self.scope.__dict__)):
AttributeError: 'NoneType' object has no attribute 'CamReady'
```

barentine, 2022-10-03:
10/02 executable installer. PYMEAcquire -i init_sim_htsms.py :

```
ERROR:PYME.Acquire.acquiremainframe:Error whilst initializing Chained Analysis GUI
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\acquiremainframe.py", line 238, in doPostInit
    cm.run(self, self.scope)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\ExecTools.py", line 205, in run
    self.codeObj(parent, scope)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Scripts\init_sim_htsms.py", line 213, in chained_analysis
    from PYME.Acquire.htsms.rule_ui import SMLMChainedAnalysisPanel, get_rule_tile, RuleChain
ModuleNotFoundError: No module named 'PYME.Acquire.htsms'
```

barentine, 2022-10-03:
Same 10/02 executable installer:
```
(base) C:\Users\aesb>python -m PYME.tileviewer.tileviewer C:\Users\aesb\PYMEData\aesb\20221003_tile_demo
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\tileviewer\tileviewer.py", line 23, in <module>
    loader=jinja2.PackageLoader('PYME.tileviewer', 'templates'),
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\jinja2\loaders.py", line 320, in __init__
    f"The {package_name!r} package was not installed in a"
ValueError: The 'PYME.tileviewer' package was not installed in a way that PackageLoader understands.
```

barentine, 2022-10-03:
Worth noting for the last error, that the executable packaged jinja2                    3.0.3              pyhd3eb1b0_0

David-Baddeley, 2022-10-03:
Missing `PYME.Acquire.htsms` package really should have been picked up in previous tests (it's missing from PYME.Acquire.setup.py, which was appropriate when we assumed it would always be a source/development install).

David-Baddeley, 2022-10-03:
not sure on the tileviewer stuff - will look into that ...

David-Baddeley, 2022-10-03:
Jinga2=3.03 on my local machine too so unlikely to be the cause ... maybe a missing templates directory??

barentine, 2022-10-04:
We're looking pretty sharp on the simulator / htsms demo with the latest build!

I think we might want to tweak the logging slightly though. Every frame spooled is a bit verbose
```
ion: keep-alive\r\nContent-Length: 201201\r\n\r\n'
DEBUG:PYME.IO.cluster_streaming:b'PUT /__aggregate_h5/aesb/2022_10_4/4_10_series_H.h5/frame00081.pzf HTTP/1.1\r\nConnection: keep-alive\r\nContent-Length: 201361\r\n\r\n'
DEBUG:PYME.IO.cluster_streaming:b'PUT /__aggregate_h5/aesb/2022_10_4/4_10_series_H.h5/frame00082.pzf HTTP/1.1\r\nConnection: keep-alive\r\nContent-Length: 201364\r\n\r\n'
DEBUG:PYME.IO.cluster_streaming:b'PUT /__aggregate_h5/aesb/2022_10_4/4_10_series_H.h5/frame00083.pzf HTTP/1.1\r\nConnection: keep-alive\r\nContent-Length: 201197\r\n\r\n'
DEBUG:PYME.IO.cluster_streaming:b'PUT /__aggregate_h5/aesb/2022_10_4/4_10_series_H.h5/frame00084.pzf HTTP/1.1\r\nConnection: keep-alive\r\nContent-Length: 201397\r\n\r\n'
DEBUG:PYME.IO.cluster_streaming:b'PUT /__aggregate_h5/aesb/2022_10_4/4_10_series_H.h5/frame00085.pzf HTTP/1.1\r\nConnection: keep-alive\r\nContent-Length: 201060\r\n\r\n'
DEBUG:PYME.IO.cluster_streaming:b'PUT /__aggregate_h5/aesb/2022_10_4/4_10_series_H.h5/frame00086.pzf HTTP/1.1\r\nConnection: keep-alive\r\nContent-Length: 201039\r\n\r\n'
DEBUG:PYME.IO.cluster_streaming:b'PUT /__aggregate_h5/aesb/2022_10_4/4_10_series_H.h5/frame00087.pzf HTTP/1.1\r\nConnection: keep-alive\r\nContent-Length: 201430\r\n\r\n'
```

barentine, 2022-10-17:
Ah, I see the colour item was commented out in response to #1292 . Will workaround for now

barentine, 2022-10-17:
Note: if you clear an analysis chain the 'start on' dropdown will not update to 'off', even though the new chain will default to off (which is displayed correctly in the 'chained analysis' panel).

barentine, 2022-10-17:
Also note that typing in `50000` into x or y position panel and hitting 'Go' updates the value to `100000`. Probably need to look at minimum step size in the simulated stage

David-Baddeley, 2022-10-17:
Re position updates - are you sure you didn't enter `500000`. The simulated stages only have a  limited travel range (I think [0, 100000]), and the default behaviour when making an out-of-bounds move is to clip to the bounds.

David-Baddeley, 2022-10-17:
I should really look at #1292 again - the commenting out was a somewhat desperate and short-term gesture. Workaround is simply to specify in the recipe text, but we might be able to get a more satisfactory workaround to the segfault and hard crash by simply specifying a different editor in the view.