python-microscopy/python-microscopy #1238: [PROPOSAL] fail hard on PYMEClusterOfOne --clusterUI if django isn't installed

barentine, 2022-05-17:
**Is your feature request related to a problem? Please describe.**
```
PYMEClusterOfOne --clusterUI
Setting icon
INFO:PYME.cluster.cluster_of_one:Launching data server: root=C:\Users\aesb\PYMEData
INFO:PYME.cluster.cluster_of_one:Launching rule server
INFO:ruleserver:Starting ruleserver on 127.0.0.1:0
INFO:root:========================================
PYMEDataServer, running on python 3.7.9 (default, Aug 31 2020, 17:10:11) [MSC v.1916 64 bit (AMD64)]

INFO:root:Serving from directory: C:\Users\aesb\PYMEData
INFO:root:Serving HTTP on 127.0.0.1 port 55134 ...
INFO:PYME.cluster.cluster_of_one:Launching node server
ERROR:PYME.cluster.cluster_of_one:django is not installed, to use clusterUI install django (2.0.x, 2.1.x)
INFO:PYME.cluster.cluster_of_one:Launching clusterUI
dict_values(['http://127.0.0.1:55131/'])
DEBUG:nodeserver:Starting to poll for tasks
```

**Describe the solution you'd like to implement**
if we enter `cluster_of_one._launch_cluster_ui` without django installed, throw a hard error and shut things down. As is we continue and open a web page that we know won't work (even if the cluster itself is fine.

**Describe alternatives you've considered**
-if we don't have Django installed and run with the `--clusterUI` flag, log the error as we do know but direct to the ruleserver page
-not worry about it since we're now packaging with django

David-Baddeley, 2022-05-17:
I guess we could do a check for django and hard fail before trying to
launch, but I'm not sure it's really worth the effort. We already have a
fairly informative log output. There are a couple of additional failure
modes that we can't easily catch (wrong django version and something else
running on the port).

Tldr: I don't object, but not high priority.

barentine, 2022-10-04:
django is now included in pyme-depends

closed 2022-10-04 by barentine