python-microscopy/python-microscopy #1244: [ENHANCEMENT] Andor control frame grab EM gain automatically

d-perez-1, 2022-06-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when 
```
ERROR:PYME.Acquire.acquiremainframe:Error whilst initializing EMCCD Camera controls GUI
Traceback (most recent call last):
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\acquiremainframe.py", line 238, in doPostInit
    cm.run(self, self.scope)
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\ExecTools.py", line 205, in run
    self.codeObj(parent, scope)
  File "C:/UserFiles/code/davis-pyme-aquire/init.py", line 76, in cam_controls
    scope.camControls['EMCCD'] = AndorControlFrame.AndorPanel(MainFrame, scope.cameras['EMCCD'], 
scope)
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\Hardware\AndorIXon\AndorControlFrame.py", line 203, in __init__
    self._init_ctrls()
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\Hardware\AndorIXon\AndorControlFrame.py", line 163, in _init_ctrls
    self.tEMGain = wx.ComboBox(pan, -1, '0', choices=['0', '%d' % self.cam.DefaultEMGain], size=[30,-1], style=wx.CB_DROPDOWN|wx.TE_PROCESS_ENTER)
AttributeError: 'iXonCamera' object has no attribute 'DefaultEMGain'
```

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Set default gain in Andor iXon or call GetEMGain

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Put in init like UOA_2.py init

**Additional context**
Add any other context or screenshots about the feature request here.

**Are you prepared to work on it?**
It's great to have help getting new features implemented. If this is something you would like to work on, let us know.
no

d-perez-1, 2022-06-03:
would also be nice to allow cooler temperature on iXon to be set

David-Baddeley, 2022-06-03:
Unfortunately the Ixon has been through about 3 different iterations which treat the EMGain slightly differently (different scaling), so we can't specify a single hard-coded constant. That was why it was manually specified in the init file - e.g. `init_UOA_2`. In my experience, some cameras want about 30-35, some ~70 (note the gain scaling is non-linear in the gain parameter, so running with the wrong parameter will put the actual gain off by a factor of up to 100). There are definitely more elegant ways of setting scaling on a per-camera basis (e.g. putting it in the per-camera noise properties and tying to the camera serial number) but it's hard to know if it's worth the effort to code this. 

Either way, it needs to be better documented, and to fail cleanly when the property is absent. The defaults that I have used in the past are pretty ad-hoc and work out to setting the EMGain so that the effective read noise is about 1 electron. The rationale for the default EM gain is:

- high enough that the read noise is insignificant compared to Poisson shot noise in the imaging (with > 10 e- from background fluorescence in most PALM/STORM imaging this is nicely satisfied at around 1 e- effective read noise)
- low enough so as not to overly limit the dynamic range and to reduce the risk of gain ageing if you do get a bright signal

barentine, 2022-06-03:
Thanks @David-Baddeley . Will add one other iXon oddity that pressing 'enter' after setting temp would appear to start spooling