python-microscopy/python-microscopy #1248: [ENHANCEMENT] PYMEImage doc: linking other sections / programatic usage note

barentine, 2022-06-28:
**Is your feature request related to a problem? Please describe.**
Trying to get someone up to speed on using PYMEAcquire for their microscope - need to help them use PYMEImage then for their data analysis but also explain programmatic usage of PYME image data types, or alternatively at least explain how to readily convert their h5 to tif. Ideally the PYMEImage page would be a bit more fleshed out.

**Describe the solution you'd like**
- links to each of the example modules explicitly mentioned in the modules section (as a good start. We ought to at least link to the psf extraction how-to page at the moment)
[image]

- links / descriptions of `blob analysis` and `programmatic invocation`
[image]
- adding a table to the top of the [data formats](https://python-microscopy.org/doc/DataFormats.html#dataformats) page along the lines of `extension | name (link to section below if there's a paragraph about it) | required dependencies (e.g. bioformats or PYME)`. Obviously doesn't have to be complete in the sense of listing every bioformats extension, but a couple big ones up front would maybe be helpful for people. Could even list `dcimg` and `dbl` files might be helpful to list.
- call the program `PYMEImage` to avoid confusing folks with `dh5view` on the `PYMEImage` doc page (even if it works)
[image]
- update the example calls on the [data model](https://python-microscopy.org/doc/DataModel.html) page to include data_xytc and data_xyztc now that we've switched over. On my installs, I note that the integrated `Shell` blows up when I type image.data because the code generating the suggestions hits the deprecation warning (this gets printed directly into the shell such that you would have to delete it to run your line)
- add example code to load appropriate IO modules / do a simple operation on a datasource. Note the first part is what's missing from the datamodel page. 

**Are you prepared to work on it?**
prepared to do the quick ones that aren't super restructured text/sphinx-y

barentine, 2022-06-28:
For a small example on working with the PYME data model (Image data), I would suggest the following
```
from PYME.IO.image import ImageStack
import matplotlib.pyplot as plt

path = 'full-path-to-file.extension'
im = ImageStack(filename=path, haveGUI=False)
plt.imshow(im.data_xyztc[:, :, 0, 0, 0].squeeze())
```

barentine, 2022-06-28:
Would also be worth showing e.g. `print(im.mdh)` or `print(im.events)` in the above / additional example

David-Baddeley, 2022-06-28:
All good ideas ... need to think a bit about where `ImageStack` usage docs go .... in some ways at least part of this should be in the `ImageStack` docstring  rather than the data models page, although at present it's probably more useful under data models.

Example code of calling the IO modules directly is probably further down the priority list, as in most cases you don't need to  / shouldn't as `ImageStack` chooses an appropriate IO module for you. Manually instantiating IO modules definitely falls into the advanced usage/hacking end of things. When we do expand those docs, I expect that the docstring of the `PYME.IO.DataSources` module (i.e. adding a docstring to the top of `__init__.py` in this directory) is probably the right place.

David-Baddeley, 2022-06-28:
For completeness with the ImageStack stuff, we should probably mention creating an ImageStack from an array, and passing in metadata e.g. Access data from an image stack, do a filter on it, return a new ImageStack from the results.

zacsimile, 2022-07-01:
I mentioned to @David-Baddeley on a call yesterday, I think a useful tool for "Trying to get someone up to speed on using PYMEAcquire for their microscope" would be a "PYMEImage Quick Start," which quickly teaches the reader how to build an init script for a synthetic microscope and launch the scope. The "Further reading" section could take them to documentation on the simulator and how to spin up a cluster of one to analyze simulated data.