python-microscopy/python-microscopy #1250: [SUPPORT] HTTP 404 error when trying to added pyme python environment

lukasfue, 2022-07-01:
I'm just trying to install a development install of PYME on a PC. I uninstalled all anaconda instances and installed miniconda 3.7. I have the david_baddeley channel added. However when I enter `conda create -n pyme pyme-depends python=3.7` it gives a HTTP 404 error before finishing.

CondaHTTPError: HTTP 404 NOT FOUND for url <https://conda.anaconda.org/david_baddeley/win-64/win-64-fftw-3.3.4-0.tar.bz2>
Elapsed: 00:00.054994
CF-RAY: 7241da5c7e5bec00-BOS

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.

(base) C:\Users\Light-saver>conda activate pyme
Could not find conda environment: pyme
You can list all discoverable environments with `conda info --envs`.

(base) C:\Users\Light-saver>conda info --envs
# conda environments:
#
base                  *  C:\Users\Light-saver\miniconda3

I've done this exact process several times, but I've never ran into this problem before.

David-Baddeley, 2022-07-03:
Hi Lucas,

I suspect this is an anaconda error rather than anything on our end ... Hopefully just a temporary glitch with their server (a 404 error could mean a lot of things, but generally it will be a server problem) ... Have you tried again?

zacsimile, 2022-07-05:
I also saw this when installing on my new work laptop. I think I solved it by going to py38.

barentine, 2022-07-07:
https://conda.anaconda.org/david_baddeley/win-64/win-64-fftw-3.3.4-0.tar.bz2 does not exist. You can navigate to your packages page @David-Baddeley and try for yourself. Oddly https://anaconda.org/david_baddeley/fftw/3.3.4/download/win-64/win-64-fftw-3.3.4-0.tar.bz2 does exist, but something is broken - you should be redirected to download if you try and open the first url that everyone's failing on. For example, try https://conda.anaconda.org/david_baddeley/win-64/dispatch-1.0-py37_0.tar.bz2 .

Maybe try rebuilding, David, incrementing build number from 0 to 1, and re upload? 

@zacsimile I assume you built locally for py38? Doesn't look like we're hosting packages yet.

zacsimile, 2022-07-07:
I did. But yes, https://anaconda.org/david_baddeley/fftw/3.3.4/download/win-64/win-64-fftw-3.3.4-0.tar.bz2 works too.

ppa007hot, 2022-07-09:
I have a problem installing PYME 22.05.25 version on my computer. I have tried 2 computers (one is win10 and one is Win11) and both gave the error below when using the .exe installation.

[image]

Then I tried miniconda 
`conda create -n pyme python=3.7 pyme-depends python-microscopy`
it was stopped when downloading fftw from an HTTP error. I manually checked the url [https://conda.anaconda.org/david_baddeley/win-64/win-64-fftw-3.3.4-0.tar.bz2,] it was no longer working.
I even tried conda install -c david_baddeley fftw, but it gave the same error.

barentine, 2022-07-09:
Hi @ppa007hot . As a temporary workaround for the miniconda approach, please create an environment
`conda create -n pyme python=3.7`
then use this link https://anaconda.org/david_baddeley/fftw/3.3.4/download/win-64/win-64-fftw-3.3.4-0.tar.bz2 to download fftw.
Activate the pyme environment
`conda activate pyme`
and then run `conda install /package-path/package-filename.tar.bz2` to install fftw (replacing the path / package name). 
Finally, run  
`conda install pyme-depends python-microscopy`

Sorry for the extra trouble.

@David-Baddeley any luck with re-uploading a build-1 fftw?

David-Baddeley, 2022-07-09:
Looking at our package build recipes, we don't seem to have one for fftw (just pyfftw). I'm pretty sure anaconda used to supply this rather than the `david_baddeley` channel. Doesn't quite explain everything (i.e. why we would seem to have a version of fftw on anaconda), but might explain why things suddenly stopped working (removal from the anaconda channel). A bit torn on what the correct long term solution is ... (is it time to bump everyone to 3.8?) as maintaining stuff against a moving backdrop of anaconda packages is a bit of a nightmare.

ppa007hot, 2022-07-09:
Hi Andrew, thank you for the detailed instructions! However the work-around didn't work for me. I follow the exact steps and when finally install python-microscopy, it still wants to download fftw. 
It looks like fftw was not properly installed even using downloaded local file?

`(pyme) C:\Users\ppa00>conda list
WARNING conda.core.prefix_data:_load_single_record(191): Ignoring malformed prefix record at: C:\Users\ppa00\anaconda3\envs\pyme\conda-meta\win-64-fftw-3.3.4-0.json
# packages in environment at C:\Users\ppa00\anaconda3\envs\pyme:
#
# Name                    Version                   Build  Channel
ca-certificates           2022.4.26            haa95532_0
certifi                   2022.6.15        py37haa95532_0
openssl                   1.1.1p               h2bbff1b_0
pip                       21.2.4           py37haa95532_0
python                    3.7.13               h6244533_0
setuptools                61.2.0           py37haa95532_0
sqlite                    3.38.5               h2bbff1b_0
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
wheel                     0.37.1             pyhd3eb1b0_0
wincertstore              0.2              py37haa95532_2`

David-Baddeley, 2022-07-10:
fftw is available in the anaconda "main" channel, at a higher version than the one we have/had. Due to the higher version number It should take precedence and get installed from there instead of python-microscopy. To resolve this, I think we need to:

a) work out why conda is not installing the version from the main channel
b) potentially delete our copy.

As an interim workaround / to help with diagnosis, try specifically telling conda to install from the main channel

```
conda install -c main fftw
```

If that comes up with package conflicts, it might help us with the diagnosis. Unfortunately I don't have easy access to a windows machine at present to test.

ppa007hot, 2022-07-11:
Hi, I just want to update that I tried `conda install -c menpo fftw` to install the exact version 3.3.4, then installed pyme-depends, and finally built a beta version of python-microscopy. It is working now.

csoeller, 2022-07-11:
Hi all, we have also hit the issue here over the weekend. A PhD student tried to install on their home PC/laptop with Win 11 and hit the 404 issue for fftw.

What is the recommended set of workaround/tests to run now? I am little confused by the various messages.

I imagine I can test it on an existing windows PC by just creating a new environment for checking things out, is that right?

David-Baddeley, 2022-07-11:
In terms of debugging, what we ulitmately need to do is work out why the fftw from the main channel is not being installed ... 

maybe the output of `conda create -n pyme37_fftw python=3.7 python-microscopy=22.05.25 fftw=3.3.9` would be useful (which should hopefully tell us where the package conflicts are).

As a workaround, installing fftw=3.3.4 from either the menpo or conda-forge channels should work.

David-Baddeley, 2022-07-11:
With any luck I can test this on a windows machine later today.

David-Baddeley, 2022-07-11:
OK, that starts to make some sense (in a rather round-about way). The conda package for fftw has a spurious compiler pin (this is important for python extensions, but not for standard dlls that get accessed via ctypes such as fftw).

The good news is that this will go away once we move to 3.8. In the meantime, I'll find and re-upload the fftw package to the anaconda channel (probably just copy the menpo one).

David-Baddeley, 2022-07-11:
Have re-uploaded fftw to the david_baddeley channel, fingers crossed it now works.

Need to look at the installer now.

zacsimile, 2022-07-11:
```
(base) C:\>conda create -n pyme python=3.7 pyme-depends
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: |
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package python conflicts for:
pyme-depends -> python[version='2.7.*|>=2.7,<2.8.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0']
pyme-depends -> cherrypy -> python[version='>=3.10,<3.11.0a0|>=3.9,<3.10.0a0|>=3.8,<3.9.0a0|>=3.5,<3.6.0a0|>=3.6|>=3.8|>=3.7|>=3.5|>=3.7.1,<3.8.0a0']
python=3.7
Note that strict channel priority may have removed packages required for satisfiability.
````

David-Baddeley, 2022-07-11:
super weird as cherrypy looks like it should be solvable ...

zacsimile, 2022-07-11:
I agree. Took a different approach and ran `conda create -n pyme python=3.7` and then `conda install pyme-depends` and got

```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: -
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package certifi conflicts for:
certifi
setuptools -> certifi[version='>=2016.09|>=2016.9.26']
pip -> setuptools -> certifi[version='>=2016.09|>=2016.9.26|>=2017.4.17']
pyme-depends -> requests -> certifi[version='>=2017.4.17']
wheel -> setuptools -> certifi[version='>=2016.09|>=2016.9.26']

Package tk conflicts for:
certifi -> python[version='>=3.10,<3.11.0a0'] -> tk[version='>=8.6.11,<8.7.0a0']
wheel -> python -> tk[version='>=8.6.11,<8.7.0a0']
pyme-depends -> pillow -> tk[version='>=8.6.10,<8.7.0a0|>=8.6.8,<8.7.0a0|>=8.6.7,<8.7.0a0']
pip -> python[version='>=3.10,<3.11.0a0'] -> tk[version='>=8.6.11,<8.7.0a0']
setuptools -> python[version='>=3.10,<3.11.0a0'] -> tk[version='>=8.6.11,<8.7.0a0']
wincertstore -> python[version='>=3.10,<3.11.0a0'] -> tk[version='>=8.6.11,<8.7.0a0']

Package vs2015_runtime conflicts for:
openssl -> vc[version='>=14.1,<15.0a0'] -> vs2015_runtime[version='>=14.0.25123,<15.0a0|>=14.0.25420|>=14.15.26706|>=14.27.29016|>=14.16.27012']
pip -> python[version='>=3.8,<3.9.0a0'] -> vs2015_runtime[version='>=14.16.27012,<15.0a0']
python=3.7 -> vc[version='>=14.1,<15.0a0'] -> vs2015_runtime[version='>=14.0.25123,<15.0a0|>=14.0.25420|>=14.15.26706|>=14.27.29016|>=14.16.27012']
wincertstore -> vc[version='>=14.1,<15.0a0'] -> vs2015_runtime[version='>=14.15.26706|>=14.27.29016|>=14.16.27012']
sqlite -> vs2015_runtime[version='>=14.16.27012,<15.0a0']
pyme-depends -> blosc[version='<=1.19'] -> vs2015_runtime[version='>=14.16.27012,<15.0a0|>=14.27.29016,<15.0a0']
vs2015_runtime
python=3.7 -> vs2015_runtime[version='>=14.16.27012,<15.0a0']
wheel -> python -> vs2015_runtime[version='>=14.16.27012,<15.0a0']
setuptools -> python[version='>=3.10,<3.11.0a0'] -> vs2015_runtime[version='>=14.16.27012,<15.0a0']
sqlite -> vc[version='>=14.1,<15.0a0'] -> vs2015_runtime[version='>=14.0.25123,<15.0a0|>=14.0.25420|>=14.15.26706|>=14.27.29016|>=14.16.27012']
vc -> vs2015_runtime[version='>=14.0.25123,<15.0a0|>=14.0.25420|>=14.15.26706|>=14.27.29016|>=14.16.27012']
openssl -> vs2015_runtime[version='>=14.16.27012,<15.0a0']
wincertstore -> vs2015_runtime[version='>=14.16.27012,<15.0a0']

Package vs2008_runtime conflicts for:
pip -> python[version='>=2.7,<2.8.0a0'] -> vs2008_runtime
vc -> vs2008_runtime[version='>=9.0.30729.1,<10.0a0']
setuptools -> python[version='>=2.7,<2.8.0a0'] -> vs2008_runtime
wincertstore -> python[version='>=2.7,<2.8.0a0'] -> vs2008_runtime
sqlite -> vc=9 -> vs2008_runtime[version='>=9.0.30729.1,<10.0a0']
pyme-depends -> python[version='>=2.7,<2.8.0a0'] -> vs2008_runtime
certifi -> python -> vs2008_runtime
openssl -> vc=9 -> vs2008_runtime[version='>=9.0.30729.1,<10.0a0']
wheel -> python -> vs2008_runtime

Package charset-normalizer conflicts for:
pip -> requests -> charset-normalizer[version='>=2.0.0,<2.1.0']
pyme-depends -> requests -> charset-normalizer[version='>=2.0.0,<2.1.0']

Package cachecontrol conflicts for:
pip -> cachecontrol
python=3.7 -> pip -> cachecontrol

Package colorama conflicts for:
python=3.7 -> pip -> colorama
pyme-depends -> sphinx -> colorama[version='>=0.3.5']
[... 60 lines cut ...]
certifi -> python[version='>=3.10,<3.11.0a0'] -> xz[version='>=5.2.5,<6.0a0']
setuptools -> python[version='>=3.10,<3.11.0a0'] -> xz[version='>=5.2.5,<6.0a0']
pip -> python[version='>=3.10,<3.11.0a0'] -> xz[version='>=5.2.5,<6.0a0']
wincertstore -> python[version='>=3.10,<3.11.0a0'] -> xz[version='>=5.2.5,<6.0a0']

Package wincertstore conflicts for:
wheel -> setuptools -> wincertstore[version='>=0.2']
wincertstore
setuptools -> wincertstore[version='>=0.2']
pip -> setuptools -> wincertstore[version='>=0.2']

Package vc conflicts for:
pyme-depends -> blosc[version='<=1.19'] -> vc[version='14.*|9.*|>=14.1,<15.0a0|>=14.2,<15.0a0']
python=3.7 -> openssl[version='>=1.1.1e,<1.1.2a'] -> vc=9
wheel -> python -> vc[version='14.*|>=14.1,<15.0a0|9.*']
pip -> python[version='>=3.8,<3.9.0a0'] -> vc[version='14.*|>=14.1,<15.0a0|9.*']
sqlite -> vc[version='14.*|9.*|>=14.1,<15.0a0']
wincertstore -> python[version='>=3.7,<3.8.0a0'] -> vc[version='14.*|9.*']
python=3.7 -> vc[version='14.*|>=14.1,<15.0a0']
setuptools -> python[version='>=3.10,<3.11.0a0'] -> vc[version='14.*|>=14.1,<15.0a0|9.*']
vc
wincertstore -> vc[version='>=14.1,<15.0a0']
openssl -> vc[version='14.*|9.*|>=14.1,<15.0a0']
Note that strict channel priority may have removed packages required for satisfiability.
```

zacsimile, 2022-07-11:
Starting to think it's a problem with this machine...

David-Baddeley, 2022-07-11:
Confirmed - I got a clean resolution and install with 

```
conda create -n pm37_test python=3.7 python-microscopy
```

David-Baddeley, 2022-07-11:
(on win10)

csoeller, 2022-07-12:
Same here (clean install) with

`conda create -n pm37_test python=3.7 pyme-depends`

This is win 10, we will test win 11 later this afternoon.

csoeller, 2022-07-12:
win 11 seems also fine with

    conda create -n pm37_test python=3.7 pyme-depends

We will do the full development install on win 11 tomorrow.

David-Baddeley, 2022-10-11:
Best option for 3.8 at present is conda-forge.

Will trigger a rebuild.

closed 2022-11-28 by David-Baddeley