python-microscopy/python-microscopy #1251: [SUPPORT] Fiducial correction

lukasfue, 2022-07-08:
I have a 3D dataset (4Pi microscopy data) that consists of two separate files. Each file is for a different color channel that was imaged sequentially. I'm trying to align these channels using fiducials and I am wondering if there's currently any way to do so with PYME given the format of my data. Both files consist of single-molecule localizations and localizations from the fiducials. So, they fiducials are not separate from the single-molecule localizations nor was this data analyzed in PYME (which are the two ways PYME prefers to handle fiducials according to the documentation). So, is there any way for me to align these two datasets using the fiducials? I saw the 'manual selection' option for fiducials, but when I try to do that it gives me the following error:

 Error whilst running Corrections>Fiducials>Manual selection
==============================================
python-microscopy=22.05.25.post0.dev[git]
python=3.7.9, platform=win32
numpy=1.19.1, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\users\laf62\code\python-microscopy\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "c:\users\laf62\code\python-microscopy\PYME\LMVis\Extras\fiducial.py", line 228, in <lambda>
    visFr.AddMenuItem('Corrections>Fiducials', 'Manual selection', lambda e: manual_selection(visFr.pipeline))
  File "c:\users\laf62\code\python-microscopy\PYME\LMVis\Extras\fiducial.py", line 79, in manual_selection
    fids = recipe.namespace['clumped_fiducials']
KeyError: 'clumped_fiducials'

I appreciate the help!

barentine, 2022-07-08:
Looks like Manual Selection could stand to be named better - I'm not entirely sure what its use case is. 

Someone will have a better answer for this, but if you're trying to get this done today @lukasfue: I assume the fiducials are brighter than your localizations? And you have the same fiducials in both color series? In a recipe (run `bakeshop`) you can first concatenate your localizations with the tablefilters.ConcatenateTables recipe module. Then if you like, in PYMEVis or wherever you can filter them to just get your fiducials and save it, and just get your localizations and save it. Then if you open your localizations in PYMEVis, followed by `Corrections > Fiducials > Load fiducial fits from 2nd file` you should be OK, right? 

To get them to show up as different colors you'll want to add a mapping. The concatenatetables recipe module will add the key `concatSource` which will be zero for the 0th input and ones for the 1st input. So after drift correcting from your 2nd file fits, just add a mapping filter. Something like
```
- tablefilters.Mapping:
    inputName: drift_corrected
    mappings:
      probe: concatSource
    outputName: c_mapped
```
Worst case you need to add a process color call after that
```
- localisations.ProcessColour:
    input: c_mapped
    output: colour_mapped
```

then you should have ~full color functionality on your drift corr'd data

David-Baddeley, 2022-07-08:
Hi Lukas,

"Manual Selection" lets you manually decide which fiducials to keep after you have loaded them and done some preliminary processing. It's generally only needed in some edge cases, as the automatic fiducial selection generally works well. In that context, the name isn't awful, but we should add some documentation for your case which I suspect is reasonably common. 

Manual selection is described quite well in the drift correction blurb from "Messy PYME Docs" which is significantly better than what is in the HTML docs, if slightly out of date (some menu items have moved). I've been meaning to update and reformat into .rst for ages.

[Drift correction using fiducials in PYME.pdf](https://github.com/python-microscopy/python-microscopy/files/9074791/Drift.correction.using.fiducials.in.PYME.pdf)

I'll split my commentary on your case into 2 sections:

a) fiducial correction when both blinking events and fiducials are localised together (single colour)
b) sequential multi-colour

## a) Fiducial correction when both blinking events and fiducials are localised together (single colour)

This is accomplished (somewhat counter-intuitively) by selecting `Corrections->Fiducials->Load fiducial fits from 2nd file` and selecting the same file again. The caveat with doing it this way is that you are relying on the filters to determine what is a localisation and what is a fiducial, and that is not always trivial (will likely require some tweaking). The default filters for fiducial and localisation data sets are such that you should have a bit of a head start here if you used largish beads for your fiducials (the default filters are set up to discriminate on size). If you used, e.g. nanogold or very small beads it will be quite a lot harder, and you might need to try and discriminate based on intensity.

I'd try getting this to work on a single colour channel before attempting the multi-colour stuff

## b) Sequential multi colour

As Andrew mentioned, the best approach here is probably to concatenate the data first using a recipe, and to use a mapping to set the `probe` column to `concatSource`. In addition to Andrew's suggestion, **you will need to offset all the time indices in the second channel so that they start immediately after the end of those from the first channel**. This can be accomplished with a mapping filter on the second channel prior to running it into `ConcatenateTables`, which maps `t` to `t + max(chan_0_t) + 1`. Unfortunately you will currently need to find  `max(chan_0_t)` manually and hard code it for each series you concatenate (unless they are of a constant length).

Once you have done this, the fiducial code is not probe-aware, so should happily track the fiducials from one acquisition to the other. 

I hope this helps,
David

David-Baddeley, 2022-07-08:
**addendum:** you should not need `ProcessColour` in your concatenation recipe - that should happen automatically when the concatenated data is loaded.

lukasfue, 2022-07-08:
Thank you both for your comments!

@barentine okay that seems reasonable. I was trying to do as you instructed and got this error from `Corrections>Fiducials>Correct`:

 Error whilst running Corrections>Fiducials>Correct
==============================================
python-microscopy=22.05.25.post0.dev[git]
python=3.7.9, platform=win32
numpy=1.19.1, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\users\laf62\code\python-microscopy\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "c:\users\laf62\code\python-microscopy\PYME\LMVis\Extras\fiducial.py", line 226, in <lambda>
    visFr.AddMenuItem('Corrections>Fiducials', 'Correct', lambda e : correct())
  File "c:\users\laf62\code\python-microscopy\PYME\LMVis\Extras\fiducial.py", line 223, in correct
    drift_correct(visFr.pipeline)
  File "c:\users\laf62\code\python-microscopy\PYME\LMVis\Extras\fiducial.py", line 55, in drift_correct
    recipe.add_modules_and_execute([filt_fiducials, dbs])
  File "c:\users\laf62\code\python-microscopy\PYME\recipes\recipe.py", line 457, in add_modules_and_execute
    self.execute()
  File "c:\users\laf62\code\python-microscopy\PYME\recipes\recipe.py", line 203, in execute
    m.execute(self.namespace)
  File "c:\users\laf62\code\python-microscopy\PYME\recipes\localisations.py", line 446, in execute
    eps=self.searchRadius, min_samples=self.minClumpSize)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\sklearn\cluster\_dbscan.py", line 160, in dbscan
    est.fit(X, sample_weight=sample_weight)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\sklearn\cluster\_dbscan.py", line 375, in fit
    neighborhoods = neighbors_model.radius_neighbors(X, return_distance=False)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\sklearn\neighbors\_base.py", line 1115, in radius_neighbors
    for s in gen_even_slices(X.shape[0], n_jobs)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\joblib\parallel.py", line 1043, in __call__
    if self.dispatch_one_batch(iterator):
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\joblib\parallel.py", line 861, in dispatch_one_batch
    self._dispatch(tasks)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\joblib\parallel.py", line 779, in _dispatch
    job = self._backend.apply_async(batch, callback=cb)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\joblib\_parallel_backends.py", line 208, in apply_async
    result = ImmediateResult(func)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\joblib\_parallel_backends.py", line 572, in __init__
    self.results = batch()
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\joblib\parallel.py", line 263, in __call__
    for func, args, kwargs in self.items]
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\joblib\parallel.py", line 263, in <listcomp>
    for func, args, kwargs in self.items]
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\sklearn\utils\fixes.py", line 211, in __call__
    return self.function(*args, **kwargs)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\sklearn\neighbors\_base.py", line 910, in _tree_query_radius_parallel_helper
    return tree.query_radius(*args, **kwargs)
  File "sklearn\neighbors\_binary_tree.pxi", line 1483, in sklearn.neighbors._kd_tree.BinaryTree.query_radius
  File "sklearn\neighbors\_binary_tree.pxi", line 1439, in sklearn.neighbors._kd_tree.BinaryTree.query_radius
MemoryError

Is this just telling me I don't have enough RAM to do this? I have 32 GB of RAM.

Also, is this for drift correction or channel registration or both? I really only need channel registration, but if it also does some drift correction, that's fine. If I combine the data into one data set for both the single molecule data and the fiducials, how will it know to register them to each other?

barentine, 2022-07-08:
Yes, you likely ran out of RAM. You could check in task manager while running if you like. How many localizations are you working with? 

Without knowing more of what you're doing (what beads, what detection channels, etc.) its hard to tell you whether you'll accomplish drift corr or channel reg at the same time. My assumption is that you're working with the same fiducials visible in both color acquisitions, in which case this will account for bulk (mean) shift (but not rotation/distortion) between each frame in your series (regardless of color). What I'm trying to say is that you'll do really basic channel registration at the same time, which is usually not great but I think you might have the advantage of working with one detection channel given that you're imaging sequential...

lukasfue, 2022-07-08:
It's about 2.7 million localizations.

Oh that makes sense. Yes, the fiducials are gold nanorodz that show up in both color acquisitions which are recorded with one detection channel. The only difference is one filter and obviously not at the same time. Not sure where to go from here since this is pretty typical size for my datasets.

David-Baddeley, 2022-07-08:
A few options,

-open the data up and try to filter for fiducials heavily first so that the
second, fiducials copy has way fewer localisation.
- you could use FindClumps for this (scales way better than dbscan) and
filter on a minimum clump size to eliminate all the short traces.
- doing more than a mean shift correction should be possible with a little
extra work, but not a simple combination of existing recipe modules