python-microscopy/python-microscopy #1252: [BUG] unusable PYMEVis gui if default fit factory filters too strict

barentine, 2022-07-08:
**Describe the bug**

```
 Error whilst running File>&Open
==============================================
python-microscopy=22.05.25.post0.dev[git]
python=3.7.9, platform=win32
numpy=1.19.1, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\userfiles\andrew\code\python-microscopy\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\LMVis\visCore.py", line 399, in OnOpenFile
    self.OpenFile(filename)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\LMVis\visCore.py", line 772, in OpenFile
    self._createNewTabs()
  File "c:\userfiles\andrew\code\python-microscopy\PYME\LMVis\VisGUI.py", line 345, in _createNewTabs
    et = 1.1*self.pipeline.selectedDataSource['tIndex'].max()*self.pipeline.mdh['Camera.CycleTime']
  File "C:\Users\aesb\Miniconda3\lib\site-packages\numpy\core\_methods.py", line 39, in _amax
    return umr_maximum(a, axis, None, out, keepdims, initial, where)
ValueError: zero-size array to reduction operation maximum which has no identity

```

**To Reproduce**
Steps to reproduce the behavior:
1. Open PYMEAcquire, spool a series to local h5 file 
2. Open PYMEClusterOfOne
3. Open the series in PYMEImage, localize it using latgaussfitfr
4. Try to open your results in PYMEVis
5. see error
6. click OK
7. realize that you see the error because the default filters throw out all your data (low NA air objective -> default sig filters tossed it all!)
8. change filters
9. try to make a new layer so you can actually see something
10. see another error
```
Traceback (most recent call last):
  File "C:\Users\aesb\Miniconda3\lib\site-packages\traits\trait_notifiers.py", line 522, in _dispatch_change_event
    self.dispatch(handler, *args)
  File "C:\Users\aesb\Miniconda3\lib\site-packages\traits\trait_notifiers.py", line 484, in dispatch
    handler(*args)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\ui\layerFoldPanel.py", line 118, in _refresh
    self.Refresh()
RuntimeError: wrapped C/C++ object of type LayerCaptionBar has been deleted
```
11. see no points
12. open my IDE

**Expected behavior**
allow me to open my localizations even if they aren't from a high NA setup. I'm OK bushwacking a bit but its annoying to have to edit code just to open a file.

David-Baddeley, 2022-07-08:
This is an area where usability has definitely regressed. We used to
display an error message (no points, try adjusting filter) in the middle of
the screen, and everything would come right when you changed the filter.
This fell by the wayside when we switched to shaders and multiple layers.

Not so sure on automatically changing the filter, but a number of avenues
to clue the user in that this might be an issue.

As I see it, we can break this down into 2 components:

- make layers more robust against bad input so that things come right when,
e.g. filter settings are adjusted

- clue the user in to the fact that there is something wrong and they need
to check the filter.

David-Baddeley, 2022-07-08:
The actual error you get is from the event display, not the point layers, but it would seem to interrupt point layer generation ...

David-Baddeley, 2022-07-08:
Should have a fix for the first traceback in #1253, but think we need a bit more to wrap this up .... I'm thinking we need a way for recipe modules to display a warning status (e.g. by setting filter background orange when there are no points) much in the same way as module errors are displayed with a red background.