python-microscopy/python-microscopy #1255: [BUG] Left section is missing in the lasted Mac version of PYMEvis

ppa007hot, 2022-07-09:
**Describe the bug**
I was trying PYME of 22.05.25 Mac version. it can be installed by conda successfully, but when I launched PYMEvis and opened my data, I can’t adjust particle size, LUT, or generate isosurface (no reaction after I choose the menu). It looks like a whole section is missing (the left panel in the picture).
[image]

**To Reproduce**
I tried two macbooks, one is 2019 macbook pro (Intel), the other is 2021 macbook pro (M1), both have the same issue.
The installation was done through `conda create -n pyme python=3.7 pyme-depends python-microscopy `

**Screenshots**

[image]

Thank you!

David-Baddeley, 2022-07-09:
That is super odd! Could you give the output of

`conda list`

As a temporary workaround (may not work, but worth trying), try dragging the bar between the left side pane and the points area to try and force a refresh.

ppa007hot, 2022-07-09:
Thank you so much for the help! Please see below for output of conda list. 
I tried dragging the bar, not working. 

(base) Yale-MBP:~ ppa007hot$ conda activate pyme
(pyme) Yale-MBP:~ ppa007hot$ conda list
packages in environment at /Users/ppa007hot/opt/anaconda3/envs/pyme:

Name                    Version                   Build  Channel
alabaster                 0.7.12                   py37_0  
babel                     2.9.1              pyhd3eb1b0_0  
backports                 1.1                pyhd3eb1b0_0  
backports.lzma            0.0.14           py37hb10d4a3_0  
blas                      1.0                         mkl  
blosc                     1.19.0               hab81aa3_0  
bottleneck                1.3.5            py37h67323c0_0  
brotlipy                  0.7.0           py37h9ed2024_1003  
bzip2                     1.0.8                h1de35cc_0  
ca-certificates           2022.4.26            hecd8cb5_0  
cairo                     1.16.0               h691a603_2  
certifi                   2022.6.15        py37hecd8cb5_0  
cffi                      1.15.0           py37hc55c11b_1  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
cheroot                   8.5.2            py37hecd8cb5_1  
cherrypy                  18.6.0           py37hecd8cb5_0  
cloudpickle               2.0.0              pyhd3eb1b0_0  
cryptography              37.0.1           py37hf6deb26_0  
cycler                    0.11.0             pyhd3eb1b0_0  
cython                    0.29.30          py37he9d5cce_0  
cytoolz                   0.11.0           py37haf1e3a3_0  
dask-core                 2021.10.0          pyhd3eb1b0_0  
docutils                  0.18.1           py37hecd8cb5_3  
expat                     2.4.4                he9d5cce_0  
fftw                      3.3.4                         0    david_baddeley
fontconfig                2.13.1               ha9ee91d_0  
freetype                  2.11.0               hd8bbffd_0  
fsspec                    2022.3.0         py37hecd8cb5_0  
future                    0.18.2                   py37_1  
gettext                   0.21.0               h7535e17_0  
giflib                    5.2.1                haf1e3a3_0  
glib                      2.69.1               h8346a28_1  
hdf5                      1.10.4               hfa1e0ec_0  
icu                       58.2                 h0a44026_3  
idna                      3.3                pyhd3eb1b0_0  
ifaddr                    0.1.6                      py_0    david_baddeley
imageio                   2.9.0              pyhd3eb1b0_0  
imagesize                 1.4.1            py37hecd8cb5_0  
intel-openmp              2021.4.0          hecd8cb5_3538  
jaraco.classes            3.2.1              pyhd3eb1b0_0  
jaraco.collections        3.3.0              pyhd3eb1b0_0  
jaraco.functools          3.3.0              pyhd3eb1b0_0  
jaraco.text               3.5.1            py37hecd8cb5_2  
jinja2                    3.0.3              pyhd3eb1b0_0  
joblib                    1.1.0              pyhd3eb1b0_0  
jpeg                      9e                   hca72f7f_0  
kiwisolver                1.4.2            py37he9d5cce_0  
lcms2                     2.12                 hf1fd2bf_0  
libcxx                    12.0.0               h2f01273_0  
libffi                    3.3                  hb1e8313_2  
libgfortran               3.0.1                h93005f0_2  
libiconv                  1.16                 hca72f7f_2  
libpng                    1.6.37               ha441bb4_0  
libtiff                   4.2.0                h87d7836_0  
libwebp                   1.2.2                h56c3ce4_0  
libwebp-base              1.2.2                hca72f7f_0  
libxml2                   2.9.14               hbf8cd5e_0  
llvm-openmp               12.0.0               h0dcd299_1  
locket                    1.0.0            py37hecd8cb5_0  
lz4-c                     1.9.3                h23ab428_1  
lzo                       2.10                 haf1e3a3_2  
markupsafe                2.1.1            py37hca72f7f_0  
[... 54 lines cut ...]
snowballstemmer           2.2.0              pyhd3eb1b0_0  
sphinx                    1.8.5                    py37_0  
sphinxcontrib             1.0                      py37_1  
sphinxcontrib-websupport  1.2.4                      py_0  
sqlite                    3.38.5               h707629a_0  
tempora                   4.1.1              pyhd3eb1b0_0  
threadpoolctl             2.2.0              pyh0d69192_0  
tk                        8.6.12               h5d9f67b_0  
toolz                     0.11.2             pyhd3eb1b0_0  
toposort                  1.5                      py37_3    david_baddeley
tornado                   6.1              py37h9ed2024_0  
traits                    6.2.0            py37h9ed2024_0  
traitsui                  7.1.0                      py_0  
typing_extensions         4.1.1              pyh06a4308_0  
ujson                     5.1.0            py37he9d5cce_0  
urllib3                   1.26.9           py37hecd8cb5_0  
wheel                     0.37.1             pyhd3eb1b0_0  
wxpython                  4.0.4            py37hebc47a7_0  
xz                        5.2.5                hca72f7f_1  
yaml                      0.2.5                haf1e3a3_0  
zc.lockfile               2.0                pyhd3eb1b0_0  
zeroconf                  0.26.3                   py37_3    david_baddeley
zlib                      1.2.12               h4dc903c_2  
zstd                      1.4.9                h322a384_0  
(pyme) Yale-MBP:~ ppa007hot$

David-Baddeley, 2022-07-10:
Really odd - just tried to replicate an install into a clean environment on my machine (2021 M1 mac) and don't get the same issue. 

Having a quick look at your package list, the only obvious anomaly is that your pyme version (`python-microscopy 22.5.25.post0.dev0 dev_0`) would seem to indicate a development install, compatible with (abridged)

```bash
conda create -n pyme python=3.7 pyme-depends
git clone python-microscopy
python setup.py develop
```

rather than 

```bash
conda create -n pyme python=3.7 pyme-depends python-microscopy
```

This in itself shouldn't be an issue (i.e. that installation method should also work), but (and I'm grasping at straws here) there is a chance that it's indicative of a previous failed installation / or the environment already exisiting.  Just in case, it's probably worth removing the environment( https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-environments.html#removing-an-environment ), or trying a clean install in a new environment with a different name. e.g.:

```bash
conda create -n pyme2 python=3.7 pyme-depends python-microscopy
```

The only other thing that might be worth a look at is the contents of your `~/.PYME` settings directory - if there was a previous installation there could be extension modules registered there that are no longer present (although this generally just results in a logged error on the console and no adverse effects). If that looks like it could be the problem it's safe to delete the `~/.PYME` directory.

ppa007hot, 2022-07-11:
Thank you, David! I did have multiple failed installations. Now I have removed everything including anaconda.
Then I installed new miniconda, and tried both formal and developmental installs (and inbetween I manually delete pyme and python-microscopy folders). I got the same issue (missing left section). 
When I open my data, there is a window reporting the following:

> Error whilst running File>&Open
> 
> python-microscopy=22.05.25[conda]
> python=3.7.6, platform=darwin
> numpy=1.21.5, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5
> 
> Traceback
> 
> Traceback (most recent call last):
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/PYME/ui/progress.py”, line 153, in func
>     return fcn(*args, **kwargs)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/PYME/LMVis/visCore.py”, line 399, in OnOpenFile
>     self.OpenFile(filename)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/PYME/LMVis/visCore.py”, line 765, in OpenFile
>     self.recipeView._layout()
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/PYME/recipes/vertical_recipe_display.py”, line 150, in _layout
>     pan = node.edit_traits(parent=item, kind=kind, view=‘pipeline_view_min’)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traits/has_traits.py”, line 1734, in edit_traits
>     view = self.trait_view(view)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/PYME/recipes/base.py”, line 291, in trait_view
>     import traitsui.api as tui
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traitsui/api.py”, line 30, in <module>
>     from .editors.api import ArrayEditor
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traitsui/editors/__init__.py”, line 22, in <module>
>     from .api import ArrayEditor
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traitsui/editors/api.py”, line 24, in <module>
>     from .code_editor import CodeEditor
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traitsui/editors/code_editor.py”, line 34, in <module>
>     class ToolkitEditorFactory(EditorFactory):
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traitsui/editors/code_editor.py”, line 46, in ToolkitEditorFactory
>     mark_color = Color(0xECE9D8)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traits/trait_factory.py”, line 40, in __call__
>     return self.maker_function(*args, **metadata)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traitsui/toolkit_traits.py”, line 43, in ColorTrait
>     return toolkit().color_trait(*args, **traits)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traitsui/toolkit.py”, line 113, in toolkit
>     _toolkit = find_toolkit(“traitsui.toolkits”)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/pyface/base_toolkit.py”, line 263, in find_toolkit
>     return import_toolkit(ETSConfig.toolkit, entry_point)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/pyface/base_toolkit.py”, line 217, in import_toolkit
>     toolkit_object = plugin.load()
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/pkg_resources/__init__.py”, line 2457, in load
>     self.require(*args, **kwargs)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/pkg_resources/__init__.py”, line 2480, in require
>     items = working_set.resolve(reqs, env, installer, extras=self.extras)
>   File “/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/pkg_resources/__init__.py”, line 788, in resolve
>     raise VersionConflict(dist, req).with_context(dependent_req)
> pkg_resources.VersionConflict: (pyface 6.1.2 (/Users/ppa007hot/opt/miniconda3/envs/pyme/lib/python3.7/site-packages), Requirement.parse(‘pyface>=7.1.0’))
> 

Indicating a pyface version conflict. Then I tried `conda update pyface`, it will downgrade the python-microscopy to 21.02.18! I don't want that...

David-Baddeley, 2022-07-12:
Hmm, pyme-depends should pin both traitsui and pyface to <=7.1.0 as higher versions don't work with wxpython (this is probably why it wants to downgrade so far, as that is probably when the pins were introduced). You could try installing pyface=7.1.0. Still really confusing, as a clean install on a similar computer worked fine for me. 

Could you show the output of:

```
conda config --show channels
```

For me (functioning install) it's:

```
channels:
  - david_baddeley
  - defaults
```

you can modify using the conda config command, or by editing .condarc.

David-Baddeley, 2022-11-28:
can we call this resolved?

David-Baddeley, 2022-12-14:
Closing as there have been a couple of releases where this can't be  replicated. Feel free to reopen if still a problem.

closed 2022-12-14 by David-Baddeley