python-microscopy/python-microscopy #1262: [ENHANCEMENT] PYMEImage feature suggestions

d-perez-1, 2022-07-19:
PYMEAcquire has been super useful for me, and I'd like to continue using it for more of my imaging experiments, but I keep finding myself wanting to use ImageJ instead of PYMEImage for quickly viewing/spot checking data. I figured I'd list some of the features that would be nice for me to have in PYMEImage so I can leave ImageJ behind. I realize some of these are a probably a lot of work, and some may exist and I just don't know where to look for them. Anyway, the list in no particular order:

- Improved histogram scaling options. One click/key press to scale histogram to max and min of the frame. One click for an "autoscale" type function, maybe this is what the stretch button does but the bounds it chooses aren't usually useful to me. Option to apply scaling from one image to another or all open images.
- Update the displayed image while dragging through z. So I can more easily scan a video for a particular thing.
- Measure mean, std, area of a selected region, like ImageJs measurements window.
- When zoomed on an image, some way to click and move the image around. Maybe hold down shift then click and drag.
- Open new images with the shell, events, etc tabs displayed. Don't show Point Finding, Analysis, Fit Status windows by default.
- Export a compressed video format (.avi for example) with given framerate, histogram scaling. For inclusion in powerpoints, etc.
- Easily switch between camera output value and photoelectrons (if gains are already in metadata).

Happy to elaborate on any point if this is helpful, and I'm technically willing to work on this but I don't really have the expertise to implement any of these.

David-Baddeley, 2022-07-19:
Thanks for the suggestions ... a few notes below:

### Histogram scaling

There are currently 2 keyboard short-cuts 

`m` - which performs min-max scaling, and
`p` - which does percentile based scaling (same as the stretch button).

Hitting these keys when either the image panel or the histogram panel has focus should work.

The auto-scaling being not so useful is a little trickier, as it's practically impossible to find an autoscaling algorithm (other than min-max) which works well across different image types. We could look at making the percentiles that the algorithm uses a configurable option so you could find something that mostly works for your data and remember that. The autoscaling algorithm is currently particularly bad on very sparse images, as we only use a subset of the pixels to compute the histogram (and by extension the scaling). This is a performance optimisation which is critical when looking at very large images (incl. tiled images which don't fit in memory), but might be a bit aggressive for standard smaller images.

As to synchronisation, there is the `Processing->Sync Windows` option, although I'm not sure if it currently synchronises histogram scaling. The (very) crude way of doing it is to right click on the histograms which lets you manually enter bounds.

### Display update when z-dragging

Not updating the display when z-dragging/time is unfortunately a conscious choice to ensure acceptable performance on large time series which are often not loaded in memory (typical single molecule blinking data). If you need to get from say frame 100 to frame 18000 in such a series, updating on every frame while you drag takes forever. There are two potential workarounds for your desired behaviour: 

1) Use the "play" function by the scalebar, and hit pause when you see something interesting
2) Use the "page-down" key (fn-down arrow on mac). Holding down `page-down` should scroll through images one at a time.

### Measurements

Agree this would be nice

### Panning

In other parts of PYME, panning is handled by dragging with the middle mouse button. The stubs for this are already in the PYMEImage code (they are used in PYMEAcquire to move the stage, rather than the image), and it would be fairly simple to add handlers that panned the image (if zoom permitted). The OpenGL based viewer already supports panning, so the other option would be to wait until we are using that.

### Opening mode

I'm going to assume that you have saved your data as .h5. PYMEImage tries to guess an opening mode based on file extension and currently assumes that .h5 implies a raw single molecule blinking series (the `lm` mode, which has the point finding and analysis panes). You can over-ride this on the command line by manually specifying a mode (`PYMEImage -m default filename` should probably do what you want). We should make the mode guessing a bit more intelligent and/or configurable.

### Video export

We've had this at one point with PYMEVis, and the simplest way would seem to involve having OpenCV as a dependency which can be problematic in terms of getting a clean install. In the interests of keeping the core reasonably easy to install, I'd lean towards coding such that it worked if OpenCV was installed, but required the user to manually install that on top of PYME. In any case, there's probably some merit in refactoring the PYMEVis movie export code (which currently dumps a folder full of .png/.jpg which you can combine with ffmpeg) so that the same code was useable in PYMEImage, and so that it could alternatively use an OpenCV backend if present.

### Photoelectrons

This has been discussed a little in a previous issue, and is generally a good idea with a few caveats, and has an open PR which I need to get to. What do you want to be in photoelectrons -  the histograms? Measurements? All subsequent processing? In terms of implementation, the easiest might be to simply create a new copy/view of the data with the different units.

### General

It would be nice to have a quick overview of what type of data you are looking at (time series, z series, etc ...) to put some of this in context.

d-perez-1, 2022-07-19:
Thanks for the reply, great to hear some of the functionality is already there.

In general I'm looking at time series of sparse images (which explains why stretch is unhelpful). Usually of single molecules/particles, either blinking or spectroscopy.

d-perez-1, 2022-10-28:
Regarding histogram scaling in PYMEImage, max/min scaling by pressing m is not behaving as I would expect. The following pictures are the same image of beads.

Pressing m with image as focus crops histogram around offset:
[image]
Pressing m with histogram as focus still leaves peaks saturated:
[image]
Manually scaling so no pixels are saturated:
[image]

Is it possible this histogram scaling function is using a subset of pixels like PYMEAcquire was, maybe from the whole movie when the image is selected or something like that?

A couple other things that would be nice:
Would be nice to have intensity of selected pixel next to x,y,z
[image]

Why does the crosshair go to seemingly random parts of the pixel?
[image]

David-Baddeley, 2022-10-28:
- Crosshair location within a pixel is due to scrolling - if the scroll is not an integer number of pixels the crosshair is offset - should be reasonably easy to fix. 

- Agree that displaying intensity underneath the cursor would be nice.

David-Baddeley, 2022-10-30:
Scroll offset is not quite as simple as I thought - it's the image display that "cogs" (crosshairs move as expected with scrolling). Fixing the image display is going to be slightly tricky, so need to decide if we fix the image display, or make everything "cog" in the same way.

barentine, 2022-11-22:
Forgive me for not digging for the commit where it went in (if I remember, it went in with a PR focused on something else), but I'm seeing the intensity display work for a click or two and then throw:
```
ERROR:PYME.DSView.displayOptions:Error in OnChange
Traceback (most recent call last):
  File "c:\userfiles\andrew\code\python-microscopy\PYME\DSView\displayOptions.py", line 332, in OnChange
    fcn()
  File "c:\userfiles\andrew\code\python-microscopy\PYME\DSView\dsviewer.py", line 280, in update
    statusText += '    I: (%s)' % ', '.join(['%3.3f' % self.do.ds[self.do.zp, self.do.xp, self.do.yp, self.do.tp, c] for c in range(self.do.ds.shape[4])])
  File "c:\userfiles\andrew\code\python-microscopy\PYME\DSView\dsviewer.py", line 280, in <listcomp>
    statusText += '    I: (%s)' % ', '.join(['%3.3f' % self.do.ds[self.do.zp, self.do.xp, self.do.yp, self.do.tp, c] for c in range(self.do.ds.shape[4])])
  File "c:\userfiles\andrew\code\python-microscopy\PYME\IO\DataSources\BaseDataSource.py", line 347, in __getitem__
    return out.squeeze(axis=tuple(to_squeeze))
ValueError: cannot select an axis to squeeze out which has size not equal to one
```
after which is no longer updates. Win 10, wx 4.0.4 msw (phoenix) wxWidgets 3.0.5

David-Baddeley, 2022-11-22:
Hmm - that's super odd any specifics on data (file) type/shape/ etc ...?

David-Baddeley, 2022-11-22:
Can't replicate on mac, with `PYMEImage --test3d`, wonder if it's file type specific ...

barentine, 2022-11-22:
PYMEAcquired h5, 128x256xcouple thousand
Single color.

David-Baddeley, 2022-11-22:
OK, can replicate with .h5, seems to be related to changing z position, rather than number of clicks (ie happens straight away when changing z, doesn't seem to happen at all when just clicking around (including out of bounds).

David-Baddeley, 2022-11-25:
Intensity display should be fixed with #1313 (super embarrassing x-z index swap)