python-microscopy/python-microscopy #1263: [ENHANCEMENT] Save Time for events as float or string

d-perez-1, 2022-07-19:
**Is your feature request related to a problem? Please describe.**
PYMEAcquire saves time for events in the hdf5 file in a format not supported by various python packages or matlab. Python sees the data type as Time64Col, and trying to view the data in matlab throws an error, and displaying the contents shows the data type as H5T_TIME (unsupported).

**Describe the solution you'd like**
Would be great if the data could be saved as a different type (i.e. float or string) so PYMEAqcuire data can more easily be used in existing analysis workflows using other software packages.

**Describe alternatives you've considered**
Using PYME to save events as a separate text file which, then loading that along with ImageData instead of getting everything straight from the .h5

**Additional context**
For matlab:
```
h5disp('8_7_series_E.h5');
Warning: H5T_TIME is not a supported HDF5 class. 
In h5info (line 125)
In h5disp>display_hdf5 (line 144)
In h5disp (line 122) 
HDF5 8_7_series_E.h5 
Group '/' 
    Attributes:
        'TITLE':  ''
        'CLASS':  'GROUP'
        'VERSION':  '1.0'
        'PYTABLES_FORMAT_VERSION':  '2.1'
    Dataset 'Events' 
........
Member 'Time':  H5T_TIME (unsupported)
```

For python 3.7 (for a different file):
```
In [9]: t.root.Events
Out[9]:
/Events (Table(1,)) ''
  description := {
  "EventDescr": StringCol(itemsize=256, shape=(), dflt=b'', pos=0),
  "EventName": StringCol(itemsize=32, shape=(), dflt=b'', pos=1),
  "Time": Time64Col(shape=(), dflt=0.0, pos=2)}
  byteorder := 'little'
  chunkshape := (221,)
```

**Are you prepared to work on it?**
If I implement my own workaround I'll share it.

David-Baddeley, 2022-07-19:
Definitely supported in python as `Time64Col` (which casts to a double, or potentially a datetime object as soon as you access the events table, by, e.g. `t.root.Events[:]`), and part of the H5 standard (https://support.hdfgroup.org/HDF5/doc1.6/UG/11_Datatypes.html), so basically means that Matlab has a broken H5 implementation, and that this is a Matlab bug. 

I'm somewhat loath to change our code to work around a Matlab bug (as it stands, the H5T_TIME tells you that the value should be interpreted as a POSIX timestamp, if we just make it double we need to know/describe that elsewhere which kind of defeats the "self documenting" nature of HDF5).  That said, there does seem to be some confusion as to how H5T_TIME should be interpreted (more recent HDF5 docs ... https://docs.hdfgroup.org/hdf5/develop/group___h5_t.html#ga364545c053f925fec65880b235e37898) - I don't think that it is really standardised across implementations, so I'll grudgingly say that it might not hurt to change to a double for matlab compatibility (given suitable testing etc ...).

We definitely don't want to do string, as it adds parsing and file size overhead, as well as loss of precision.

d-perez-1, 2022-07-19:
Thanks for info. If its just a matlab issue I can work around it myself when using our existing matlab analysis. I wasn't sure if there would be issues with other platforms as well warranting a change.

David-Baddeley, 2022-07-19:
My attitude evolved as I was writing the first reply (as you might be able to tell), it seems that the `H5T_TIME` type is poorly defined and not portable (which also dilutes/eliminates much of the self documenting advantage. This might be a case of pragmatism and if double (float64) works without major modifications to the rest of the software, it might be reasonable to change.

closed 2022-07-25 by David-Baddeley