python-microscopy/python-microscopy #1267: [BUG] FFBase._getROI finnicky for 2D fits when called from recipe module (array data source issue?)

barentine, 2022-07-20:
**Describe the bug**
Something about the common-code for 2D and 3D get ROIs works for some datasources and not for others - unclear exactly what the difference is yet

```
----> 8 outputs = FitPoints().apply(inputImage=something, inputPositions=inp)

c:\userfiles\code\python-microscopy\PYME\recipes\base.py in apply(self, *args, **kwargs)
    213                 namespace[getattr(self, k)] = v
    214 
--> 215         self.execute(namespace)
    216 
    217         return {k : namespace[k] for k in self.outputs}

c:\userfiles\code\python-microscopy\PYME\recipes\measurement.py in execute(self, namespace)
    228 
    229             #print x/ps, y/ps
--> 230             r[i] = ff.FromPoint(x/ps, y/ps, roiHalfSize=self.roiHalfSize)
    231 
    232         res = tabular.FitResultsSource(r, sort=False)

c:\userfiles\code\python-microscopy\PYME\localization\FitFactories\LatGaussFitFR.py in FromPoint(self, x, y, z, roiHalfSize, axialHalfSize)
    151 
    152     def FromPoint(self, x, y, z=None, roiHalfSize=5, axialHalfSize=15):
--> 153         X, Y, data, background, sigma, xslice, yslice, zslice = self.getROIAtPoint(x, y, z, roiHalfSize, axialHalfSize)
    154 
    155         dataMean = data - background

c:\userfiles\code\python-microscopy\PYME\localization\FitFactories\FFBase.py in getROIAtPoint(self, x, y, z, roiHalfSize, axialHalfSize)
    111             zslice - z slice into original data array
    112         """
--> 113         xslice, yslice, zslice, dataROI, sigma, bgROI = self._get_roi(x, y, z, roiHalfSize, axialHalfSize)
    114 
    115 

c:\userfiles\code\python-microscopy\PYME\localization\FitFactories\FFBase.py in _get_roi(self, x, y, z, roiHalfSize, axialHalfSize)
     79         y = int(round(y))
     80         if (z is None): # use position of maximum intensity
---> 81             z = self.data[x, y, :].argmax()
     82 
     83         roiHalfSize = int(roiHalfSize)

IndexError: too many indices for array: array is 2-dimensional, but 3 were indexed
```

**To Reproduce**
Steps to reproduce the behavior:
1. Load a PYMEAcquire h5 series
2. run the FitPoints recipe module on it - no errors, cool
3. mess around with your original datasource, make a new datasource from it via e.g. `ImageStack(data=[some numpy thing on your old datasource])`
4. run FitPoints on this new datasource
5. see error

**Expected behavior**
not break trying to find z for a 2D fit

barentine, 2022-07-20:
So the erroneous datasource is an `PYME.IO.DataSources.ArrayDataSource.ArrayDataSource`
while things work fine with a `PYME.IO.DataSources.BaseDataSource.XYTCWrapper`.

which makes sense as far as whether the datasource will automatically broadcast a dim z or not. 

I guess my question now is: what is preferable to 'fix'?

David-Baddeley, 2022-07-20:
Simplest fix would be to change line 226 of recipes/measurement.py

from:
```                ff = fitMod.FitFactory(img.data[:, :, t, self.channel], md) ```
 to
```                ff = fitMod.FitFactory(np.atleast_3d(img.data[:, :, t, self.channel)], md) ```

Have it changed in the editor here, albeit on a wip branch. Will PR (hopefully on top of master rather than the wip branch.

barentine, 2022-07-20:
sounds good! Thanks David

PR #1268 (2022-07-20): issue1267
fixes #1267.
closed 2022-07-20 by David-Baddeley