python-microscopy/python-microscopy #1278: [ENHANCEMENT] Reload Python module on Push to RuleServer in `debug` mode

zacsimile, 2022-08-31:
**Is your feature request related to a problem? Please describe.**
It would be nice if, when I'm testing scripts I want to push to the rule server, I could launch the server in a debug mode where it will re-import Python modules on push from another command line. As it stands, if I modify code, I have to close and then re-open the server to re-import.

**Are you prepared to work on it?**
Yes

David-Baddeley, 2022-08-31:
This actually falls into the category of "fiendishly difficult", as you
need to do a deep reload and identify all the modules which are downstream
of a module (i.e. imports the changed module, or import a module that
imports the changed module, ad infinitum). If you are following best
practice and having your main logic outside the recipe module with the
module just acting as glue, then a lot of the time you are going to want to
reload this logic file and not just the recipe module. The only two
packages (CherryPy and django) that I am aware of which do any form of
change detection and reloading do a hard reload and basically completely
restart the process. It's even a bit harder for us, as we have 4 separate
types of server that run as part of the PYME-CLUSTER, and you'd need to
decide which of these needed reloading and do them in the right order. A
further complication is what to do about analysis in progress - do you
abort it or not? CherryPy and django get away with aborting (in most
situations) because the webframesworks are generally considered
stateless, although I suspect you could find edge cases. We also do some
reloading in the acquisition protocols, but that is strictly a shallow
reload, and we get away with it because we don't import those modules
anywhere else.

David-Baddeley, 2022-08-31:
PS - you would also need to get rid of any old object instances - these
stick around and use the old version of the code following a reload.

David-Baddeley, 2022-11-30:
closing as no simple implementation

closed 2022-11-30 by David-Baddeley