python-microscopy/python-microscopy #1285: [BUG] Save snapshot is broken on OSX with wx 4.1.1

zacsimile, 2022-09-24:
```
 Error whilst running View>Save Snapshot
==============================================
python-microscopy=22.09.20.post0.dev[git]
python=3.8.11, platform=darwin
numpy=1.23.1, wx=4.1.1 osx-cocoa (phoenix) wxWidgets 3.1.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/zachmarin/Code/python-microscopy/PYME/ui/progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "/Users/zachmarin/Code/python-microscopy/PYME/LMVis/Extras/snapshot.py", line 135, in <lambda>
    vis_fr.AddMenuItem('View', 'Save Snapshot', lambda e: save_snapshot(vis_fr.glCanvas))
  File "/Users/zachmarin/Code/python-microscopy/PYME/LMVis/Extras/snapshot.py", line 110, in save_snapshot
    settings.configure_traits(kind='modal')
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traits/has_traits.py", line 2174, in configure_traits
    rc = toolkit().view_application(
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/toolkit.py", line 217, in view_application
    return view_application.view_application(
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/view_application.py", line 95, in view_application
    ui = view.ui(
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/view.py", line 462, in ui
    ui.ui(parent, kind)
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/ui.py", line 244, in ui
    self.rebuild(self, parent)
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/toolkit.py", line 132, in ui_modal
    ui_modal.ui_modal(ui, parent)
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/ui_modal.py", line 46, in ui_modal
    _ui_dialog(ui, parent, BaseDialog.MODAL)
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/ui_modal.py", line 65, in _ui_dialog
    BaseDialog.display_ui(ui, parent, style)
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/ui_base.py", line 71, in display_ui
    ui.owner.init(ui, parent, style)
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/ui_modal.py", line 139, in init
    sw = panel(ui, window)
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/ui_panel.py", line 273, in panel
    sg_sizer, resizable, contents = fill_panel_for_group(
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/ui_panel.py", line 432, in fill_panel_for_group
    fp = FillPanel(
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/ui_panel.py", line 589, in __init__
    self.add_items(content, panel, self.sizer)
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/ui_panel.py", line 909, in add_items
    editor.prepare(item_panel)
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/editor.py", line 236, in prepare
    self.init(parent)
  File "/Users/zachmarin/opt/miniconda3/envs/pyme/lib/python3.8/site-packages/traitsui/wx/file_editor.py", line 96, in init
    sizer.Add(control, 1, wx.EXPAND | wx.ALIGN_CENTER)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL))" failed at /Users/zachmarin/Software/wxPython-4.1.1/ext/wxWidgets/src/common/sizer.cpp(2184) in DoInsert(): Vertical alignment flags are ignored with wxEXPAND
```

David-Baddeley, 2022-10-13:
Partially resolved by updating to traitsui=7.4.1 (with code on current head)

This works if you type the filename in, but gives the error below if you click the button which should pop up the file chooser:
```
Traceback (most recent call last):
  File "/Users/david/opt/miniconda3/envs/pyme_as/lib/python3.8/site-packages/traitsui/wx/file_editor.py", line 147, in show_file_dialog
    dlg = self._create_file_dialog()
  File "/Users/david/opt/miniconda3/envs/pyme_as/lib/python3.8/site-packages/traitsui/wx/file_editor.py", line 209, in _create_file_dialog
    default_path=self._file_name.text(),
AttributeError: 'TextCtrl' object has no attribute 'text'
```

zacsimile, 2023-01-18:
In a continuing saga, it looks like the snapshot only grabs 1/4 of the field of view.

The window: 
[image]

The resulting snapshot:
[image]

Is this related to the previous issue of the main window only displaying in 1/4 of the available space?

David-Baddeley, 2023-03-15:
https://github.com/enthought/traitsui/pull/1993 should fix the clicking on button issue.

David-Baddeley, 2023-09-18:
Think this should now be fixed

closed 2023-09-18 by David-Baddeley