python-microscopy/python-microscopy #1298: [BUG] Measure2D module breaks with latest skimage.measure.regionprops

csoeller, 2022-10-13:
**Describe the bug**
When using the `measure2D` module in recipes it raises an error due to a naming change of some image properties in the `regionprops`method.

I am aware of #524 but am suggesting to fix this breakage anyway ahead of unifying the various interfaces.

**To Reproduce**
Steps to reproduce the behavior:
1. Open image in `dh5view`
2. add `Measure2D`module in suitable recipe
3. Execute recipe
4. See error

```
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYMEsrc/pyme38/PYME/recipes/recipe.py", line 222, in execute
    m.execute(self.namespace)
  File "/Users/csoe002/Documents/src/PYMEsrc/pyme38/PYME/recipes/measurement.py", line 753, in execute
    m.addFrameMeasures(i, *self._measureFrame(i, labels, intensity))
  File "/Users/csoe002/Documents/src/PYMEsrc/pyme38/PYME/recipes/measurement.py", line 676, in addFrameMeasures
    self._keys.remove('convex_image')
ValueError: list.remove(x): x not in list

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYMEsrc/pyme38/PYME/ui/progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYMEsrc/pyme38/PYME/DSView/modules/recipes.py", line 102, in RunCurrentRecipe
    self.outp = self.activeRecipe.execute(input=self.image)
  File "/Users/csoe002/Documents/src/PYMEsrc/pyme38/PYME/recipes/recipe.py", line 263, in execute
    raise RecipeExecutionError('Recipe execution failed in module %s' % self._failing_module._module_name, self ) from e
PYME.recipes.recipe.RecipeExecutionError: Recipe execution failed in module measurement.Measure2D
```

**Reason**
Latest `skimage.measure.regionprops` has renamed some of the `XXX_image` properties to `image_XXX`. The patch below fixes this in a backwards compatible manner and is a little more robust in doing it, if ok I can submit as PR.

```patch
diff --git a/PYME/recipes/measurement.py b/PYME/recipes/measurement.py
index 6ee1170b7..51668bb0b 100644
--- a/PYME/recipes/measurement.py
+++ b/PYME/recipes/measurement.py
@@ -669,14 +669,21 @@ class Measure2D(ModuleBase):
                     #first time we've called this - determine our data type
                     self._keys = ['t', 'x', 'y'] + [r for r in dir(measurements[0]) if not r.startswith('_')]
                     
-                    self._keys.remove('euler_number') #buggy!
+                    if 'euler_number' in self._keys:
+                        self._keys.remove('euler_number') #buggy!
                     
                     # remove all the image measurements - as these will potentially generate export errors
                     # and are not super-useful
-                    self._keys.remove('convex_image')
-                    self._keys.remove('filled_image')
-                    self._keys.remove('image')
-                    self._keys.remove('intensity_image')
+                    # the naming has changed recently, so check for two possible namings
+                    for key in ['convex_image',
+                                'filled_image',
+                                'image_convex',
+                                'image_filled',
+                                'image',
+                                'intensity_image',
+                                'image_intensity']:
+                        if key in self._keys:
+                            self._keys.remove(key)
                     
                     if not contours is None:
                         self._keys += ['contour']
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
- python-microscopy=22.10.03.post0.dev[git]85d65a224ec6
- python=3.8.13, platform=darwin
- numpy=1.23.2, wx=4.2.0 osx-cocoa (phoenix) wxWidgets 3.2.0

David-Baddeley, 2022-10-13:
Definitely keen to fix promptly - suggested changes look good - a PR would be welcome!

David

David-Baddeley, 2022-10-13:
@csoeller - making a new release today, so will just copy the changes across rather than waiting for a PR

PR #1303 (2022-10-13): Fix Measure2D for recent skimage
fixes #1298
closed 2022-10-13 by David-Baddeley