python-microscopy/python-microscopy #1307: [BUG] --root option doesn't work for PYMEClusterOfOne

zacsimile, 2022-10-23:
If you pass a custom `--root` option to `pymeclusterofone`, it creates a `LOGS` directory at your location of choice, but if you run any tasks, the data is still saved in the default `PYMEData` folder.

zacsimile, 2022-10-23:
However, specifying `dataserver-root` in `config.yaml` does work.

David-Baddeley, 2022-10-26:
This is a bit more complex - it does work for the dataserver itself, but not for anything which bypasses the dataserver because the data resides on the local node, which is the case for most IO. Probably also not for scoring of tasks (`is_local()` and similar). In practice, it will probably be respected for the `__aggregate` endpoints, but not direct saving. The simplest solution is probably simply to remove the --root option.

PR #1317 (2022-11-30): issue #1307
fixes #1307 by removing root directory option, leaving PYME.config as the sole method of specifying it. Also adds an option to set the number of workers launched
closed 2022-11-30 by David-Baddeley