python-microscopy/python-microscopy #1310: [SUPPORT]

d-perez-1, 2022-10-28:
**What are you trying to do?**
Estimate pupil function from z stack

**Where are you stuck?**
I've used PYMEImage PSF extraction, and can see estimate of Cramer-Rao lower bound. Andrew suggested there is code to estimate pupil function, but doesn't know where it is.

Also wondering what dashed lines on this plot are:
[image]

David-Baddeley, 2022-10-28:
I'll double check next time I'm in front of the computer, but I'm pretty
sure the pupil extraction is on the "Processing" menu. You will need the
"psf_tools" module loaded (should be there by default after PSF extraction).

As to the dotted curves, they are the theoretical CRLB for a simulated
*astigmatic* PSF (NA 1.45, lambda around 700, I forget how much
astigmatism, but about typical for astigmatic localisation), and give a
rough indication of how good your PSF is for 3D localisation.

David-Baddeley, 2022-10-28:
`Processing->Extract Pupil` from the menu is the correct combination. If you want to get better sampling of the pupil you might need to make the area of the PSF larger (i.e. change the half sizes used for extraction).

d-perez-1, 2022-10-28:
Great thank you!

closed 2022-10-28 by d-perez-1