python-microscopy/python-microscopy #1319: [BUG] Error when open PYMEAcquire

Yujin-Bao, 2022-12-03:
**Describe the bug**
`ImportError` when opening `PYMEAcquire`. It seems it is due to the last commit of `displaySettingsPanel.py`, which is "make histogram background change colour on saturation".

**To Reproduce**
Open `PYMEAcquire` (with PCO camera) and you will see the error and traceback.

**Screenshots**
[image]

barentine, 2022-12-03:
Hi @Yujin-Bao , am I correct that this is a development install and you didn't rerun `python setup.py develop` after updating? Give that a try and let us know if that fixes things

Yujin-Bao, 2022-12-03:
Hi @barentine, thanks for the hint! It works after I reran python `setup.py develop`!

barentine, 2022-12-04:
Great! Feel free to close the issue then. Thanks

David-Baddeley, 2022-12-04:
Think we can close this as resolved, but make a mental not that there should be some way of indicating when a rebuild is needed for source installs (safest is to assume it always is - do the docs reflect this?)

closed 2022-12-04 by David-Baddeley