python-microscopy/python-microscopy #1321: [PROPOSAL] IDS Peak API version of ueye / ucam camera classes

barentine, 2022-12-19:
**Is your feature request related to a problem? Please describe.**
IDS has deprecated their UI cameras, e.g. IDS-UI-3240ML-NIR-GL, and their new U3 cameras run on IDSPeak. More info [here](https://www.1stvision.com/cameras/IDS-imaging-software).

Note that uEye (UI) cameras appear to also be supported by IDS Peak (see [here](https://www.1stvision.com/cameras/IDS/IDS-manuals/en/ueye-intro.html))

**Describe the solution you'd like to implement**
A camera class using IDS Peak API

**Describe alternatives you've considered**
buying old cameras

David-Baddeley, 2022-12-20:
Sounds good to me ...

barentine, 2023-03-15:
Just an update that we have an IDS Peak only camera inbound - will give this a try at some point

barentine, 2024-04-26:
addressed with merge of #1454

closed 2024-04-26 by barentine