python-microscopy/python-microscopy #1325: [PROPOSAL] test imports have some unmarked expected fails on win10 py3.7 pyme-depends 1.18, dev install

barentine, 2023-01-24:
**Describe the bug**
<details> <summary> pytest output </summary>

```
(pyme) C:\code\python-microscopy\tests>pytest
================================================= test session starts =================================================
platform win32 -- Python 3.7.16, pytest-7.1.2, pluggy-1.0.0
rootdir: C:\code\python-microscopy
collected 122 items

PYME\test_imports.py F.F..                                                                                       [  4%]
PYME\Acquire\test_tweeter.py s                                                                                   [  4%]
PYME\Analysis\test_piezo_movement_correction.py ..                                                               [  6%]
PYME\Analysis\points\test_clumps.py .....                                                                        [ 10%]
PYME\Analysis\points\test_cluster_morphology.py ...                                                              [ 13%]
PYME\Analysis\points\test_coordinate_tools.py ..........                                                         [ 21%]
PYME\Analysis\points\test_multiview.py ..                                                                        [ 22%]
PYME\Analysis\points\test_ripleys.py ..                                                                          [ 24%]
PYME\Analysis\points\test_spherical_harmonics.py ...x...s                                                        [ 31%]
PYME\Analysis\points\test_traveling_salesperson.py .....                                                         [ 35%]
PYME\Analysis\points\DistHist\test_dist_hist.py ....                                                             [ 38%]
PYME\DSView\LUT\test_lut.py .                                                                                    [ 39%]
PYME\IO\test_cluster_spooler.py F                                                                                [ 40%]
PYME\IO\test_countdir.py F                                                                                       [ 40%]
PYME\IO\test_dataserver.py FFFFFFsE                                                                              [ 46%]
PYME\IO\test_h5.py ...F                                                                                          [ 50%]
PYME\IO\test_pipeline_io.py .X                                                                                   [ 51%]
PYME\IO\test_pzf.py .........                                                                                    [ 59%]
PYME\experimental\test_octree.py ...                                                                             [ 61%]
PYME\experimental\test_treap.py ..                                                                               [ 63%]
PYME\experimental\test_triangle_mesh.py sssssssssssssssss                                                        [ 77%]
PYME\experimental\test_triangular_mesh.py xxxxxxxxx                                                              [ 84%]
PYME\localization\test_MultifitSR.py .s                                                                          [ 86%]
PYME\localization\test_dumbell.py .                                                                              [ 86%]
PYME\localization\test_localization_algorithms.py ......                                                         [ 91%]
PYME\localization\FitFactories\Interpolators\test_interpolators.py ...                                           [ 94%]
PYME\recipes\test_acquisition.py .                                                                               [ 95%]
PYME\recipes\test_measurement.py ..X                                                                             [ 97%]
PYME\recipes\test_processing.py .                                                                                [ 98%]
PYME\recipes\test_recipes.py .                                                                                   [ 99%]
PYME\recipes\test_tablefilters.py .                                                                              [100%]

======================================================= ERRORS ========================================================
_________________________________ ERROR at teardown of test_mulithread_result_filing __________________________________

    def teardown_module():
        global proc, tmp_root
>       proc.send_signal(signal.SIGINT)

PYME\IO\test_dataserver.py:36:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <subprocess.Popen object at 0x0000020F933DEF48>, sig = <Signals.SIGINT: 2>

    def send_signal(self, sig):
        """Send a signal to the process."""
        # Don't signal a process that we know has already died.
        if self.returncode is not None:
            return
        if sig == signal.SIGTERM:
            self.terminate()
        elif sig == signal.CTRL_C_EVENT:
            os.kill(self.pid, signal.CTRL_C_EVENT)
        elif sig == signal.CTRL_BREAK_EVENT:
            os.kill(self.pid, signal.CTRL_BREAK_EVENT)
        else:
>           raise ValueError("Unsupported signal: {}".format(sig))
E           ValueError: Unsupported signal: 2

C:\Users\Administrator\anaconda3\envs\pyme\lib\subprocess.py:1335: ValueError
[... 232 lines cut ...]
yaml                      0.1.7            vc14h4cb57cf_1  [vc14]  anaconda
zc.lockfile               2.0                pyhd3eb1b0_0
zeroconf                  0.24.0                   py37_3    david_baddeley
zipp                      3.11.0           py37haa95532_0
zlib                      1.2.11           vc14h1cdd9ab_1  [vc14]  anaconda
```
</details>

**To Reproduce**
Steps to reproduce the behavior:
1. `conda create -n pyme python=3.7 pyme-depends`
2. install msvc business
3. clone the head
4. `python setup.py develop`
5. `conda install pytest`
6. some failures

**Expected behavior**
pass the tests.

Less worried about cluster tests given I haven't double checked firewall issues yet, etc., but would ideally not hit import errors on countdir or neuroglancer.

**proposal**
- Unless we want to ship neuroglancer with pyme-depends we should mark the neuroglancer dsview module import test as skip if neuroglancer can't be imported
- unless we want to write a `countdir.c` for windows we should check `sys.win32`, separate countdir import test from `test_IO_imports`

David-Baddeley, 2023-01-24:
- reasonable to put countdir import behind a platform check
- hard to special case neuroglancer in the tests, maybe see if we can move the neuroglancer import to execution rather than import in the module itself.

David-Baddeley, 2023-01-24:
(the neuroglancer module is not imported by name, but rather gets imported automatically  by searching the DSView/modules directory)

closed 2023-04-13 by David-Baddeley