python-microscopy/python-microscopy #1330: [BUG] Hitting enter in some parts of PYMEAcquire GUI starts a time series acquisition

barentine, 2023-02-02:
**Describe the bug**
Enter get's grabbed in parts of the MainFrame other than the HDFSpoolFrame, resulting in accidentally starting spooling acquisitions, sometimes without noticing for awhile

**To Reproduce**
Steps to reproduce the behavior:
1. Go to camera integration time settings
2. Click on the number
3. type in whatever you want
4. hit enter
5. congratulations, you are saving to disk 

**Expected behavior**
Not hook enter to start an acq. Frankly not even from the spool frame - clicking a button with the visual cue of what you're about to do is nice, particularly when it opens shutters / fires lasers...

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10]
 - Python version:
 - numpy version:
 - wxpython version [if a UI bug]:

**Additional context**
I think many have experienced this - I ran into it again recently. Didn't see in it in quick issue search, but sorry if I missed it

David-Baddeley, 2023-02-02:
Suspect this is due to adding a BTN_DEFAULT (or something along those
lines) flag to the button. Good for visual styling, and when the spool
panel was a dialog, but safe to remove to avoid this.

David-Baddeley, 2023-02-07:
line 229 in HDFSpoolFrame.py looks like a good candidate.

David-Baddeley, 2023-02-07:
it's a `.SetDefault()` call rather than a flag ...

closed 2023-04-13 by David-Baddeley