python-microscopy/python-microscopy #1331: [ENHANCEMENT] make OS dark mode settings look better / be usable

barentine, 2023-02-02:
**Is your feature request related to a problem? Please describe.**
New laptop, I use dark mode (Mac, OSX 12.6.2), PYME look's ridiculous, which I'm "ok" with, but it's impossible to read text in some places.

**Describe the solution you'd like**
base GUI colors on
```
appearance = wx.SystemSettings.GetAppearance()
appearance.IsDark()
```
so we can make things readable. [Looks like we can even hook changes for those running light/dark mode based on time of day, etc.](https://discuss.wxpython.org/t/how-to-react-to-dark-mode-change/35905)

**Describe alternatives you've considered**
- changing my OS settings when I want to run PYME.
- make all text pink

David-Baddeley, 2023-02-02:
Seeing as we generally don't actually specify colours in many places (we
mostly just use the windowing system defaults - the exceptions being the
green text in the analysis settings and the histogram controls) dark mode
etc ... should just work (if the operating system handles it properly).
That said we are likely to run into problems with both wx.aui and traitsui,
both of which do hard code some colours.

It is possible that we need to do a little more (setting the colour
database??), but only if it's something like a single function call when
you start the program. I definitely don't want to get into the weeds of
defining colours in each panel etc ... It's not portable or maintainable
and we really need to let the OS take care of that!

David-Baddeley, 2023-02-02:
Basically, everything should be using
https://docs.wxpython.org/wx.SystemColour.enumeration.html#wx-systemcolour
for the colours, and so should get the dark mode stuff right without us
needing to get involved. I'm not sure why this it not the case ...

barentine, 2023-02-02:
Ah, sorry, forgot to add my screenshot! I agree - we need something easy for sure. I think there are a couple places where we hardcode colors and don't want to

[image]

David-Baddeley, 2023-02-02:
OK, the folding pane title bars are hardcoded on our part (we can change
that, but not entirely trivial). The tabs, on the other hand are from AUI,
and might be hard (although there is a chance that AUI has a dark mode by
now).

We could certainly make the unpainted image background black - potentially
across themes.

Have you checked pymevis (lots of traitsui stuff)?

David-Baddeley, 2023-02-02:
OK, biggest usability issues I can see would be:

1) notebook tabs. We would need to subclass `wx.lib.agw.aui.tabart.DefaultTabArtProvider` (and ideally `wx.lib.agw.aui.tabart.DefaultDockArtProvider`) over-ride the relevant methods, and call `.SetArtProvider()` and `.SetAutoNotebookTabArt()` methods on the `AuiManager` object in `AUIFrame.py`.

2) text boxes in traitsui generated GUIs (which appear with white text on a white background). Not quite sure how to go about changing these, but could be tricky (might involve some monkey patching or similar).

Until we've hit these two, I'm not sure it's worth spending much time on the asthetically problematic but perfectly usable fold panels. Addressing fold panels is complicated as we use background colours here to indicate recipe execution errors (i.e. not a simple change in one place).

barentine, 2023-02-02:
PYMEVis could also use some help:
[image]

[image]

barentine, 2023-04-12:
> 1. notebook tabs. We would need to subclass `wx.lib.agw.aui.tabart.DefaultTabArtProvider` (and ideally `wx.lib.agw.aui.tabart.DefaultDockArtProvider`) over-ride the relevant methods, and call `.SetArtProvider()` and `.SetAutoNotebookTabArt()` methods on the `AuiManager` object in `AUIFrame.py`.

the issue for the tabs is in `wx/lib/agw/aui/aui_utilities.py` with `GetBaseColour`. For Mac's, it basically hardcodes the base color to white for some reason, while letting other OS use `wx.SystemSettings.GetColour(wx.SYS_COLOUR_3DFACE)`:
```
In [1]: import wx

In [2]: if wx.Platform == "__WXMAC__":
   ...:     try:
   ...:         import Carbon.Appearance
   ...:     except ImportError:
   ...:         CARBON = False
   ...:     else:
   ...:         CARBON = True
   ...: 

In [3]: CARBON
Out[3]: False

In [4]: wx.Platform
Out[4]: '__WXMAC__'

In [5]:     if wx.Platform == "__WXMAC__":
   ...:         k = Carbon.Appearance.kThemeBrushToolbarBackground if CARBON els
   ...: e 52
   ...:         if hasattr(wx, 'MacThemeColour'):
   ...:             base_colour = wx.MacThemeColour(k)
   ...:         else:
   ...:             brush = wx.Brush(wx.BLACK)
   ...:             brush.MacSetTheme(k)
   ...:             base_colour = brush.GetColour()
   ...: 

In [6]: wx.MacThemeColour
Out[6]: <function wx._core.MacThemeColour>

In [7]: wx.MacThemeColour(52)
Out[7]: wx.Colour(236, 236, 236, 255)
```

zacsimile, 2023-12-26:
+1 This is hard to use.

[image]

zacsimile, 2024-04-18:
This has become a problem for me again. The easiest way to fix the traits stuff is to replace `traitsui.wx.constants.OKColor` with `wx.BLACK` if we're in dark mode. I'm having some trouble monkey patching this as the `constants.py` file makes a call to `wx.SystemSettings`, which forces the `wx.App` to already be up. But, it seems we import the traits prior to launching the app. Not quite sure where/how in the PYME code base to make the injection.

David-Baddeley, 2024-04-19:
If we are somewhat serious about having things change over automatically when you change from light to dark mode, we should use `OKColor=wx.SystemSettings.GetColour(wx.SYS_COLOUR_WINDOW)` rather than `wx.BLACK`.

David-Baddeley, 2024-04-19:
Also for consistency - the actual dark background is (23,23,23), not black

zacsimile, 2024-04-19:
Thanks! Yeah I was just following the style of what was in `traitsui` already. In my PR I have `OKColor = wx.SystemSettings.GetColour(wx.SYS_COLOUR_INFOBK)` commented as an alternative. I found this visually more appealing than `OKColor = wx.SystemSettings.GetColour(wx.SYS_COLOUR_WINDOW)` but perhaps this is not a good argument. 

My main issue is figuring out how to inject it properly at runtime. Still haven't found a good way to do this.

David-Baddeley, 2024-04-19:
Injecting at runtime looks like it will be a dog of a task thanks to how traitsui is structured (a combination of relative imports and the fact that they are assigned to static variables when the various editor modules are loaded). It's the relative imports which really kill us - e.g. `from .constants import OKColor` in the editor. I haven't quite worked out if it's the relative bit (which might mean that `from traitsui.wx import constants' actually results in a different instance of the module) of the direct `from xx import OKColor`bit).

To get to a "skinable" traitsui, we would need a fairly major changes across multiple parts of the traitsui code.

David-Baddeley, 2024-04-20:
Actually I'm really confused - in theory we should be able to patch this by importing and monkey-patching immediately after app creation, but for some reason it doesn't work. The relative import should not be an issue - we only get one copy in sys.modules regardless of how it's imported.

David-Baddeley, 2024-04-20:
OK - I think I've got it. When importing `traitsui.wx.constants`, `traitsui.wx.__init__` executes first. This imports all the wx ui modules (creates the toolkit), so everything is there before you can monkey patch.

David-Baddeley, 2024-04-20:
Truely hideous monkey patch ...

```python
class VisGuiApp(wx.App):
    def __init__(self, filename, use_shaders, cmd_args, *args):
        self.filename = filename
        self.use_shaders = use_shaders
        self.cmd_args = cmd_args
        wx.App.__init__(self, *args)
        
        
    def OnInit(self):
        
        ### Begin traitsui monkey-patch
        old_white = wx.WHITE
        wx.WHITE = wx.SystemSettings().GetColour(wx.SYS_COLOUR_WINDOW)
        import traitsui.wx.constants as constants
        wx.WHITE = old_white
        ### End traitsui monkey-patch
        
        self.main = VisGUIFrame(None, self.filename, use_shaders=self.use_shaders, cmd_args=self.cmd_args)
        self.main.Show()
        self.SetTopWindow(self.main)
        return True

``

David-Baddeley, 2024-04-20:
Slightly nicer monkey patch ....

```python
import sys
import os.path

from importlib.abc import MetaPathFinder
from importlib.util import spec_from_file_location

class MyMetaFinder(MetaPathFinder):
    def find_spec(self, fullname, path, target=None):
        if 'traitsui.wx.constants' in fullname:
            #print(fullname, path, target)

            return spec_from_file_location(fullname, os.path.join(os.path.dirname(__file__), 'traitsui_constants.py'))

        return None # we don't know how to import this
    
sys.meta_path.insert(0, MyMetaFinder())
````

Will commit shortly - At least for the simple stuff ...

David-Baddeley, 2024-04-20:
That probably needs more context/explanation - we install an import hook to effectively replace constants.py with our own version.

David-Baddeley, 2024-04-20:
[image]

With #1500 we should be functional, if not particularly asthetic

David-Baddeley, 2024-04-20:
now somewhat more asthetic, although could still use work ...

[image]

[image]

[image]

TODOs ... 
- contrast between pipeline outputs and background
- contrast between active and inactive tabs
- icons for PYMEImage toolbar
- fold-panel icons
- free space in spooler
- layer ui
- metadata viewer
- (maybe) histlimits