python-microscopy/python-microscopy #1360: [SUPPORT] What is PYMEImage 'X-Z' or 'Y-Z' slice supposed to do?

barentine, 2023-05-09:
So I open a two-color stack, crop a piece, and want to scroll through a side-view. 

I click 'X-Z' to get the side view slice, and when I hit the play button, the crosshairs move and the 't' updates in the bottom left hand corner, but the displayed image does not change.

Is this intended? Or a bug?

David-Baddeley, 2023-05-09:
This is intended, although I can see an argument for wanting to play the non-visible dimension (i.e. swapping the t slider for a y slider when the view changes to XZ). The main reason this is not currently done is that playing along anything other than the z or t axes has unacceptable performance (at least for large series). This is due to the lazy data loading, and the fact that the majority of series are chunked along z/t (loading an XZ slice requires loading a 2D frame for every visible z line and pulling a single line out of it - this entire loading process is repeated for each y postion when scrolling along the y axis). Getting acceptable playback would require implementing additional caching and dramatically increasing the memory footprint of PYMEImage. 

As currently implemented, you can step through XZ or YS slices using the i, j,k,l keys (as used in games as arrow key alternatives - I believe i-k move in y and j-l move in x).

David-Baddeley, 2023-05-09:
And by unacceptable performance I mean that pushing the motion keys for long enough that they repeat  (analogous to the current slider play) will typically leave the program unresponsive for long enough that it appears to have crashed while it is catching up with the key presses.