python-microscopy/python-microscopy #1361: [BUG] Numpy version compatibilty / pinning

David-Baddeley, 2023-05-09:
I've just come across a couple of issues with deprecated numpy functionality both related to indexing with non-integers (where I thought we'd got everything), but also a more recent deprecation (happened sometime between 1.22 and 1.24) relating to bools, which break our built packages. After fixing these we probably need to look at pinning numpy again to prevent new breakages - but ideally keeping these pins moving.

barentine, 2023-05-16:
Any updates on this, @David-Baddeley ? Including where specifically the issues are? I get nervous hearing about indexing issues when working with e.g. 5D datasources and the like. Thanks!

David-Baddeley, 2023-05-16:
Been pretty snowed under the past couple of weeks ... Indexing issues were
in the biplane/ratiometric fitting code, but are now fixed.