python-microscopy/python-microscopy #1373: [BUG] DBSCAN seems to be broken

lukasfue, 2023-06-06:
**Describe the bug**
DBSCAN gives an error when trying to apply it data.
- I used the same data and parameters that have worked in the past and still get the error

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Analysis > Clustering > DBSCANClump
2. Apply desired parameters
3. See error:
 Error whilst running Analysis>Clustering>DBSCAN Clump
==============================================
python-microscopy=22.01.12[unknown]
python=3.7.9, platform=win32
numpy=1.19.1, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\PYME\LMVis\Extras\clusterAnalysis.py", line 93, in OnClumpDBSCAN
    recipe.add_modules_and_execute([clumper,])
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\PYME\recipes\recipe.py", line 457, in add_modules_and_execute
    self.execute()
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\PYME\recipes\recipe.py", line 203, in execute
    m.execute(self.namespace)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\PYME\recipes\localisations.py", line 446, in execute
    self.searchRadius, self.minClumpSize)
TypeError: dbscan() takes from 1 to 2 positional arguments but 3 were given

**Expected behavior**
Execution of DBSCAN algorithm

**Screenshots**
This was with 4Pi SMS data that I've applied DBSCAN to before and it worked fine (about 8 months ago)
Here are the parameters I used:
[image]

David-Baddeley, 2023-06-06:
Was fixed in https://github.com/python-microscopy/python-microscopy/pull/1201

Upgrading to a more recent PYME version should fix things.

closed 2023-06-06 by David-Baddeley

csoeller, 2023-06-07:
Yes, I'd also guess you need a `git pull` or `git pull master`. I have never fully gotten into the git command line interface and would recommend a GUI interface for like minded users. I am using _gitkraken_ and via the github student/teacher program can access the pro version for free.

David-Baddeley, 2023-06-07:
How did you install the base environment? conda? pip? My guess is that the pytables package didn't bundle the HDF5 libraries. This shouldn't be an issue for the conda package, but could be for a pip install (which might assume the libraries were already installed as would often be the case on Mac or Linux).

David-Baddeley, 2023-06-07:
should add that `git status` generally tells you if you have any uncommitted local changes, not if there are any upstream changes.

lukasfue, 2023-06-08:
I used "conda create -n pyme pyme-depends python=3.7". Before uninstalling, entering `git pull` or `git status` said I was up to date with the python-microscopy master branch.

csoeller, 2023-06-08:
I am not sure if you are familiar with the conda version of Python virtualenvs which we use heavily to make various PYME installs that do not interfere with each other. We highly recommend this approach and we just then have launcher scripts to launch the various PYME apps from the suitable conda environment. The approach is described in [Installing-PYME-with-py3-win10](https://github.com/csoeller/pyme-install-docs/blob/master/Installing-PYME-with-py3-win10.md) and a brief doc on [making suitable windows launchers](https://github.com/csoeller/pyme-install-docs/blob/master/PYME-windows-launchers.md).

That way you can make installs based on different Python versions, experimental installs from different git branches etc and also try a new install without wiping an existing install.

Perhaps this is overkill for you and possibly may seem even more confusing, it all depends on personal pereferences.

csoeller, 2023-06-08:
PS - I see you do use named environments but wonder if you keep several around. That is were switching between envs can become useful.

David-Baddeley, 2023-06-08:
@lukasfue - That implies you have a conda package for pytables, which should have included the HDF5 libraries. The corollary is that the specific version of the package that you have installed is broken. My suggestion:

- do a `conda list` to determine which version of pytables you have  (it would also be interesting to know if it is from the main anaconda channel(s) or from conda-forge)
- install a different (likely earlier version) - `conda install pytables=x.x`. Repeat until you find one that works.

If this problem is reproducible, we might need to pin pytables in pyme-depends to ensure that we get a good one.

lukasfue, 2023-06-12:
@David-Baddeley - I ended up installing pytables=3.7, and I was able to then launch visgui without issue and DBSCAN worked in the fresh install, so I suppose the original problem was somehow related to my previous installation. With that said, it seems that maybe the pytables dependency needs updating? Whatever it's currently set to doesn't seem to work since Yujin and I (both my PC and Macbook) had this issue when trying to install PYME, but updating pytables to 3.7 resolved it in all cases.