python-microscopy/python-microscopy #1377: [ENHANCEMENT] Would it be worth getting sphinx to put more docstrings on the website than we currently have?

barentine, 2023-06-09:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when someone asks me what a function does and I send them a screenshot of a docstring rather than a link to our documentation. This is mostly an issue for functions people add from the GUI, since the recipe docstrings display if adding a module.

**Describe the solution you'd like**
If someone asks me what's up with Area calculation for PYMEVis > Analysis > Measure objects, I'd like for them to 
- have a tool tip that tells them something e.g. at least what to search on our site maybe
- be able to search our doc and find something. For example: [objectMeasure](https://www.python-microscopy.org/doc/api/PYME.Analysis.points.objectMeasure.html) would ideally contain the docstrings for the functions in objectMeasure, e.g. `measure`, not just a blank page

**Describe alternatives you've considered**
manually writing doc

David-Baddeley, 2023-06-09:
This is likely a bug in either docstring formatting or sphinx invocation - We should show all the docstrings. Solution is to look for errors in the output - the most likely cause is would be for `objectMeasure` to import a module which is neither available in the build environment nor mocked up `docs/conf.py`. In general we want to generate docs in a stripped down environment and mock modules which are not present rather than generating docs in a full environment as doing the later tends to result in the duplication of lots of 3rd party docs.

David-Baddeley, 2023-06-09:
In terms of GUI menu functions, the long term vision here is to make *everything* recipe modules under the hood, and to piggy back off the documentation there.

David-Baddeley, 2023-06-09:
For objectMeasure in particular, we also really need to get onto #524

David-Baddeley, 2023-07-28:
and now fixed on the website (turns out you need to build the c coded extensions for objectMeasure to get generated)

David-Baddeley, 2023-08-14:
@barentine
Having fixed the documentation build issue, can we close this? We haven't really addressed the tooltip issue (or general documentation quality issues), but this seems overly specific for those.

barentine, 2023-08-17:
Yes! Thanks @David-Baddeley . Really nice to have this working

closed 2023-08-17 by barentine