python-microscopy/python-microscopy #1380: [SUPPORT] UnsatisfiableError when attempting to do a development install

lukasfue, 2023-06-12:
**What are you trying to do?**
I'm trying to do a development install and running into a package conflict.
Note: 
- Windows 11 machine (I'm not sure if PYME even works on Windows 11)
- Using Miniconda 3.7 

**Where are you stuck?**
Here is the error I receive:
[image]

In case its helpful, here is the conda list output:
[image]

I've tried reinstalling Miniconda 3.7, but that did not resolve the conflict.

David-Baddeley, 2023-06-12:
Super odd ... But the strict channel priority bit might be a clue, as could
the fact that all the packages in the base env seem to be from conda forge.
Do you have conda-forge in your channel list? Where is it?

Output of conda config --show could be useful.

David-Baddeley, 2023-06-12:
Unfortunately reinstalling doesn't sort out a broken config.

David-Baddeley, 2023-06-12:
To sort out a broken conda config, you need to remove `<home>/.condarc` before reinstalling miniconda but it might be possible to sort out without that:

I suspect it comes down to having `conda-forge` as the highest priority channel. The simplest fix might be

`conda config --remove channels conda-forge`

If you really want conda-forge, you can add it back at the bottom of the list (`conda config --append channels conda-forge`) so that it get used only if the default channels don't have a package. It seems to be pretty essential that it's not at the top of the list (i.e. `conda config --add channels conda-forge` seems to have a high chance of getting you into trouble). In general, though, I find it safest to leave it out of the default channels list altogether and manually specify it whenever I want to install a conda-forge package (i.e. `conda install -c conda-forge somepackage`).

lukasfue, 2023-06-13:
Removing conda forge seemed to result in a similar but slightly different error:

UnsatisfiableError: The following specifications were found to be incompatible with each other:
Output in format: Requested package -> Available versions
Package python conflicts for:
pyme-depends -> cherrypy -> python[version='>=3.10,<3.11.0a0|>=3.11,<3.12.0a0|>=3.9,<3.10.0a0|>=3.8,<3.9.0a0|>=3.5,<3.6.0a0|>=3.6|>=3.8|>=3.7|>=3.5|>=3.7.1,<3.8.0a0']
python=3.7
pyme-depends -> python[version='2.7.*|>=2.7,<2.8.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0']
Note that strict channel priority may have removed packages required for satisfiability.

In any case, I was able to get the plugin working on that machine with the one-click install version thanks to yours and Andrews advice in the other post I made yesterday and that was the ultimate goal. So, thank you for the assistance!

closed 2023-06-13 by lukasfue