python-microscopy/python-microscopy #1381: [ENHANCEMENT] Ability to install plugins in the one-click install version of PYME.

lukasfue, 2023-06-12:
**Is your feature request related to a problem? Please describe.**
Some users like the simplicity of the one-click install, but it seems that it isn't possible to install plugins without a development install of PYME (or at least it wasn't obvious to me how to do so).

**Describe the solution you'd like**
Making it possible to install plugins to the one-click install PYME version and instructions on how to do so (or just the latter if it is already possible).

barentine, 2023-06-12:
@lukasfue if you hit the windows key and type `anaconda prompt` the search should show you the anaconda prompt @David-Baddeley is talking about under `Apps` with some parenthesis around a name associated with your executable install. Something like: 
[image]

More broadly though, sounds like we could use:
- [ ] a Plugins page under the Installation docs. Ideally this should show people how to install both conda and source plugins, and give enough details on how to find the anaconda prompt gifted them by their executable install (if that's the route they took)
[image]
- [ ] An update to the plugin template repository to actually handle dist settings properly (note @csoeller raised this some time back [here](https://github.com/python-microscopy/pyme-plugin/issues/8)). Note for others we discussed the need to update that repository last week - it's currently set up to help folks with a deprecated module discovery, and the new yaml config is simpler.
- [ ] a Plugin's page on the main website? Somewhere we can list plugins so folks can easily find them. `ch_shinkwrap` might be a great example, but you cannot currently discover it (or it's preprint) from the PYME website or docs

lukasfue, 2023-06-13:
Thank you David and Andrew. The key was finding that Anaconda Prompt specific to PYME.

Best,
Lukas

closed 2023-06-13 by lukasfue