python-microscopy/python-microscopy #1384: [SUPPORT] PYMEVis post-processing localizated data from astigamatism z-stacking PAINT

Yujin-Bao, 2023-06-13:
I am trying to post-process the localized data from astigamatism z-stacking PAINT in PYMEVis, but I have problem understanding the 'view' tab and doing drift correction.

**Where are you stuck?**
I captured 4 slices of frames with a step size of 400 nm. For each slice I captured 50,000 frames and then move to another slice. The drift correction was conducted simutaneously from a transmitted-light channel, so the lateral drift values were recorded.

[image]
I filtered the fitResults_z0 to the range -400~0 nm and the colorbar on the right indicates the z position. The xz view shows repetitive color and each period is from red (lowest z) to purple (highest z). My understanding is that each repetitive period stands for a slice I took, and the software has merge the slices in the correct sequence instead of overlapping them with each other because of the metadata have been recorded. I am wondering if my understanding is correct.

For the drift correction, I usually use two ways: mapping (reverse the recorded lateral drift) and rcc. However, the problem I am having is that neither of these two methods give me good drift correction. I think this is because when the software does z-stacking, it still records the lateral drift on top of the previous slice instead of recording drift for individual slice. But I have no idea how to correct it.

David-Baddeley, 2023-06-13:
Hi Yujin,

can I safely assume you acquired this using the z-stepped option in PYMEAcquire, albeit with a longer dwell time than the default? If so, PYMEVis will automatically shift the z-planes into the correct position (which is why you see 4 slices when you colour-code by fitted z position). The z position in PYMEVis is calculated  as `z = z_focus + z_fit` where z_focus is the stage position, and z_fit is the fitted z position (`fitResults_z0`). 

On the basis of the information you've provided int the post it's a bit hard to draw any conclusions about drift correction:

- Is it predominately lateral or axial drift that is an issue?
- Is the drift well corrected within a single plane?

The one thing I will note is that rcc is likely to require overlap between the planes in order for it to work (e.g. by expanding the filter range for [-400, 0] to [-400, 400]). Is there any reason you clipped away all the positive fits?

When using transmitted light drift tracking in combination with z-stacking, the comparison is made to the plane in the transmitted light calibration stack which corresponds to the target position, so it should adapt automatically. One potential source of issues here would be if there was lateral motion during acquisition of the calibration stack.

Yujin-Bao, 2023-06-14:
Hi David,

Thanks for the reply! Yes I used z-stepped option and set step to 0.4 um, slices to 4 and dwell to 50,000.

For the drift correction, the lateral drift is an issue because the axial drift has been corrected by transmitted-light focus lock. And the drift correction by both mapping and rcc work quite well within a single plane.

I did clipping because for single plane astigmatism, only a certain z range gives good precision. I clipped the range of 400 nm not to get overlaps between slices. In this data set I think either [-200, 200] or [-400, 0] is OK. But maybe I was wrong and I should let them overlap? But if this is the case, one localized point could be reconstructed from different slices and shown as different color overlapping.

From what you said it seems drift correction by mapping should work well. But I do not quite understand 'lateral motion' here. Do you mean the stage shifted laterally when the piezo moved from one plane to another?

Best,
Yujin

David-Baddeley, 2023-06-14:
Once you move past the debugging stage, you will definitely want overlap so as not to get striping artefacts in z (your detection probability drops off as you move away from a nominal z=0, so the localisation density will appear to drop off as well). Ultimately you might want to consider a smaller axial step size (e.g. 200 nm) and increased overlap to get approximately uniform detection probability through z. 

If you use `fitResults_z` to colour points, you will certainly get overlapping colours, but I'd argue that `fitResults_z` is only really useful for debugging when you specifically want to look at behaviour within a slice - in the end you'll want to use the combined z. 

Yep - if the stage moves laterally when the piezo moves you could be in for a hard time. I've seen this happen in inverted microscopes when the sample is on the slide and there is excess mountant between the coverslip and the slide. For low-working distance lenses you can end up lifting the entire slide when trying to focus a sample on the slide. 

If I understand correctly, it's the mapping between slices rather than the drift correction per se which is problematic. This could be something like lateral motion during stepping, but there are a number of other factors which could play a role here. Coma in the PSF can lead to an apparent  lateral movement of points between top and bottom of a slice. If your microscope isn't properly set up in the infinite conjugate you can also get magnification differences as you step. Feel free to flick me a copy of the data and/or the PSF if you can't figure it out. Doing a z stack of a field of beads (as you might use for PSF calibration) can also be useful when trying to chase down magnification changes etc ...