python-microscopy/python-microscopy #1386: [BUG] Save Snapshot error

lukasfue, 2023-06-16:
**Describe the bug**
I'm trying to save a snapshot with View > Save Snapshot, but get an error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to View > Save Snapshot
2. See error

**Expected behavior**
A pop-up window to select snapshot settings and then a saved snapshot.

Here is the error:

 Error whilst running View>Save Snapshot
==============================================
python-microscopy=23.05.17.post2.post0.dev[git]
python=3.7.12, platform=win32
numpy=1.21.6, wx=4.1.1 msw (phoenix) wxWidgets 3.1.5

Traceback
=========
Traceback (most recent call last):
  File "c:\users\laf62\code\python-microscopy\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "c:\users\laf62\code\python-microscopy\PYME\LMVis\Extras\snapshot.py", line 135, in <lambda>
    vis_fr.AddMenuItem('View', 'Save Snapshot', lambda e: save_snapshot(vis_fr.glCanvas))
  File "c:\users\laf62\code\python-microscopy\PYME\LMVis\Extras\snapshot.py", line 110, in save_snapshot
    settings.configure_traits(kind='modal')
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traits\has_traits.py", line 2171, in configure_traits
    args,
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\toolkit.py", line 218, in view_application
    context, view, kind, handler, id, scrollable, args
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\view_application.py", line 101, in view_application
    args=args,
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\view.py", line 462, in ui
    ui.ui(parent, kind)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\ui.py", line 244, in ui
    self.rebuild(self, parent)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\toolkit.py", line 132, in ui_modal
    ui_modal.ui_modal(ui, parent)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\ui_modal.py", line 46, in ui_modal
    _ui_dialog(ui, parent, BaseDialog.MODAL)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\ui_modal.py", line 65, in _ui_dialog
    BaseDialog.display_ui(ui, parent, style)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\ui_base.py", line 71, in display_ui
    ui.owner.init(ui, parent, style)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\ui_modal.py", line 139, in init
    sw = panel(ui, window)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 274, in panel
    panel, content[0], ui
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 433, in fill_panel_for_group
    panel, group, ui, suppress_label, is_dock_window, create_panel
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 589, in __init__
    self.add_items(content, panel, self.sizer)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 909, in add_items
    editor.prepare(item_panel)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\editor.py", line 236, in prepare
    self.init(parent)
  File "C:\Users\laf62\.conda\envs\pyme\lib\site-packages\traitsui\wx\file_editor.py", line 96, in init
    sizer.Add(control, 1, wx.EXPAND | wx.ALIGN_CENTER)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL))" failed at ..\..\src\common\sizer.cpp(2184) in wxBoxSizer::DoInsert(): Vertical alignment flags are ignored with wxEXPAND

zacsimile, 2023-06-16:
See #1285. Try upgrading traits to 7.4.

lukasfue, 2023-06-20:
I think this may have been caused by conda-forge. I reinstalled with conda-forge removed and this problem is now resolved.

closed 2023-06-20 by lukasfue