python-microscopy/python-microscopy #1387: watershed function[SUPPORT]

Z-Qing, 2023-06-18:
Hi, everyone. 
I installed the pyme using "python setyp.py develop" and I realized that in the setup.py file in ctrib directory the cpmatch is commented out. If I simplely add the subdirectory in the setup file, I will get the error " config = setup_module.configuration(*args)
TypeError: configuration() takes 0 positional arguments but 1 was given". 
Is there any way to bypass this problem and get the watershed function work?

Thanks for your help.

David-Baddeley, 2023-06-18:
The version of cpmath we bundle is extremely out of date, and I suspect the include line was simply commented out because something in the build process had stopped working. I don't think I've looked at this since we moved to py3.

I could look into this a bit more, but the only place cpmath is used is in the watershed function in PYMEImage. In the interests of future maintainablity, I wonder if it instead makes sense to replace the functionality using the watershed function in skimage, which is actively maintained, and to remove cpmath entirely. This would also be a good opportunity to add a recipe module so that it can be scripted and isn't only manual-only.

David-Baddeley, 2023-06-18:
Oops, there is already a recipe module which uses the skimage version. We should just modify the menu one to use that, so we are consistent. Temporary workaround would be to add the watershed module in the recipe editor, rather than choosing watershed from the menu.

Z-Qing, 2023-06-23:
The watershed function I tried to use is in PYMEVis. After generate 3d Gaussian, there is a processing|watershed tab in the pop-up window. But using a recipe module is a good ideal. Thanks!

closed 2024-06-25 by Z-Qing