python-microscopy/python-microscopy #1397: [BUG] MDHandler error in SKFilter recipe module

barentine, 2023-06-27:
**Describe the bug**

```
 Error whilst running Recipes>Run Current Recipe	F5
==============================================
python-microscopy=23.06.15.post0.dev[git]
python=3.7.16, platform=win32
numpy=1.21.5, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\userfiles\andrew\code\python-microscopy\PYME\recipes\recipe.py", line 237, in execute
    m.execute(self.namespace)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\recipes\base.py", line 285, in execute
    v.mdh.copyEntriesFrom(mdh) # copy / overwrite with module processing parameters
  File "c:\userfiles\andrew\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 405, in copyEntriesFrom
    self.setEntry(en, mdToCopy.get(en))
  File "c:\userfiles\andrew\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 619, in setEntry
    self.__dict__[entPath[0]].setEntry('.'.join(entPath[1:]), value)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 619, in setEntry
    self.__dict__[entPath[0]].setEntry('.'.join(entPath[1:]), value)
AttributeError: 'dict' object has no attribute 'setEntry'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "c:\userfiles\andrew\code\python-microscopy\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\DSView\modules\recipes.py", line 102, in RunCurrentRecipe
    self.outp = self.activeRecipe.execute(input=self.image)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\recipes\recipe.py", line 282, in execute
    raise RecipeExecutionError('Recipe execution failed in module %s' % self._failing_module._module_name, self ) from e
PYME.recipes.recipe.RecipeExecutionError: Recipe execution failed in module skfilters.SKF_threshold_multiotsu
 Error whilst running Recipes>Run Current Recipe	F5
==============================================
python-microscopy=23.06.15.post0.dev[git]
python=3.7.16, platform=win32
numpy=1.21.5, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\userfiles\andrew\code\python-microscopy\PYME\recipes\recipe.py", line 237, in execute
    m.execute(self.namespace)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\recipes\base.py", line 285, in execute
    v.mdh.copyEntriesFrom(mdh) # copy / overwrite with module processing parameters
  File "c:\userfiles\andrew\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 405, in copyEntriesFrom
    self.setEntry(en, mdToCopy.get(en))
  File "c:\userfiles\andrew\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 619, in setEntry
    self.__dict__[entPath[0]].setEntry('.'.join(entPath[1:]), value)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 619, in setEntry
    self.__dict__[entPath[0]].setEntry('.'.join(entPath[1:]), value)
AttributeError: 'dict' object has no attribute 'setEntry'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "c:\userfiles\andrew\code\python-microscopy\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\DSView\modules\recipes.py", line 102, in RunCurrentRecipe
    self.outp = self.activeRecipe.execute(input=self.image)
  File "c:\userfiles\andrew\code\python-microscopy\PYME\recipes\recipe.py", line 282, in execute
    raise RecipeExecutionError('Recipe execution failed in module %s' % self._failing_module._module_name, self ) from e
PYME.recipes.recipe.RecipeExecutionError: Recipe execution failed in module skfilters.SKF_threshold_multiotsu

```

**To Reproduce**
Steps to reproduce the behavior:
1. Open an h5 2 color image
2. run this recipe:
```
- base.ExtractChannel:
    inputName: input
    outputName: star_orange
- filters.GaussianFilter:
    dimensionality: XYZ
    inputName: star_orange
    outputName: smoothed_star_orange
- skfilters.SKF_threshold_multiotsu:
    dimensionality: XYZ
    inputName: smoothed_star_orange
    outputName: otsu3class
```

7. See error

**Additional context**
I saved this H5 after making a composite from two OBF stacks.

David-Baddeley, 2023-06-28:
Could you provide a copy/print of the metadata. The most likely issue is that something has set a metadata entry to a dict, which is not permitted.

barentine, 2023-06-28:
Of course! Thanks for looking, David
```
image.mdh
<NestedClassMDHandler>:

voxelsize.x: 0.04999999802819659
voxelsize.y: 0.050000001768542766
voxelsize.z: 0.050000001768542766
voxelsize.units: um
Camera.ReadNoise: 0
Camera.NoiseFactor: 1
Camera.ElectronsPerCount: 1
Camera.TrueEMGain: 1
Camera.ADOffset: 0
ChannelNames: ['ORANGE STED', 'RED STED']
tIndex: 0
Origin.x: 0
Origin.y: 0
Origin.z: 0
cropping.xslice: (0, 385, 1)
cropping.yslice: (0, 228, 1)
cropping.zslice: (0, 114, 1)
EstimatedLaserOnFrameNo: 0
```

barentine, 2023-06-28:
And of the intermediate Gaussian Filtered'd:
```
>>> recipes.activeRecipe.namespace['smoothed_star_orange'].mdh
<NestedClassMDHandler>:

voxelsize.x: 0.04999999802819659
voxelsize.y: 0.050000001768542766
voxelsize.z: 0.050000001768542766
voxelsize.units: um
Camera.ReadNoise: 0
Camera.NoiseFactor: 1
Camera.ElectronsPerCount: 1
Camera.TrueEMGain: 1
Camera.ADOffset: 0
ChannelNames: ['ORANGE STED']
tIndex: 0
Origin.x: 0
Origin.y: 0
Origin.z: 0
cropping.xslice: (0, 385, 1)
cropping.yslice: (0, 228, 1)
cropping.zslice: (0, 114, 1)
EstimatedLaserOnFrameNo: 0
Parent: Filtered Image 1
Processing.ExtractChannel.Channeltoextract: 0
Processing.GaussianFilter.Sigmas: [1.0, 1.0, 1.0, 0.0]
Processing.GaussianFilter.Dimensionality: XYZ
Processing.GaussianFilter.Sigmat: 0.0
Processing.GaussianFilter.Sigmax: 1.0
Processing.GaussianFilter.Sigmay: 1.0
Processing.GaussianFilter.Sigmaz: 1.0
```

PR #1408 (2023-07-27): Tentative fix for #1397
Fixes #1397. @barentine - untested, could you please try out?
closed 2023-07-27 by David-Baddeley