python-microscopy/python-microscopy #1400: [ENHANCEMENT] Export ImageStack to series of images suitable for video-generation by ffmpeg

barentine, 2023-07-05:
**Is your feature request related to a problem? Please describe.**
I'm frustrated when I don't know how to quickly get PYMEImage rendering into a video other than doing a screen-recording (which I'm not particularly great at)

**Describe the solution you'd like**
Similar to PYMEVis animation, I'd like to export every frame from a stack as a png or so such that I can run FFMPEG and get a movie suitable for presentation

**Describe alternatives you've considered**
screen recording
imageJ

David-Baddeley, 2023-07-05:
Definitely worth considering, but potentially a bit more involved than it might first seem. The rendering code is pretty well embedded in the GUI and the current snapshot functionality (`View->Save as PNG`) is essentially a screen-grab. To automatically render every frame in a series, in a smooth exporter you'd want to either abstract or reproduce the rendering code in a form which didn't involve drawing things to a `wx.DC`. Reproduction will be bad for maintenance, and too much abstraction might not be great for performance (operating system graphics routines are generally pretty efficient - so pulling all rendering out into code which doesn't use the windowing libraries might not be ideal). 

With the current code structure it wouldn't be too hard to abstract the colour mapping code, but if we move to an openGL renderer in the future (somewhat on the cards) this logic will likely move into a shader, so we'd end up maintaining two versions. If you want any of the overlays (scale bar, annotations, etc ...) to be present in the exported video then abstracting the rendering code becomes quite tricky.

This leads me to the conclusion that, despite being a bit clunky, a screen capture approach (as we do for animation in PYMEVis) might be the most pragmatic. Basically just call the logic behind `View->Save as PNG` in a loop (there would be some fiddling here to make sure the display updated between frames - you might need to hook wx.EVT_IDLE to do your saving and incrementing).

barentine, 2023-08-17:
Closed with #1403

closed 2023-08-17 by barentine