python-microscopy/python-microscopy #1401: [BUG] cropping metadata slice starts always starts at zero

barentine, 2023-07-05:
**Describe the bug**
The origin is updated, but the crop slice doesn't help you e.g. go back and easily crop the same from another channel or so. Here I took a 25 pixel crop in Y starting from pixel 70 (0.05 um pixels). It shows the 25 pixel slice, but starts at 0. 
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to export cropped
2. crop something
3. look at its metadata
4. wish that the slice were a bit more useful

**Expected behavior**
show me the slice it came from with respect to its parent image. A slice metadata that just shows the current image size is kind of useless

David-Baddeley, 2023-07-24:
hmm, the slice metadata is calculates on lines 167-171 of IO/dataExporter.py

There is an error here (we are using the shape of the data after cropping in the call to `indices()`, rather than before), but that shouldn't give rise zeros for the start of the slice.

David-Baddeley, 2023-07-24:
that's for .hdf - were you talking .hdf or .tif?

PR #1407 (2023-07-26): Fix cropping metadata
Fixes #1401
closed 2023-07-26 by David-Baddeley