python-microscopy/python-microscopy #1409: [SUPPORT] issue of conflicts with pyme-depends and python>3.7 ?

csoeller, 2023-07-26:
This is similar to #1380, but subtly different. I have tried to `conda install` `pyme-depends` with python versions > 3.7. With all version 3.8-3.11 I get a conflict, again with cherrypy as in #1380, but there is **no** conda-forge channel in my channel list.

This is on windows 11 with a recent miniconda3 install. I list my channels, conda info and the conflict upon installing below. Curiously, forcing `conda-forge` manually into the channel list leads to a successful install (after much thinking about solving the environment), i.e. the following command ultimately succeeds: `conda create -n pyme-py38 -c conda-forge python=3.8 pyme-depends`.

Anyway, here are the details. Any hints appreciated.

```
conda create -n pyme-py38 python=3.8 pyme-depends
Retrieving notices: ...working... done
Collecting package metadata (current_repodata.json): done
Solving environment: unsuccessful attempt using repodata from current_repodata.json, retrying with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: -
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package python conflicts for:
pyme-depends -> cherrypy -> python[version='>=3.10,<3.11.0a0|>=3.11,<3.12.0a0|>=3.5,<3.6.0a0|>=3.8|>=3.6|>=3.7|>=3.5|>=3.7.1,<3.8.0a0']
python=3.8
pyme-depends -> python[version='2.7.*|>=2.7,<2.8.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0']
```

These are my channels:

```
   conda config --show channels
     channels:
         - david_baddeley
         - defaults
```
and conda info: 

```
conda info

     active environment : base
    active env location : C:\ProgramData\miniconda3
            shell level : 1
       user config file : C:\Users\XXXX\.condarc
 populated config files : C:\Users\XXXX\.condarc
          conda version : 23.5.2
    conda-build version : not installed
         python version : 3.11.4.final.0
       virtual packages : __archspec=1=x86_64
                          __win=0=0
       base environment : C:\ProgramData\miniconda3  (read only)
      conda av data dir : C:\ProgramData\miniconda3\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/david_baddeley/win-64
                          https://conda.anaconda.org/david_baddeley/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\ProgramData\miniconda3\pkgs
                          C:\Users\XXXX\.conda\pkgs
                          C:\Users\XXXX\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\XXXX\.conda\envs
                          C:\ProgramData\miniconda3\envs
                          C:\Users\XXXX\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.5.2 requests/2.29.0 CPython/3.11.4 Windows/10 Windows/10.0.22621
          administrator : False
             netrc file : None
           offline mode : False
```

David-Baddeley, 2023-07-27:
This looks like it's a packaging issue with cherrypy in the anaconda channels.

So far I haven't had any success installing without using conda-forge for python >=3.8 (I think the default channels are just not up to scratch these days), so it seems likely that conda-forge will be recommended going forward - at least for 3.8 and above. The resolution time can be somewhat improved by using mamba (or the libmamba solver with conda) instead of the default conda.

csoeller, 2023-12-09:
Closing for now; hopefully will still show up in searches when others bump into this (or I have forgotten about the issue in future).

closed 2023-12-09 by csoeller

csoeller, 2024-01-22:
One more comment on this issue (though closed). We have now made a little set of code to automate installing environments with PYME and more recent Python version using conda-forge for package dependencies, including full `arm64` installs on mac. This suggests up to Python 3.9 everything seems fine, 3.10 exposes some automated float->int argument conversions that are not done anymore but looks fixable. See also [PYME-test-env](https://github.com/csoeller/PYME-test-env).